esphome/issues #2326: Not a bug, but misspeling

karolzlot, 2021-08-17:
### The problem

[image]

### Which version of ESPHome has the issue?

-

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

-

### What platform are you using?

ESP32

### Board

-

### Component causing the issue

-

### Example YAML snippet

```yaml
-
```

### Anything in the logs that might be useful for us?

```txt
-
```

### Additional information

-

oxan, 2021-08-17:
@jesserockz Seems the typo fix introduced another typo :rofl:

jesserockz, 2021-08-17:
Haha, thanks. Will get this sorted

closed 2021-08-18 by jesserockz