esphome/issues #2327: DSMR - Gas consumed empty

bakboord4, 2021-08-18:
### The problem

DSMR internal component.
Sensor Gas consumed is not reporting any values.
The Marcel Zuidwijk's solution with ESP-Link _does_ report Gas consumed.
Issue is also reported here:  https://github.com/zuidwijk/dsmr/issues/7
My Gasmeter and or Energy meter is replaced.

### Which version of ESPHome has the issue?

ESP 2021.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8.7

### What platform are you using?

ESP8266

### Board

Wemos D1 mini - SlimmeLezer by M. Zuidwijk

### Component causing the issue

DSMR

### Example YAML snippet

```yaml
gas_delivered:
      name: "Gas Consumed"
```

bakboord4, 2021-08-19:
addition to the problem: 
problem states: My Gasmeter and or Energy meter is replaced.
checked, should read: My gasmeter is replaced.

glmnet, 2021-08-21:
I don't understand this issue correctly. Do you just need the GAS_MBUS_ID configurable?

bakboord4, 2021-08-21:
@glmnet 
Thank you for looking in to this issue. First, I am a knowledgable user but not a developer.
Unfortunate the link to issue/7 in zuidwijk/dsmr is gone. As I understood/remember the discussion in /zuidwijk/dsmr, a replaced gasmeter gets a new number i.e old meter 1 -> new meter 2. Replacement of gas meters is not uncommon.  I try to contact M. Zuidwijk for more information.

zuidwijk, 2021-09-01:
> @glmnet
> Thank you for looking in to this issue. First, I am a knowledgable user but not a developer.
> Unfortunate the link to issue/7 in zuidwijk/dsmr is gone. As I understood/remember the discussion [...]

Sorry for my lack of response ... I was busy soldering 600 devices :-/

I think, and correct me if I'm wrong, the best solution would be a configurable GAS_MBUS_ID.

```yaml
sensor:
  - platform: dsmr
    gas_delivered:
      name: "Gas Consumed"
      gas_id: 1/2/etc
```

When it is not configured in the yaml, then it uses it's default value of '1' as that is most common.

zuidwijk, 2021-09-01:
> @zuidwijk actually I made an attempt already, see [esphome/esphome#2157](https://github.com/esphome/esphome/pull/2157) I comment using this approach:
> 
> ```yaml
> # Set custom gas_mbus_id
> dsmr:
>   gas_mbus_id: 2
> ...
> ```
> 
> @bakboord4 do you think you can extract a capture with the data? If it isn't working for you I might need that

That's a fine solution too (even better I'd say). As long as it is configurable :)

This is mine:
```
/ISK5\2M550E-1012

1-3:0.2.8(50)
0-0:1.0.0(210901220240S)
0-0:96.1.1(***)
1-0:1.8.1(006919.627*kWh)
1-0:1.8.2(006368.188*kWh)
1-0:2.8.1(000001.545*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.570*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00011)
0-0:96.7.9(00003)
1-0:99.97.0(1)(0-0:96.7.19)(171027165014S)(0000070436*s)
1-0:32.32.0(00007)
1-0:32.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(227.0*V)
1-0:31.7.0(004*A)
1-0:21.7.0(00.701*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(***)
0-1:24.2.1(210901220008S)(02855.757*m3)
!5CB6
```

I've got bus id  1 for gas `0-1` -> `0-1:24.2.1(210901220008S)(02855.757*m3)`

**specification**

7.3 Gas Data
The following is only applicable if Gas meters are connected. The OBIS reference Channel number (n) will be defined by the installing order of the M-Bus devices. So a Gas meter can be installed on channel 1, 2, 3 or 4.
Device-Type `0-n:24.1.0.255`
Equipment identifier `0-n:96.1.0.255`
Last 5-minute value (temperature converted), gas delivered to client in m3, including decimal values and capture time `0-n:24.2.1.255`

bakboord4, 2021-09-02:
@glmnet. your request for a capture. Have only one slimmelezer meaning the download of the esp-link firmware with esptool etc. May I ask if the above reaction of @zuidwijk (thanks) provide you enough information?
To the suggested solution. From a user perspective it is unclear if an user can remember or is informed that the gasmeter was broke and replaced or that the non-reporting may be caused by for instance a software and or hardware problem. 
Secondly, does an user have the knowledge of gasmeter numbering 1 to 4? 
Is it possible to have a look in HA DSMR integration where to my knowledge the gasmeter automatically is found. That solution in combination with esp-link from Zuidwijk worked in my case.

bakboord4, 2021-09-02:
@glmnet @zuidwijk 
Not that easy to convert slimmelezer from ESPHome firmware to ESP-link firmware and integrated into HA-test with DSMR.
Two screenprints as requested. 

One log; at the bottom you will see gasmeter telegram. @zuidwijk's assumption is correct. 

Second screenprint is the proof of my last remark in above reaction. It shows the entity Gas Comsumption of the gasmeter device created by the DSMR integration of HA. 

[image]
 
[image]

closed 2021-09-15 by glmnet