esphome/issues #2328: Template Number state is always "unknown"

JamesSwift, 2021-08-18:
### The problem

I decided to try out the new template number as it seems a so much simpler concept to configure than how I used to do it. 

However, although in the ESPHome logs I can see the value being set successfully, Home Assistant always reports the value as `unknown`.

The value is sometimes set by the esp 8266, sometimes by calling a service from within home assistant:

```
service: number.set_value
target:
  entity_id: number.kitchen_window_hold_until
data:
  value: 10
```

[image]

### Which version of ESPHome has the issue?

v2021.8.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.8.7

### What platform are you using?

ESP8266

### Board

d1 mini

### Component causing the issue

template number

### Example YAML snippet

```yaml
number:
  - platform: template
    name: "Kitchen Window Hold Until"
    id: kitchen_window_hold_until
    min_value: 0
    max_value: 86400
    step: 1
```

### Anything in the logs that might be useful for us?

```txt
[19:35:24][D][number:010]: 'Kitchen Window Hold Until' - Setting
[19:35:24][D][number:029]:   Value: 10.000000
```

oxan, 2021-08-18:
You should either enable optimistic mode, or supply the lambda and set_action parameters to read/write the number from somewhere (if optimistic mode is off, the template number doesn't track any state itself).

@jesserockz would it make sense to require either optimistic mode to be on, or lambda/set_action to be specified (in config validation)?

JamesSwift, 2021-08-18:
Thannks @oxan, I read the docs but missed the "defaults to false" under optimistic, I assumed it would default to true. Sorry about that. 

Could I also ask, as the number is a float, what are the limits on precision? I ideally want to store a unix timestamp in there.

I noticed if I set:
```
number:
  - platform: template
    name: "Kitchen Window Hold Until"
    id: kitchen_window_hold_until
    min_value: 1111111111
    max_value: 9999999999
    optimistic: true
    step: 1
```

Home assistant reports:
[image]

oxan, 2021-08-18:
> Could I also ask, as the number is a float, what are the limits on precision? I ideally want to store a unix timestamp in there.

ESPHome uses single-precision floats. The exact limits escape me at the moment, but it's not enough for unix timestamps. You'd get steps of 30 seconds or something like that.

jesserockz, 2021-08-18:
> @jesserockz would it make sense to require either optimistic mode to be on, or lambda/set_action to be specified (in config validation)?

Well its the same as any of the other template platforms... Its up to the user to set the right fields based on use case.

oxan, 2021-08-18:
> Well its the same as any of the other template platforms... Its up to the user to set the right fields based on use case.

Yeah sure, but generally we try pretty hard to only allow configurations that will work and make sense (it's even in the contribution guide to be as strict as possible with validation). I can't think of a usecase for numbers that are non-optimistic and don't have a set_action, as it seems to me that any value set from HA will then just be lost. But maybe I'm not creative enough :) (As you said, it's the same for covers&switches, but I'm also not sure whether it makes sense there).

jesserockz, 2021-08-18:
Well for switches it can make sense to not do it as it gives you a momentary switch, but for numbers maybe it does

JamesSwift, 2021-08-18:
Thanks for your help on this. Really appreciate this awesome project and the people who maintain it

oxan, 2021-08-18:
> Well for switches it can make sense to not do it as it gives you a momentary switch, but for numbers maybe it does

For momentary switches you'd still need either `turn_on_action`/`turn_off_action`, right? Otherwise you'd have a momentary switch that does nothing when triggered, which wouldn't be particulary useful.

closed 2021-09-04 by jesserockz