esphome/issues #2329: 2021.8.0 breaks DHT SI7021

srnoth, 2021-08-18:
### The problem

After upgrading to 2021.8.0 (only changes to config are to add the new thermostat entries), the log is full off:

[18:59:37][W][dht:167]: Falling edge for bit 40 failed!
[18:59:37][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).

Returning to 1.20.4 resolved the issue immediately.

The upgrade does *not* appear to break my DHT22 sensors, just the 1-wire SI7021 (Sonoff) sensors.

### Which version of ESPHome has the issue?

2021.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Latest

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

DHT SI7021 

### Example YAML snippet

```yaml
sensor:
  - platform: dht
    pin: D6
    model: SI7021
    temperature:
      name: AC Temperature
      id: temperature
      accuracy_decimals: 1
      filters:
        - sliding_window_moving_average:
            window_size: 5
            send_every: 5
    humidity:
      name: AC Humidity
      id: humidity
      accuracy_decimals: 0
      filters:
        - sliding_window_moving_average:
            window_size: 5
            send_every: 5
    update_interval: 1s
```

### Anything in the logs that might be useful for us?

```txt
[18:59:37][W][dht:167]: Falling edge for bit 40 failed!
[18:59:37][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
```

jesserockz, 2021-08-19:
Weird as `dht` has not changed between 1.20 and 2021.8 apart from spelling in a comment...

srnoth, 2021-08-19:
Confirmed the same behavior on a brand new Sonoff TH16 with only the DHT sensor defined, nothing else.

srnoth, 2021-08-24:
Fixed in 2021.8.1

closed 2021-08-24 by srnoth