esphome/issues #2334: Waveshare ePaper 2.9" Rev2.1 - flickering screen refreshing problem.

dougle03, 2021-08-19:
### The problem

Following the example within the ESPHome.io page `https://next.esphome.io/components/display/waveshare_epaper.html`
I've paired up a Waveshare ePaper 2.92 board version 2.1 to a Wemos D1 Mini and loaded the code.
I get the basic itprint text on boot up, but on next refresh the screen shows noise, then flashes the correct display then fades to nearly nothing until the next refresh where the cycle repeats.
[image]

### Which version of ESPHome has the issue?

2021.9.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Not applicable

### What platform are you using?

ESP8266

### Board

Wemos D1 (Clone)

### Component causing the issue

Waveshare ePaper 2.9" rev2.1

### Example YAML snippet

```yaml
yaml below.

spi:
  clk_pin: D0
  mosi_pin: D1
  #miso_pin: D7 #not connected
display:
  - platform: waveshare_epaper
    cs_pin: D2
    dc_pin: D3
    busy_pin: D4
    reset_pin: D5
    model: 2.90inv2
    #id: mydisplay
    update_interval: 30s
    full_update_every: 180
    rotation: 90°
    #brightness: 100%
    lambda: |-
      //Header
      it.strftime(0, 0, id(font4), TextAlign::TOP_LEFT, "%a %H:%M", id(esptime).now());
      it.print(0, 50, id(font3), "TESTING");
      
```
```

### Anything in the logs that might be useful for us?

```txt
Nothing in the logs of note.
```

### Additional information

https://user-images.githubusercontent.com/59865048/130116613-ad30d677-3dce-4061-958c-a3fba39d5742.mp4

dougle03, 2021-08-19:
Note: powered by a lab bench power supply to the USB of the D1. Ground and 3.3v from the D1's pins.

ephirial, 2021-08-20:
Hey, I can confirm this behaviour. I have tested the same display with a Wemos D1 and I have the same issue. I have tested a bunch of different pin configurations, all with the same result.
I have also tested it on a ESP32, also with the same issue.

dougle03, 2021-08-20:
Seems like it's an actual bug then. It's a shame as I have plans for this display.

swifty99, 2021-08-28:
Hi, 
same here.

To mitigate the problem I use:
`  on_boot:
    priority: 700 
      # remove display dirt, bad workaround, shoulb be better
      - delay: 2sec      
      - component.update: my_display 
      - delay: 100ms
      - component.update: my_display 
      - delay: 100ms
      - component.update: my_display 
      - delay: 100ms
      - component.update: my_display 
      - delay: 100ms
`

dougle03, 2021-08-28:
Thanks, I'll give that a go when I get home ;)

dougle03, 2021-08-28:
Just need to get home now to try it! 
[image]

dkagedal, 2021-09-15:
I have this problem as well. I'm not sure how the code above was meant to help, other than to try to update the display "enough times". Anyway it didn't help to me.

Looking at the code, and at the specs and examples from Waveshare, I'm wondering about the code at https://github.com/esphome/esphome/blob/8f3a739da77e06b5ef18115248ee469ed0caa95f/esphome/components/waveshare_epaper/waveshare_epaper.cpp#L375

Should't there be a call to `this->wait_until_idle_()` after the Master Activation command?

And what is the `0xFF` command? I didn't find that in the specs.

dkagedal, 2021-09-16:
Tried adding that, and it didn't help. But I don't really know what I'm doing here.

dougle03, 2021-09-16:
It does look like there is a bug in the module that's presumably messing up the refresh process. People have these displays working fine using Arduino IDE and the libs available; but the ESPHome implementation does not work properly with v2.1 screens. I've tried to buy a v2 screen but can't find them anywhere on sale.
Perhaps the original dev would take a look at some point? 
Anyone know who the dev was?

balloob, 2021-09-16:
Please do not tag random people to get attention to your issue.

dkagedal, 2021-09-16:
It seems to get it right when redrawing it initially after boot, but later refreshes are corrupted. So I tried

```
    full_update_every: 1
```

And that solved the issue!

dougle03, 2021-09-16:
> Please do not tag random people to get attention to your issue.
Sorry, did not mean to tag you, was attempting to tag the original dev for the mod. Apologies again.

dougle03, 2021-09-16:
> It seems to get it right when redrawing it initially after boot, but later refreshes are corrupted. So I tried
> 
> ```
>     full_update_every: 1
> ```
> 
> And that solved the issue!

Excellent, will give that a try later this evening. Thanks

dougle03, 2021-09-16:
>  full_update_every: 1

That did it. Now it's stable. Thanks very much. I'll close this issue now, but I suspect a change is needed to the ESPHome docs page.

dkagedal, 2021-09-17:
This is really just a workaround. The problem remains that partial updates are broken, so i think we can keep this issue open

dougle03, 2021-09-17:
Yes, I agree. I only closed because it solved my immediate problem, but yes, having it flicker in the corner of my eye every 15 seconds is pretty annoying...

LukasK13, 2021-09-20:
Please reopen this issue.
The problem is not only the text being blurry but also the flickering during the partial update which should not occur. Definitely not resolved.

dougle03, 2021-09-21:
Reopened to resolve the issues

LukasK13, 2021-09-21:
@K-r-i-s-t-i-a-n you implemented esphome/esphome#1538. Did this work with v2?

Dianoga, 2021-09-21:
I'm seeing this exact same issue. Same screen hooked up to a D1 mini exhibiting the same effect. I also tried switching to a Heltec Wifi kit 32 to see if it was board specific but the same issue is still happening.

Dianoga, 2021-09-21:
Partial update is the entire reason I'm using this screen. Currently I'm looking at the differences between the code generated by esphome and the sample code provided by waveshare as the sample code is properly working with the partial refresh.

Dianoga, 2021-09-21:
I don't have it working and have in fact made everything worse (full updates don't work either) but as best as I can tell right now these changes _should_ match the behavior of the example code from waveshare.

https://github.com/Dianoga/esphome/commit/696d319d4baca2884c576eb48be57edd31d5eea8

dougle03, 2021-09-22:
My yaml that now works, although a full refresh is done every 15 seconds (Which is a bit rubbish).
`spi:
  clk_pin: D0
  mosi_pin: D1 #DIN
  #miso_pin: D7 #not connected
display:
  - platform: waveshare_epaper
    cs_pin: D2
    dc_pin: D6
    busy_pin: D7
    reset_pin: D5
    model: 2.90inv2
    id: mydisplay
    update_interval: 15s
    full_update_every: 1
    rotation: 90
    lambda: |-`

LukasK13, 2021-11-01:
I had a look at the example code [here](https://github.com/waveshare/e-Paper/blob/master/Arduino/epd2in9_V2/epd2in9_V2.cpp) and the (incomplete) manual [here](https://www.waveshare.com/w/upload/7/79/2.9inch-e-paper-v2-specification.pdf) and implemented a hacky fix for the artial update. Unfortunately, a partial update has to be done 2 s after each full update in order to make the first partial update work. Therefore I had to include a 2 s delay in the code which is rather unprofessional.
You can find my fix [here](https://github.com/LukasK13/esphome/tree/fix-waveshare-epaper-2in9v2).

Dianoga, 2021-12-12:
How do we get a fix merged into esphome? I don't have enough C knowledge to put together a PR that I would have any faith in. @LukasK13 are you able to submit a PR with your changes? Even if it's not a perfect fix it would be better then the broken we have now.

joekeslar, 2022-01-23:
Is there a status on the merged fix?

Tookir, 2022-05-16:
Hi,

Same problem here, I just bought it for a project, it annoys me a little since it is not marked that it does not work in v2.1. Anyone have a solution, other than a total refresh?

emdekext, 2022-05-21:
I agree, please fix this asap, thanks. 
There's also another open issue, it seems exactly the same problem: [https://github.com/esphome/issues/issues/2459](url)

Kaibob2, 2022-06-03:
I just bought a 2.9" Waveshare E-Ink Panel V2 and seeing the exact same behaviour as pictured in the start post. 
Reducing `full_update_every:` to 1 fixes the grainy display, but a full update is not what the display is intended for.

Kaibob2, 2022-06-23:
Meanwhile I also added the above as local `external_component` and this fixed the issue almost perfectly. The mentioned 2 seconds delay (in the original thread) in the partial update code make buttons pretty unresponsive when a lot of display updates occur while pressing them, but that's not a real problem.

Tookir, 2022-06-23:
Can you share you esp code for exemple Kaibob2 ? I try to add id by external_component github link but it's not working. (errors at compilation).
Thanks for your help.

Kaibob2, 2022-06-23:
This is what you can use:

```
external_components:
  - source:
      type: git
      url: https://github.com/velaar/esphome
      ref: dev
    components: [ waveshare_epaper]
```

When you have ESPhome 2022.6.0 or newer installed you will need `display` also additional as external component as there was a breaking change from that version onwards. 
`Add display_type property to DisplayBuffer https://github.com/esphome/esphome/pull/3430 by [@guillempages](https://github.com/guillempages) (breaking-change)`
In that case it would be like this then:

```
external_components:
  - source:
      type: git
      url: https://github.com/velaar/esphome
      ref: dev
    components: [ waveshare_epaper, display]
```

Kaibob2, 2022-06-23:
Another option would be to download the git repository and copy the `display` and `waveshare` folder into a folder of your choice in the esphome directory. In my case it's named `my_components`.
All components in this directory will then be used instead of the default ones during compilation. This method makes the source code a little more independent from changes in a repository of someone else.
If you decide to do so it will be like that:

```
external_components:
  - source:
      type: local
      path: /config/esphome/my_components
```

Kaibob2, 2022-06-23:
Glad I could help. I'm very thankful for this fix, too.
Most of the info on my display is not time critical either, the few info which is I perform a manual update with `component.update` as soon as a new value arrives. My standard values are:

```
reset_duration: 200ms
update_interval: 30s
full_update_every: 200
```

remyderuysscher, 2023-01-02:
> This is what you can use:
>
> ```
> external_components:
>   - source:
>       type: git
>       url: https://github.com/velaar/esphome
>       ref: dev
>     components: [ waveshare_epaper]
> ```
>
> When you have [...]

Excellent, spend many hours to debug this screen.  This solves it.  Should be merged in ESPHome someday.

ExZylan, 2023-06-22:
> Excellent, spend many hours to debug this screen. This solves it. Should be merged in ESPHome someday.

With the latest version of esphome 2023.6.0, this external package no longer works. Any solution?

Compiling the program without the external package still produces errors in the fast screen refresh of my Waveshare 2.9".

I had to revert to an earlier version of esphome to be able to compile the program with the external package.

Kaibob2, 2023-06-29:
The fix is pretty simple if you finally find the breaking change related to this issue: https://github.com/esphome/esphome/pull/4926
So, if you're already using the external component from https://github.com/velaar/esphome and have `waveshare_epaper` and  `display` integrated like this:
```
external_components:
  - source:
      type: local
      path: /config/esphome/my_components
```
then just navigate with a file editor to your my_components folder and change 
`time::ESPTime time`
to
`ESPTime time`
in line 371, 377, 380 and 383 in file `display_buffer.cpp`

Afterwards do the same in line 241, 253, 265 and 276 in file `display_buffer.h`

Save both files, click on clean build files in the ESPhome Dashboard for the specific node which won't compile and install like usual.

beyondgrave, 2023-06-29:
This almost worked for me. I can now get past the time::ESPTime time error, but now get an error about set_transparency. Are you using any images on your epaper display??

Kaibob2, 2023-06-29:
No, only Icons, lines and Text . Sorry.

beyondgrave, 2023-06-29:
Weird. I can't even get icons to work... I guess I'll keep fighting with it

Kaibob2, 2023-06-30:
For the icons i added this 

```
font:
  - file: 'materialdesignicons-webfont.ttf'
    id: icon_font_24
    size: 24
    glyphs: [
      "\U000F1974",  #car-clock
      "\U000F0298",  #gas-station
      "\U000F1077",  #face-woman
      "\U000F0643",  #face-man
      "\U000F050F",  #thermometer
      "\U000F058E",  #water-percent
      "\U000F11DC",  #window-open-variant
      "\U000F06E8",  #lightbulb-on
      "\U000F01AD",  #currency-eur
      "\U000F0D8E",  #mailbox-up-outline
      "\U000F1886"   #mdi-flower-pollen-outline  
      ]
```
and then use the icons like this:
```
    pages:
      - id: mainscreen
        lambda: |-
          if (id(online).state){

            //Temperatur Garten
            it.printf(2, 36, id(icon_font_24), "\U000F050F");          
            if (id(t_garten).has_state())
            {              
              it.printf(24, 41, id(arial18), "%.1f°", id(t_garten).state);
            }
          }
```
Only for example, the whole mainscreen code is huuuge.

beyondgrave, 2023-06-30:
Ah, I see. I was using still using the Image function in ESPhome and referring to the mdi-icons. Seems like the custom components don't have support for the new/changed Image features (transparency stuff). Using the Font function as a workaround is very clever. I'll have to try that, but I'll have to do a lot of re-coding

Kaibob2, 2023-06-30:
This whole custom component solution is quite flaky and I guess it will be a constant struggle to keep it working with ongoing changes in the main ESPhome features. I hope that there will be a better supported e-ink display soon. Maybe there is already one available and I just don't know.
Good luck and happy re-coding :)

stephenjamieson, 2023-07-06:
@jesserockz, apologies for the direct ping, but is there any chance that this will get merged? Or does it need work?

velaar, 2023-07-06:
> time::ESPTime time

Fixed this in my github version. This manual fix is no longer needed.

Kaibob2, 2023-07-21:
As there are quite a few `display` related breaking changes in 2023.7.0, has anybody using the waveshare ePaper already updated to this version and can confirm that it's still working?

phoenixswiss, 2023-07-21:
@Kaibob2  It is broken. I have two 7.5v2 waveshare screens configured with model: 7.50inV2

One was working fine on 2023.6.5 and the other was having problems with artifacts.

After updating to 2023.7.0, the one with artefacts doesn't work with model: 7.50inV2 --> error message: "component api took a long time for an operation". I changed it to 7.50inV2alt and it works with that configuration, but has even more artifacts than before.

The other, which worked fine with 2023.6.5, now has artifacts with model: 7.50inV2 and with model: 7.50inV2alt doesn't even work. I also now have two vertical lines on the screen (a new hw problem). 

In other words: I had and still have a lot of problems with the ePaper display integration in ESPhome in general. I have flashed my old Arduino project to my Lolin D32 Pros and both displays work fine.

Kaibob2, 2023-07-21:
Thanks for the response. Have you used the external component solution mentioned earlier in this thread or did you use the Standard Esphome Display component?

phoenixswiss, 2023-07-21:
I'm using the stock ESPhome displac component

```
display:
  - platform: waveshare_epaper
    id: epaper_display
    reset_duration: 200ms
    setup_priority: 1000
    cs_pin: 5
    dc_pin: 0
    busy_pin: 15
    reset_pin: 2
    model: 7.50inV2
    update_interval: never 

[...]

- component.update: epaper_display
```

I don't want to accuse anyone here, especially not the hard working ESPhome developers/community, but as a matter of fact I bought three brand new waveshare 7.5v2 ePaper screens and connected them to the same Lolin D32 Pro ESP32 just like the ones I have been running for two years. 

Since using the ESPhome Display component in my new project, all three brand new displays have some sort of temporary problem (artefacts) or even permanent damage (2 vertical lines on one screen and a completely toasted driver hat after running for a few days on another display). It may just be an unfortunate coincidence.

Kaibob2, 2023-07-21:
I see. Funny that it works like that for you. I suppose you circumvent the partial Update Problem by setting update_interval: never
Maybe try the custom component solution and see if it helps.

Kaibob2, 2023-07-21:
This issue https://github.com/esphome/issues/issues/4307 also deals with this display.

phoenixswiss, 2023-07-21:
I've just noticed something strange. 

If you use a "strong" light source like an LED from a smartphone and point it directly at the display after the display has refreshed ~3 seconds ago, it goes dark at this point. 

But it shouldn't, because the update of the epaper has already been completed and the pixels shouldn't react to light after this step. This only happens when I'm running ESPhome on my ESP32. When I flash my Arduino project on the ESP, the display remains white where I have pointed the LED light source. It seems that ESPhome doesn't put the display to sleep after updating. 

Waveshare warns of display damage if the display is left on for an extended period of time.

https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT_(B)_Manual#ESP32.2F8266:~:text=Note%20that%20the%20screen%20cannot%20be%20powered%20on%20for%20a%20long%20time.%20When%20the%20screen%20is%20not%20refreshed%2C%20please%20set%20the%20screen%20to%20sleep%20mode%20or%20power%20off%20it.%20Otherwise%2C%20the%20screen%20will%20remain%20in%20a%20high%20voltage%20state%20for%20a%20long%20time%2C%20which%20will%20damage%20the%20e%2DPaper%20and%20cannot%20be%20repaired!

velaar, 2023-07-22:
> As there are quite a few `display` related breaking changes in 2023.7.0, has anybody using the waveshare ePaper already updated to this version and can confirm that it's still working?

So here goes my Friday evening. But it seems to be compiling and working on 2023.7.0.

I have done absolutely zero testing on it and switch to IDF for device-reasons (see no reason why it shouldn't work with default framework though) 

Here is a a yaml to test 

```
esphome:
  name: crystalball
  on_boot:
    priority: 200
    then: 
    - delay: 1sec
    - component.update: my_display

esp32:
  board: esp32dev
  framework:
     type: esp-idf
     version: recommended
     # Custom sdkconfig options
     sdkconfig_options:
       CONFIG_COMPILER_OPTIMIZATION_SIZE: y
     # Advanced tweaking options
     advanced:
       ignore_efuse_mac_crc: false

# Enable logging
logger:
  level: INFO

api:

external_components:
  - source:
      type: git
      url: https://github.com/velaar/esphome
      ref: test_23_7
    components: [ waveshare_epaper, display]

```

Kaibob2, 2023-07-23:
I'm on vacation for the next 10 days and I suppose it's wiser to test this afterwards :)
Thanks for the effort, I'll report back as soon as I update.

MadJalapeno, 2023-07-24:
https://github.com/esphome/issues/issues/2334#issuecomment-1646498120 fixed it for me, thank you @velaar

zigomatichub, 2023-07-27:
I don't have issue after update with model to 7.50in-bV2.
Sorry for my mistake

Kaibob2, 2023-08-06:
https://github.com/esphome/issues/issues/2334#issuecomment-1646498120 fixed it for me, too.
Thanks for that!

Nik19792000, 2023-10-01:
https://github.com/esphome/issues/issues/2334#issuecomment-1646498120 fixed it for me too. I've spent few hours before I finally found this issue discussion.

boyfromgermany, 2024-01-21:
Hello 
i have the 2.9in version with red color support
if i use 2.9in-b it does nothing at all
and if i use 2.9inV2 it shows the text for 1 second but then flickers until it is black 

i used the fix above and the custom component without success

does someone have the same display ?

pablogventura, 2024-01-28:
> Hello i have the 2.9in version with red color support if i use 2.9in-b it does nothing at all and if i use 2.9inV2 it shows the text for 1 second but then flickers until it is black
> 
> i used the fix above and the custom component without success
> 
> does someone have the same display ?

I am in the same situation, i cant be able to make work this e-ink display. I only see blinking noise. I tried all ideas in this topic.
[image]

[12 more replies]