esphome/issues #2336: ESPHome is not showing my device online

engrfaiz, 2021-08-20:
### The problem

I am using home assistant with esp32. i have write custom code for sensors and write yml accordingly. after uploading code to esp32. i cannot see online status on esphome. can you please guide?

### Which version of ESPHome has the issue?

2021.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8.0

### What platform are you using?

ESP32

### Board

ESP32 DEV Kit

### Component causing the issue

Testing

### Example YAML snippet

```yaml
includes:
    - my_custom_sensor.h  

manual_ip:
    # Set this to the IP of the ESP
    static_ip: 172.xxxxxxxx
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.2.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

sensor:
- platform: custom
  lambda: |-
    auto my_sensor = new MyCustomSensor();
    App.register_component(my_sensor);
    return {my_sensor->temperature_sensor, my_sensor->pressure_sensor};

  sensors:
  - name: "My Custom Temperature Sensor"
    unit_of_measurement: °C
    accuracy_decimals: 1
  - name: "My Custom Pressure Sensor"
    unit_of_measurement: hPa
    accuracy_decimals: 2
```

### Anything in the logs that might be useful for us?

```txt
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 172.20.10.5: [Errno 101] Network is unreachable). Re-Trying in 1 seconds
INFO Connecting to 172.20.10.5:6053 (172.20.10.5)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 172.20.10.5: [Errno 101] Network is unreachable). Re-Trying in 1 seconds
INFO Connecting to 172.20.10.5:6053 (172.20.10.5)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 172.20.10.5: timed out). Re-Trying in 2 seconds

These are logs on ESP Flasher tool

[10:48:40][I][app:106]: ESPHome version 2021.8.0 compiled on Aug 19 2021, 04:40:17
[10:48:40][C][wifi:499]: WiFi:
[10:48:40][C][wifi:359]:   SSID: 'iPhone'
[10:48:40][C][wifi:360]:   IP Address: 172xxxxxxx
[10:48:40][C][wifi:362]:   BSSID: F2:C4:2A:03:35:7A
[10:48:40][C][wifi:363]:   Hostname: 'testing'
[10:48:40][C][wifi:367]:   Signal strength: -54 dB ▂▄▆█
[10:48:40][C][wifi:371]:   Channel: 6
[10:48:40][C][wifi:372]:   Subnet: 255.255.255.0
[10:48:40][C][wifi:373]:   Gateway: 192.168.2.1
[10:48:40][C][wifi:374]:   DNS1: 0.0.0.0
[10:48:40][C][wifi:375]:   DNS2: 0.0.0.0
[10:48:40][C][logger:189]: Logger:
[10:48:41][C][logger:190]:   Level: DEBUG
[10:48:41][C][logger:191]:   Log Baud Rate: 115200
[10:48:41][C][logger:192]:   Hardware UART: UART0
[10:48:41][C][captive_portal:148]: Captive Portal:
[10:48:41][C][ota:029]: Over-The-Air Updates:
[10:48:41][C][ota:030]:   Address: 172.20.10.5:3232
[10:48:41][C][ota:032]:   Using Password.
[10:48:41][C][api:095]: API Server:
[10:48:41][C][api:096]:   Address: 172.20.10.5:6053
[10:48:51][D][sensor:131]: 'My Custom Temperature Sensor': Sending state 7.12000 °C with 1 decimals of accuracy
[10:48:51][D][sensor:131]: 'My Custom Pressure Sensor': Sending state 9.12000 hPa with 2 decimals of accuracy
```

### Additional information

This is my_custom_sensor.h code

#include "esphome.h"

class MyCustomSensor : public PollingComponent 
{
    public:
      Sensor *temperature_sensor = new Sensor();
      Sensor *pressure_sensor = new Sensor();
      MyCustomSensor() : PollingComponent(15000) { }

      void setup() override {  }

      void update() override 
      {
        // This is the actual sensor reading logic.
    
        temperature_sensor->publish_state(7.12);

    
        pressure_sensor->publish_state(9.12);
      }
};

DAVe3283, 2021-08-20:
IP address of the device is `172.20.10.5/24`, but the gateway is `192.168.2.1`, which is not on the subnet. The gateway needs to be on the subnet for the ESP to be able to talk outside that subnet.

Additionally, the logs indicate your ESPHome instance doesn't have a valid route to the `172.20.10.x` subnet, and can't even send packets to the ESP (which wouldn't be able to send any back due to its bad gateway config).

You need to validate your network setup. This doesn't look like a ESPHome problem, but a network config problem.

engrfaiz, 2021-08-23:
Thanks for your response. actually i am not good in network related things. but here is new ip and Gateway. still i cannot see esp32 online on home assistant. can you please tell me what values should i change.
Thank You

[10:03:50][I][app:106]: ESPHome version 2021.8.0 compiled on Aug 22 2021, 18:02:31
[10:03:50][C][wifi:499]: WiFi:
[10:03:50][C][wifi:359]:   SSID: 'iptime'
[10:03:50][C][wifi:360]:   IP Address: 192.168.0.190
[10:03:50][C][wifi:362]:   BSSID: 70:5D:CC:7E:4D:B6
[10:03:50][C][wifi:363]:   Hostname: 'testing'
[10:03:50][C][wifi:367]:   Signal strength: -53 dB ▂▄▆█
[10:03:50][C][wifi:371]:   Channel: 1
[10:03:50][C][wifi:372]:   Subnet: 255.255.255.0
[10:03:50][C][wifi:373]:   Gateway: 192.168.0.1
[10:03:50][C][wifi:374]:   DNS1: 168.188.1.1
[10:03:50][C][wifi:375]:   DNS2: 168.126.63.1
[10:03:50][C][logger:189]: Logger:

DAVe3283, 2021-08-23:
If you are not good with networking, you might consider simplifying your setup. The default of using DHCP with mDNS works really well, and would probably solve your problem. Make sure to connect it to the same WiFi as your Home Assistant is connected to; don't use a guest WiFi or anything like that. Then you don't need to know anything about static IP addresses.

closed 2022-01-05 by github-actions[bot]