esphome/issues #2338: BH1750 illuminance sensor is not available in statistics-graph card

paulius2k, 2021-08-20:
### The problem

Hello,

I have an BH1750 illuminance device with ESPHome FW.

The problem is that this sensor is not available in the Statistics Graph Card (https://www.home-assistant.io/lovelace/statistics-graph/)

I checked the requirements:  https://www.home-assistant.io/more-info/statistics/
and to me it seams that the needed attributes of the sensor are created, i.e. this is what I see in the HA developer tools:
```
state_class: measurement
unit_of_measurement: lx
friendly_name: BH1750 Illuminance
device_class: illuminance
```

What else is needed for this sensor to become available in Statistics Graph Card?

Thank you!

### Which version of ESPHome has the issue?

2021.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8.8

### What platform are you using?

ESP8266

### Board

d1_mini_pro

### Component causing the issue

bh1750 light sensor

### Example YAML snippet

```yaml
`/config/esphome/bh1750.yaml`:

esphome:
  name: bh1750
  platform: ESP8266
  board: d1_mini_pro

sensor:
  - platform: bh1750
    name: "BH1750 Illuminance"
    device_class: "illuminance"
    state_class: "measurement"
    address: 0x23
    measurement_duration: 69
    update_interval: 30s
    force_update: true
```
```

oxan, 2021-08-25:
Those attributes seem correct to me, so I think this might be more of a HA issue than an ESPHome issue.

paulius2k, 2021-09-04:
As discussed in https://github.com/home-assistant/frontend/issues/9882, this feature was restricted only to several types of sensors: https://rc.home-assistant.io/blog/2021/08/25/release-20219/#long-term-statistics-unlocked-for-all-sensors

It works fine in 2021.9.2

closed 2021-09-04 by paulius2k