esphome/issues #2339: WA2812 pulse to 100% brightness when brightness slider is changed.

matphillips, 2021-08-20:
### The problem

I have a few LED strips being controlled by ESPHome and HomeAssistant.
This is tested and confirmed with LED strips connected to both ESP32 and ESP8266, and with the NeoPixel platform, and FastLED.

Changing the brightness setting, either programatically or using the HomeAssistant interface causes a brief pulse to 100%, so if the brightness is set to 30% and you move it to 31%, it goes up to 31 via 100. 

This only happens when there is an 'effect' in play. Any effect, even something simple like:
```
- addressable_rainbow:
          name: "Rainbow"
          speed: 10
          width: 70

```
With Effects turned off, the brightness changes smoothly and without a flash.
Easy to test by using a mouse to pick a brightness, and then the left & right arrow keys on the keyboard to make small changes to the brightness, the pulse is easy to see when the brightness barely changes.

I imagine many people have LED strips hooked up, can anyone else test and validate this issue?

### Which version of ESPHome has the issue?

2021.8.8

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.8.0

### What platform are you using?

ESP8266

oxan, 2021-08-21:
I'm surprised this used to work at all, transitions and effects don't play nice together.

matphillips, 2021-08-21:
Thanks! Yeah, it seemed to work just fine previously. Initially I thought it was a small hardware failure on my device when I first noticed it, and I dont know when it started.
In my implementation, I have a custom clock display running as an effect, and it has a light sensor to set the brightness for the ambient level of the room. So the flashing was really noticeable because it will change its own brightness quite a lot when the ambient level is hovering just above and below a threshold point. So this will help a lot.

Appreciate you finding the cause and resolving it.

closed 2021-08-23 by jesserockz

matphillips, 2021-09-06:
Fault present again in version 2021.9.3
Did the fix get removed?