esphome/issues #2340: Belgian Sagemcom T211 invalid string length error

PietPetrus, 2021-08-21:
### The problem

Bought a "Slimmelezer", opened the ports on the digital meter, plugged in and did a network setup.
Home Assistant discovers the device but the slimmelezer spits out the following error:

`           ^
Invalid string length
[E][dsmr:138]: 0-0:96.1.4(50215)
`

### Which version of ESPHome has the issue?

Firmware: 2021.8.0 (Aug 19 2021, 16:39:05)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8.8

### What platform are you using?

ESP8266

### Board

Wemos D1 mini - SlimmeLezer by M. Zuidwijk

### Component causing the issue

DSMR

butsjoh, 2021-08-23:
I am experiencing the same issue ...

siskosgithub, 2021-09-07:
I'm also experiencing the same issue, sadly cannot use the slimmelezer from Marcel because of this. Any updates on when this will be fixed? Or is there any workarround?

Sharnobile, 2021-09-07:
> I'm also experiencing the same issue, sadly cannot use the slimmelezer from Marcel because of this. Any updates on when this will be fixed? Or is there any workarround?

Same here. If you have any updates... thanks!

vandenbogerd, 2021-09-07:
Same here… no use for the Slimme Lezer here also…

gijbelsy, 2021-09-10:
Same problem.

`Invalid string length
[E][dsmr:138]: 0-0:96.1.4(50215)
           ^`

PietPetrus, 2021-09-11:
Temporary solution until this gets merged into the new ESPHome release.

Edit the device yaml like this:

```
esphome:
  platformio_options:
    lib_deps: glmnet/Dsmr@0.5
    lib_ignore: glmnet/Dsmr@0.3
```

Build firmware and do an OTA update.

Zebber, 2021-09-11:
> Temporary solution until this gets merged into the new ESPHome release.
> 
> Edit the device yaml like this (add the lines concerning platformio_options):
> 
> ```
> esphome:
>   platformio_options:
>     lib_deps: glmnet/Dsmr@0.5
>     lib_ignore: glmnet/Dsmr@0.3
> ```
> 
> Build firmware and do an OTA update.
> 
> This essentially replaces the old library with the new one, fixing the problem.

I have tried this via the home assistant add on, and the error is still there, unfortunately. Did I maybe miss any steps?

I opened ESPHome in Home Assistant, clicked 'edit' on the slimme lezer, added the platform_io_options bit to the esphome block, and finally clicked install and chose the wireless option. As far as I can tell, it did use the new library.

Am I missing a step here or did I go about it the completely wrong way?

butsjoh, 2021-09-11:
> Temporary solution until this gets merged into the new ESPHome release.
>
> Edit the device yaml like this (add the lines concerning platformio_options):
>
> ```
> esphome:
>   platformio_options: [...]

Thnx for giving instructions on how to get this working but building firmware is nothing something you can expect all users todo easily. I can figure it out probably but some more detailed instructions would be handy. I suggest a wiki page or something that explains the step. That would already take a way some frustration that is building up.

Zebber, 2021-09-11:
> I have tried this via the home assistant add on, and the error is still there, unfortunately. Did I maybe miss any steps?
>
> I opened ESPHome in Home Assistant, clicked 'edit' on the slimme lezer, [...]

Ok, very weird, but repeating this steps seems to have worked. (I repeated the steps because I was adding a static ip address to the device).

@butsjoh If you're using home assistant like I am, you can probably use the same steps as I did. Unfortunately, that's as far as my knowledge on esphome currently reaches.

butsjoh, 2021-09-11:
@Zebber So the problem is actually just the web interface that is build into the firmware that is just displaying the value wrong ? So home assistant still receives the values but the yaml fix will parse it correctly? Sorry for the dumb question maybe but my assumption was that the "slimme lezer" was not getting any data at all.

Zebber, 2021-09-11:
The device is not able to parse the data correctly. I.e. it "reads" some data, but thinks it's gibberish.

The ESPHome addon for home assistant provides an easy way to prepare new firmware and uploading that to the device.  I would suggest trying that. Here is a resource that might prove helpful (it was to me!): https://www.zuidwijk.com/category/faq/slimmelezer/

gijbelsy, 2021-09-11:
I can confirm the work-around is performing well:
[https://community.home-assistant.io/t/slimmelezer-invalid-string-length/337527](url)

PietPetrus, 2021-09-11:
@butsjoh  I'll see if I can upload a clean firmware binary to help you out. Without the credentials etc.
I expected this from slimmelezer seller but he's busy selling I guess.

But by no means am I an active developer working on ESPHome. I already did the pull request and the active dev merged it onto platformio.
Besides waiting for a merge in esphome and a new release there is not much you can do.

zuidwijk, 2021-09-11:
> @butsjoh I'll see if I can upload a clean firmware binary to help you out. Without the credentials etc.
> I expected this from slimmelezer seller but he's busy selling I guess.
>
> But by no means am I [...]

Hey, I've looked into it and seen the workaround, nice job! I didn't knew that workaround. Yeah I'm very very busy soldering all the devices and trying to get the backorders back to 0.

I'll try to compile the new code and if it works, I'll post it here and on my website :)

zuidwijk, 2021-09-11:
Can you please test below firmware?
[slimmelezer-2021.8.2-state_class-and-BE-fix.bin.zip](https://github.com/esphome/issues/files/7148345/slimmelezer-2021.8.2-state_class-and-BE-fix.bin.zip)

Cpainy, 2021-09-11:
Can confirm that that firmware works Marcel. Thanks @PietPetrus, @zuidwijk and others

PietPetrus, 2021-09-12:
@zuidwijk Thanks for replying!
I understand the issue with the backorders, that's why I didn't bother you with software issue at first 😁.

If you encounter more problems (on a Belgian meter) you may let me know and I'll look into it if I have time.

butsjoh, 2021-09-12:
@zuidwijk Fully understand the pressure on you related to the orders that have been made. I am trying the patch right know but feel confident that it is going to work. Apart from the workaround which seems to be a valid route to take i was asking if you could explain in a wiki page how to build the patch that you just made. I guess these steps that you do are a routine job and then we people will have to bother you less if they want to take a shortcut and move faster. But thnx for the support ! Much appreciated !

siskosgithub, 2021-09-12:
> Temporary solution until this gets merged into the new ESPHome release.
>
> Edit the device yaml like this (add the lines concerning platformio_options):
>
> ```
> esphome:
>   platformio_options: [...]

Nice, thanks for the update. Didn't know this was possible :) 
Can confirm it is now working here 👍

butsjoh, 2021-09-12:
> Can you please test below firmware?
> [slimmelezer-2021.8.2-state_class-and-BE-fix.bin.zip](https://github.com/esphome/issues/files/7148345/slimmelezer-2021.8.2-state_class-and-BE-fix.bin.zip)

This works perfect ! Thnx

domoticz410, 2021-09-12:
> > Can you please test below firmware?
> > [slimmelezer-2021.8.2-state_class-and-BE-fix.bin.zip](https://github.com/esphome/issues/files/7148345/slimmelezer-2021.8.2-state_class-and-BE-fix.bin.zip)
> 
> This works perfect ! Thnx

@zuidwijk 
I've successfully updated the firmware (OTA - update with  the bin file from the above ZIP), but now I do not see the SlimmeLezer  asking for an IP-address anymore.  Before the update, Slimmelezer got 192.168.226 from my PfSense and after the OTA update no more activity in this slimmelezer.  It is also not showing the  Wifi-"Slimmelezer" entry so I can start all over...  how to get out of this?

zuidwijk, 2021-09-12:
> > > Can you please test below firmware?
> 
> > > [slimmelezer-2021.8.2-state_class-and-BE-fix.bin.zip](https://github.com/esphome/issues/files/7148345/slimmelezer-2021.8.2-state_class-and-BE-fix.bin.zip)
> 
> > 
> 
> > This works perfect ! Thnx
> 
> 
> 
> @zuidwijk 
> 
> I've successfully updated the firmware (OTA - update with  the bin file from the above ZIP), but now I do not see the SlimmeLezer  asking for an IP-address anymore.  Before the update, Slimmelezer got 192.168.2.226 from my PfSense and after the OTA update no more activity in this slimmelezer.  It is also not showing the  Wifi-"Slimmelezer" entry so I can start all over...  how to get out of this?

After update you need to connect it to your WiFi again (your WiFi credentials isn't in the firmware 😉 )

domoticz410, 2021-09-12:
> After update you need to connect it to your WiFi again (your WiFi credentials isn't in the firmware 😉 )

This time it worked ... getting in the values and I see no more error.
BUT....   maybe for other Home Assistant users like me, who did not yet play with ESP-devices (and flashing them) and who are using Windows (e.g. Win10) , it would be great to store a document on your website to instruct us to flash a new firmware to the SlimmeLezer.
Here are some of my experiences:
1e.   Using OTA update did NOT work
2e.   So need to install via USB-cable to Windows-laptop   (I got COM3 as port)
3e.  That meant for me:   
  a.  Open Windows Powershell
  b.  Install latest Python (python.org) for Windows  --- pleanty of youtube's on this
  c.  Install ESPTOOL via " python -m pip install esptool"  because pip install esptool gave error  --- also youtube's on this
4e.  Now that ESPTOOL was installed, it was not that easy to launch it and select the firmware bin-file, so I ended up downloading   https://github.com/Nerdiyde/ESPEasyFlasher
 -> that gives a very easy web-window with 2 entries:   1) port (and if your USB is plugged in, you see port selected)
     2)   a selector to firmware bin-file    Warning:   you need to copy the firmware file under the same folder as the ESPEasyFlasher folder you unzipped the files in -- then you can select it
5e.  AND only now, finally, pushing flash button gives a perfectly working flashing of new firmware in SlimmeLezer.
6e.   Important:  also telling people that you need to start from scratch again, meaning:
      a.  look for SlimmeLezer Wifi on your laptop
      b.  logon to 192.168.4.1 until you get a menu with Wifi SSIDs to select from in your network
      c.   and slect again your Wifi-SSid to use + password

Would be great if all would be nicely on 1 page........it did cost ne quite some time to hunt down all this info on different pages and youtube movies......

Anyhow...all seems to be working ... finally I can now move to Home Assistant to get the SlimmeLezer recognized there 

Thanks ;)

closed 2021-09-15 by glmnet

DirkTas67, 2021-09-19:
When will this be fixed ? I really like esphome because you can change things on the fly, many things have been changed in my YAML file and so if I flash this BIN file, everything is gone, this is not ideal.

So any idea when this will be fixed in ESPHOME ?

Regards
Dirk

siskosgithub, 2021-09-20:
@DirkTas67 they have already fixed it in the dev channel's i can see in the other topics.
For now I have added the following to my code, this way you can still use your own code.
`esphome:
  platformio_options:
    lib_deps: glmnet/Dsmr@0.5
    lib_ignore: glmnet/Dsmr@0.3`