esphome/issues #2341: ESP32 not waking up from deep sleep

hirani89, 2021-08-21:
### The problem

I have a ESP32­WROOM­32UE connected to rs485. But it does not wake up from deep sleep.

I am using the rx and tx pins to communicate to rs485.

I also have a voltage divider connected to IO35 to read battery voltage.

other than that, nothing is connected to the esp.

### Which version of ESPHome has the issue?

1.20.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.7.4

### What platform are you using?

ESP32

### Board

ESP32­WROOM­32UE

### Component causing the issue

deep_sleep

### Example YAML snippet

```yaml
external_components:
  # use  GitHub
  - source:
      type: git
      url: https://github.com/martgras/esphome
      ref: modbus_component
    components: [ modbus_controller ]
esphome:
  name: test
  platform: ESP32
  board: esp32dev

# Enable logging
# logger:

# Enable Home Assistant API
api:

ota:
  password: "xxxxx"

wifi:
  ssid: "XXXXX"
  password: "xxxxx"
  manual_ip:
    static_ip: 192.168.0.189
    gateway: 192.168.0.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "ryduSVVUvTTU"
[... 199 lines cut ...]
[21:41:47][C][ota:032]:   Using Password.
[21:41:47][C][api:095]: API Server:
[21:41:47][C][api:096]:   Address: 192.168.0.189:6053
[21:41:47][C][deep_sleep:020]: Setting up Deep Sleep...
[21:41:47][C][deep_sleep:023]:   Sleep Duration: 1800000 ms
[21:41:47][C][deep_sleep:026]:   Run Duration: 30000 ms
[21:41:48][D][adc:085]: 'Sensor Battery': Got voltage=2.15V
[21:41:48][D][sensor:127]: 'Sensor Battery': Sending state 3.94787 V with 2 decimals of accuracy
[21:41:58][D][adc:085]: 'Sensor Battery': Got voltage=2.15V
[21:42:00][D][sensor:127]: 'Sensor Battery': Sending state 3.95311 V with 2 decimals of accuracy
[21:42:08][D][adc:085]: 'Sensor Battery': Got voltage=2.14V
[21:42:08][D][sensor:127]: 'Sensor Battery': Sending state 3.93389 V with 2 decimals of accuracy
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.0.189:6053 (192.168.0.189)
WARNING Couldn't connect to API (Error connecting to 192.168.0.189: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.0.189:6053 (192.168.0.189)
WARNING Couldn't connect to API (Error connecting to 192.168.0.189: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.0.189:6053 (192.168.0.189)
WARNING Couldn't connect to API (Error connecting to 192.168.0.189: timed out). Trying to reconnect in 2 seconds
INFO Connecting to 192.168.0.189:6053 (192.168.0.189)
WARNING Couldn't connect to API (Error connecting to 192.168.0.189: timed out). Trying to reconnect in 3 seconds
INFO Connecting to 192.168.0.189:6053 (192.168.0.189)
WARNING Couldn't connect to API (Error connecting to 192.168.0.189: timed out). Trying to reconnect in 5 seconds
INFO Connecting to 192.168.0.189:6053 (192.168.0.189)
```

insipiens, 2021-09-21:
Have you connected GPIO0 and RST so it can wake up?

martgras, 2021-10-06:
An Esp32 doesn't need gpio pins for wakeup.
Are you sure it doesn't wake up after 30 minutes? 
For testing I would reduce the the sleep time. 
Btw I realize this issue is a bit old already but did you notice that the modbus component changed quite a bit since then. See https://github.com/martgras/esphome/blob/testing/esphome/components/modbus_controller/readme.md

devjklein, 2024-05-21:
@hirani89 Were you ever able to resolve this issue? I am using an ESP32-WROOM-32 and getting the same issue. My log states `Sleep duration: 60000000us` but it never wakes up. This occurs with nothing connected to the board.