esphome/issues #2343: ESP32 S2 WROOM - not able to integrate

homonto, 2021-08-21:
### The problem

so I am happy user of ESPHome with some ESP32S boards. Recently I bought ESP32-S2-WROOM and I am completely not able to integrate this board with ESPHome. Is it because S2 is not supported or what?
In Arduino or other platforms there is a big difference between ESP32 and S2 but as per ESPHome: "All ESP32-based devices are supported by ESPHome. Simply select ESP32 when the ESPHome wizard asks you ..." - but when I start the wizard it says: "Unable to identify the connected device (12882)." When I skip wizard and try to install it says: "installation failed". I am connecting using usb-c cable from my macbook to the board. Arduino has no issues - any sketch done with Arduino or even esptool works but ESPHome apparently not. I am a big fan of ESPHome for my HA but...
Any help would be appreciated. Thank you

### Which version of ESPHome has the issue?

Current version: 2021.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.8.8

### What platform are you using?

ESP32

### Board

ESP32-S2-WROOM

### Component causing the issue

integration does not work

### Example YAML snippet

```yaml
esphome:
  name: test
  platform: ESP32
  board: esp32dev
  arduino_version: latest

# Enable logging
logger:
  level: DEBUG
# Enable Home Assistant API
api:

ota:
  password: "asdfasdfas"
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: True
  manual_ip:
    static_ip: 192.168.1.184
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "test Fallback Hotspot"
    password: !secret esp_fallback_wifi_password

captive_portal:

sensor:
  - platform: wifi_signal
    update_interval: 5s
    id: test_wifi
    name: "test_wifi"
```

### Anything in the logs that might be useful for us?

```txt
I am not at the log level as it does not integrate even
During installation it shows:

"installation failed" and 2 eyes ;-(
```

oxan, 2021-08-21:
That text on the website is outdated, there's no official support for the S2 (and C3) yet. However, you can try setting `arduino_version` to `dev`, it might work (but be warned that that's using an dev version of Arduino, so it might break at any time).

homonto, 2021-08-21:
thank you for your answer
I am using Arduino IDE and this ESP32-S2 works perfectly OK there
but ESPHome is... much easier for IoT than Arduino

I tried but unsuccessful - it does not install - same like before

oxan, 2021-08-22:
Well, it might work, but there's no stable release of the Arduino framework for the S2 yet. 

If it doesn't work, you probably need to set the correct board. I'm not sure if there's any S2 boards in the ESPHome database yet.

homonto, 2021-08-22:
yeap, but Arduino itself has completely NO ISSUE with S2 and in Arduino I set simply "ESP32S2 DEV MODULE" - which one to put into ESPHome in the field "board" ?

oxan, 2021-08-22:
See the [documentation](https://esphome.io/components/esphome.html) for boards option, you need to put a PlatformIO board identifier there.

homonto, 2021-08-22:
thank you but since I am not an expert it does not help me at all ;-(

oxan, 2021-08-22:
I missed that you included the board you're using in the initial issue, it seems the ESP32-S2-WROOM isn't supported by PlatformIO yet, so it can't be used with ESPHome. 

I'd suggest to wait until these chips are officially supported.

homonto, 2021-08-22:
thank you, considering that S2 is almost 2y old... ;-) OK, I am not complaining - just moving back to Arduino IDE then. Pity, because what in Arduino takes hundreds of lines, is done in 20 lines in ESPHome for my non-S2 ESP32 boards ;-(
but ok, life

teamsuperpanda, 2021-12-06:
> This works for my S2. You can try that, too.
> 
> ```
> esp32:
>   board: esp32-s2-kaluga-1
> 
> esphome:
>   platformio_options:
>     platform_packages:
>       - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0
>     platform:
>       - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
> ```

Using that I got this error. Do you know what changes I would need to make?
```
src/esphome/components/esp32_ble_tracker/queue.h:11:10: fatal error: esp_gap_ble_api.h: No such file or directory

*************************************************************************
* Looking for esp_gap_ble_api.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_gap_ble_api.h"
* Web  > https://platformio.org/lib/search?query=header:esp_gap_ble_api.h
*
*************************************************************************
```

jwillikers, 2022-01-10:
I have a working example using the ESP-IDF framework instead of the Arduino framework for the ESP32-S2 here: https://github.com/jwillikers/esphome-funhouse.
A lot of functionality is missing and it won't build on `aarch64` Linux yet, but it works.

homonto, 2022-01-10:
ESP32-S2 is fully functional in Arduino both old 1.x and new 2.x
but I wanted ESPHome

jwillikers, 2022-01-10:
> ESP32-S2 is fully functional in Arduino both old 1.x and new 2.x but I wanted ESPHome

It's an ESPHome project.