esphome/issues #2344: pmsa003i no sensor data

omegacore, 2021-08-21:
### The problem

Set up an esp8266 thing with no other components, flashes, boots, and connects to wifi no trouble. ESPHome logs shows I2C bus has a device at address 0x12 as expected. Default configuration used from documentation, added an update rate of 5s to speed up testing.

Logging set to VERY_VERBOSE only shows a ping request, no other errors or information returned by the sensor. HA shows no data values in the entities (which loaded fine).

Deleted and re-added device just to be sure.

Happen to have bought two of these, tried with both.

### Which version of ESPHome has the issue?

2021.8.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

thing

### Component causing the issue

pmsa003i

### Example YAML snippet

```yaml
captive_portal:

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:

ota:

i2c:

sensor:
  - platform: pmsa003i
    pm_1_0:
      name: "PM1.0"
    pm_2_5:
      name: "PM2.5"
    pm_10_0:
      name: "PM10.0"
    pmc_0_3:
      name: "PMC <0.3µm"
    pmc_0_5:
      name: "PMC <0.5µm"
    pmc_1_0:
      name: "PMC <1µm"
    pmc_2_5:
      name: "PMC <2.5µm"
    pmc_5_0:
      name: "PMC <5µm"
    pmc_10_0:
      name: "PMC <10µm"
    update_interval: 5s
```

### Anything in the logs that might be useful for us?

```txt
`[15:54:33][C][i2c:028]: I2C Bus:
[15:54:33][C][i2c:029]:   SDA Pin: GPIO2
[15:54:33][C][i2c:030]:   SCL Pin: GPIO14
[15:54:33][C][i2c:031]:   Frequency: 50000 Hz
[15:54:33][I][i2c:033]: Scanning i2c bus for active devices...
[15:54:33][I][i2c:040]: Found i2c device at address 0x12
[15:54:33][C][logger:189]: Logger:
[15:54:33][C][logger:190]:   Level: VERY_VERBOSE
[15:54:33][C][logger:191]:   Log Baud Rate: 115200
[15:54:33][C][logger:192]:   Hardware UART: UART0
[15:54:33][C][pmsa003i:021]:   Address: 0x12
[15:54:33][C][captive_portal:148]: Captive Portal:
[15:54:33][C][ota:029]: Over-The-Air Updates:
[15:54:33][C][ota:030]:   Address: aqsensor.local:8266
[15:54:33][C][api:095]: API Server:
[15:54:33][C][api:096]:   Address: aqsensor.local:6053
[15:54:34][VV][api.service:327]: on_ping_request: PingRequest {}
[15:54:34][VV][api.service:043]: send_ping_response: PingResponse {}
[15:54:38][VV][api.service:327]: on_ping_request: PingRequest {}
[15:54:38][VV][api.service:043]: send_ping_response: PingResponse {}`
```

omegacore, 2021-08-22:
False alarm. Something was amiss with my esp8266 i2c pins. I swapped to pins 13 and 12 and everything started working.

omegacore, 2021-08-22:
Actually it looks like the sensor might take a while to boot and the esp is starting before it's ready so the initial read fails and the component is marked as failed.

omegacore, 2021-08-24:
Per ssieb on discord I tried setting setup_priority to 250 to give the pmsa more time to initialize. This worked and the sensor now initializes consistently.

sjtrny, 2021-08-28:
I can't replicate this problem. I'm guessing it's one of two things:

1. ESP8266 specific or related hardware issue. I've been testing on an ESP32 (esp32doit-devkit-v1).
2. Different PMSA003i hardware.

Were did you get your PMSA003i from? I've been testing with an Adafruit device purchased January 8, 2021.

omegacore, 2021-09-06:
Adafruit, in August 2021.

closed 2022-01-20 by github-actions[bot]