esphome/issues #2347: Fujitsu Climate IR - "Footer fail"

marciogranzotto, 2021-08-22:
### The problem

99% of the time the footer checking on https://github.com/esphome/esphome/blob/b0fa317302df4f6adeab56034e0b6099f58aa4c4/esphome/components/fujitsu_general/fujitsu_general.cpp#L301 fails.

### Which version of ESPHome has the issue?

v2021.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8.8

### What platform are you using?

ESP32

### Board

ESP32 DevKit v1

### Component causing the issue

fujitsu_general

### Example YAML snippet

```yaml
substitutions:
  room: "Test"

esphome:
  name: test_ac
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${room} Ac Fallback Hotspot"
    password: !secret wifi_password

captive_portal:

# Enable logging
logger:
  level: VERBOSE
  
# Enable Home Assistant API
api:

ota:

# esp32_ble_tracker:

remote_receiver:
  id: receiver
  pin:
    number: GPIO15
    inverted: true
  dump: raw
  tolerance: 40%

remote_transmitter:
[... 269 lines cut ...]
#define FUJITSU_AC_ONE_SPACE   1182U
#define FUJITSU_AC_ZERO_SPACE  367U
#define FUJITSU_AC_TRL_MARK    448U
#define FUJITSU_AC_TRL_SPACE   8100U
```

ESPHome:
```cpp
const uint16_t FUJITSU_GENERAL_HEADER_MARK = 3300;
const uint16_t FUJITSU_GENERAL_HEADER_SPACE = 1600;

const uint16_t FUJITSU_GENERAL_BIT_MARK = 420;
const uint16_t FUJITSU_GENERAL_ONE_SPACE = 1200;
const uint16_t FUJITSU_GENERAL_ZERO_SPACE = 420;

const uint16_t FUJITSU_GENERAL_TRL_MARK = 420;
const uint16_t FUJITSU_GENERAL_TRL_SPACE = 8000;

const uint32_t FUJITSU_GENERAL_CARRIER_FREQUENCY = 38000;
```

`FUJITSU_AC_BIT_MARK    448` seems to be way closer to the values I've been getting then `420`:
```
455, 459, 433, 485, 429, 455, 460, 464
```

marciogranzotto, 2021-08-23:
Well, actually the timings are working if I just remove the footer checking. I don't think we need it since it decodes the code perfectly without that check. I'll open a PR

WeekendWarrior1, 2021-08-24:
What IR receiver are you using?
Different models handle different packet lengths better or worse, and this could be why you're seeing this.

I don't think removing data validation methods is the right way to fix this issue.

marciogranzotto, 2021-08-24:
I'm using a TSOP4838.
Looks like the footer validation doesn't add anything to it. It already validates the header and the expected message body.
Without the footer validation works nearly always, with it, nearly never

closed 2021-08-30 by jesserockz