esphome/issues #2348: MQTT Lights unavailable on HA

joe-sydney, 2021-08-22:
### The problem

I have been using MQTT for all my esphome devices (api disabled) for perhaps over a year now... no problems, except when I upgraded to 2021.8.0. 

MQTT Lights was shown as unavailable on HA 2021.8.x. I then downgraded HA to 2021.7.x and still the same problem.

I then downgraded esphome to 1.20.4 and the lights became available again.

MQTT itself seems to be working, eg I can control the lights via MQTT.

### Which version of ESPHome has the issue?

2021.8.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

Both 2021.7 and 2021.8 (not HA specific)

### What platform are you using?

ESP8266

### Component causing the issue

light

ephirial, 2021-08-23:
Hey, I have the same issue. 
In my case, the "state" attribrute is missing in the mqtt state topic.

I have this configuration:
`light:
  - platform: monochromatic
    name: "wohnzimmer_schreibtisch"
    id: wohnzimmer_schreibtisch
    output: output_ledstripe

output:
  - platform: esp8266_pwm
    id: output_ledstripe
    pin: D3`

And this is what the device sends via MQTT state topic:
`{"color_mode":"brightness","brightness":229,"color":{}}`

The "state" element is missing, which should always send ON or OFF.

joe-sydney, 2021-08-23:
> Hey, I have the same issue.
> In my case, the "state" attribrute is missing in the mqtt state topic.
>
> I have this configuration:
>
> ```
> light:
>   - platform: monochromatic
>     name: [...]

Nice to see you found some clues - I panicked and didn't wait to see what was going on :)

oxan, 2021-08-23:
This is fixed by esphome/esphome#2186.

closed 2021-08-23 by oxan