esphome/issues #2350: Shelly 2.5 negative active power readings

miroslavpetrov, 2021-08-23:
### The problem

When using ESPhome for Shelly 2.5 I get negative active power readings. 
The wiring is 100% correct. The issue is not present with tasmota or the factory fw.
[image]

### Which version of ESPHome has the issue?

2021.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.8.5

### What platform are you using?

ESP8266

### Board

Shelly 2.5

### Component causing the issue

ADE7953

### Example YAML snippet

```yaml
# Basic Config
substitutions:
  devicename: shelly_25

esphome:
  name: ${devicename}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret ssid1
  password: !secret ssid1_pass

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

i2c:
  sda: GPIO12
  scl: GPIO14

sensor:
  - platform: ade7953
    voltage:
      name: ${devicename} Voltage
    current_a:
      name: ${devicename} Current B
    current_b:
      name: ${devicename} Current A
    active_power_a:
      name: ${devicename} Active Power B
      filters:
        - multiply: -1
    active_power_b:
      name: ${devicename} Active Power A
      filters:
        - multiply: -1
    update_interval: 60s

  # NTC Temperature
  - platform: ntc
    sensor: temp_resistance_reading
    name: ${devicename} Temperature
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    icon: "mdi:thermometer"
    calibration:
      b_constant: 3350
      reference_resistance: 10kOhm
      reference_temperature: 298.15K
  - platform: resistance
    id: temp_resistance_reading
    sensor: temp_analog_reading
    configuration: DOWNSTREAM
    resistor: 32kOhm
  - platform: adc
    id: temp_analog_reading
    pin: A0

status_led:
  pin:
    number: GPIO0
    inverted: yes

binary_sensor:
  - platform: gpio
    pin: GPIO16
    name: "ade7953 IRQ pin"
    internal: true
  - platform: gpio
    pin:
      number: GPIO13
    name: ${devicename} Switch1
    on_press:
      then:
        - switch.toggle: shelly_relay_1
  - platform: gpio
    pin:
      number: GPIO5
    name: ${devicename} Switch2
    on_press:
      then:
        - switch.toggle: shelly_relay_2

switch:
  - platform: gpio
    id: shelly_relay_1
    name: ${devicename} Relay1
    pin: GPIO4
    icon: "mdi:electric-switch"
    restore_mode: RESTORE_DEFAULT_OFF
  - platform: gpio
    id: shelly_relay_2
    name: ${devicename} Relay2
    pin: GPIO15
    icon: "mdi:electric-switch"
    restore_mode: RESTORE_DEFAULT_OFF
```

G6EJD, 2021-08-23:
CT fitted the wrong way round, unclip and reverse it, check arrow on base for correct energy flow direction from source to load.

miroslavpetrov, 2021-08-24:
@G6EJD Shelly 2.5 is not using current transformers.

G6EJD, 2021-08-24:
My mistake- sorry

oxan, 2021-08-25:
You have a `multiply: -1` filter in your config, that makes the values negative... Try it without.

closed 2021-08-25 by oxan