esphome/issues #2353: DSMR component using W instead of kW

mmakaay, 2021-08-23:
### The problem

DSMR uses kW for all its values, see for example this small excerpt from the specs:

[image]

However, the DSMR component uses "W" as the unit type.

### Which version of ESPHome has the issue?

2021.8.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.8.0

### What platform are you using?

ESP8266

### Board

ESP-01

### Component causing the issue

dsmr

### Additional information

This was raised as an issue in the comments for the DSMR PR:
https://github.com/esphome/esphome/pull/1881#discussion_r667531234
Frenck requested to post this as a separate issue.

mmakaay, 2021-08-29:
Fixed in
https://github.com/esphome/esphome/pull/2206

mmakaay, 2021-09-09:
I thought this was taken care of in the abovementioned PR, but that one only tackled Wh -> kWh changes.
There are also some sensor values that use kW, but that are still reported as W.
Therefore, I'm reopening this issue.

oxan, 2021-09-14:
> I thought this was taken care of in the abovementioned PR, but that one only tackled Wh -> kWh changes.

It was also for a completely different component, unrelated to DSMR ;)

closed 2021-09-14 by jesserockz