esphome/issues #2354: Logs and other WebUI functionality broken without https

omegacore, 2021-08-24:
### The problem

Webui seems broken when using http to connect.

Connected with http to ha, opened webui, clicked on Logs button to view logs of an active esp, and nothing happens.

Opened another browser window, reconnected to ha with https, and Logs button works.

Seems a warning or a note in the logs might help some people if this was an intentional change.

### Which version of ESPHome has the issue?

2021.8.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Component causing the issue

logger

OttoWinter, 2021-08-30:
Try pressing ctrl+shift+r while in the dashboard view to clear any caches

omegacore, 2021-09-08:
Tried today and issue is gone...

closed 2022-01-22 by github-actions[bot]