esphome/issues #2355: Monochromatic Light don't dim with HA

xtimmy86x, 2021-08-24:
### The problem

After the last update 2018.8.1 all my monochromatic lights don't show the dim bar in Home Assistant 2018.8.8
with the previous version 2018.8,0 all is ok

### Which version of ESPHome has the issue?

2018.8.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2018.8.8

### What platform are you using?

ESP32

### Board

ESP32 dev

### Component causing the issue

monochromatic light

glyndon, 2021-08-24:
same here. Posted about it on the HA forum: https://community.home-assistant.io/t/2021-8-1-breaks-monochromatic-light/332648

kernelpanic85, 2021-08-25:
Just upgraded and seeing the same issues as well

3ative, 2021-08-25:
Same:
All my mono lights have **'color modes'** (_not needed_) and no **'Brightness'** now. :(
[image]

wendelit, 2021-08-25:
I've been trying to understand the ESP home code to try and figure out this problem, and this is a wild guess, but can it be because the protocol buffer code is complied and needs to be re-compiled to take this update into account ?

jesserockz, 2021-08-25:
The issue is that there was a mistake in the color_mode value for monochromatic which made it `2` instead of `3`.

There was also a mistake of including this change in 2021.8.1 when it should have waited until the next release as Home Assistant needs to be updated to recognise `3` (as well as `2` for a period of time).

I could revert the change in ESPHome, but it will need to be made again soon anyway.

closed 2021-08-25 by jesserockz