esphome/issues #2356: Slowing down ESP32 CPU Speed

ripvega, 2021-08-24:
### The problem

I’m working with an ESP32 and an HX711 for measuring weight. I need to measure down to 1gm resolution.

The HX711 works at 80Mhz. The ESP32 is at 240Mhz and this is causing fluctuations in the readings.

[This explains it well](https://www.youtube.com/watch?v=5bdsvoAcUzE).

Arduino has rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M) to slow down the ESP32. Can this be done with ESPHome too?

### Which version of ESPHome has the issue?

1.20.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8.1

### What platform are you using?

ESP32

### Component causing the issue

HX711

G6EJD, 2021-08-25:
Maybe try a different I2C bus speed, default is 50KHz which is very slow, maybe outside of the sensors specification:

frequency (Optional, float): Set the frequency the I²C bus should operate on. Defaults to 50kHz. Values are 50kHz, 100kHz, 200kHz, … 800kHz

ripvega, 2021-08-25:
Thanks for the reply.

I've achieved what I needed using:

```
esphome:
  name: feeder
  platform: ESP32
  board: esp32dev
  platformio_options:
    board_build.f_cpu: 80000000L
```
This has matched the ESP32 to the HX711 and I now have 0.5gm accuracy.

closed 2021-08-25 by ripvega