esphome/issues #2358: MQTT light on HA - Invalid supported_color_modes

joe-sydney, 2021-08-25:
### The problem

Light bulb configured with cwww component does not appear on Home Assistant when upgraded to 2021.8.
HA throws the following exception:
Aug 25 10:50:45 abc hass[27314]:     raise vol.Error(f"Invalid supported_color_modes {sorted(color_modes)}")
Aug 25 10:50:45 abc hass[27314]: voluptuous.error.Error: Invalid supported_color_modes []

The mqtt homeassistant/light message contains the following:
"color_mode":true,"supported_color_modes":[],"brightness":true

Downgrading esphome to 1.20.4 got the light appearing on HA again.

### Which version of ESPHome has the issue?

2021.8

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.7.4

### What platform are you using?

ESP8266

### Component causing the issue

cwww? mqtt?

### Example YAML snippet

```yaml
output:
  - platform: esp8266_pwm
    id: output_warm_white
    pin: GPIO12
  - platform: esp8266_pwm
    id: output_daylight
    pin: GPIO14

light:
  - platform: cwww
    name: ${name}
    id: light1
    cold_white: output_daylight
    warm_white: output_warm_white
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
```

### Anything in the logs that might be useful for us?

```txt
HA logs:
Aug 25 10:50:45 abc hass[27314]:     raise vol.Error(f"Invalid supported_color_modes {sorted(color_modes)}")
Aug 25 10:50:45 abc hass[27314]: voluptuous.error.Error: Invalid supported_color_modes []

The mqtt homeassistant/light message contains the following:
"color_mode":true,"supported_color_modes":[],"brightness":true
```

joe-sydney, 2021-08-25:
Closing as 2021.8.2 has fixed it.

oxan, 2021-08-25:
This happens because HA doesn't support CWWW modes, and apparently we don't expose the CT emulation over MQTT.

Lefuneste83, 2021-08-31:
ESPHOME 2021.8.2
HA Core 2021.8.8
HA Supervisor 2021.06.8

Behaviour is confirmed. I have a home built [QuinLED-Quad](https://quinled.info/2018/09/07/quinled-quad/) to control my office main light in cwww dual white mode. It has been working flawlessly for at least 2 years with full control of the color balance between the 2 channels. After upgrading to latest version of Esphome and rebuilding the ESP32 firmware against it, I suddenly lost the light entity enabling me to control the light itself. I still had reported the 3 others entities : ESP32 onboard temp, Onboard LED toggle switch and Wifi signal strength. I have just spent 2 hours getting totally nuts, trying every possible rollback solutions (software and hardware) and I was about to roll back the entire HA instance when I found this post. 
I am not sure what causes the issue as the topic for the light control is properly queued in the MQTT server. I drilled down the problem to the way HA interprets the entity itself, not being able to expose it as a usable entity in Lovelace, which is super weird as the yaml code for the device shows no anomaly at build time, the entity register properly in MQTT, and it used to work flawlessly for years. What is the origin of the breaking change ? Is it related to some form of color coding not exposed properly anymore to HA ? As stated above I have managed to immediately recover the missing entity by forcefully integrating the device through the add-hoc ESPHOME integration, but this required to enable the api in the device yaml, which was not originally required with this device. So what would be the "proper" way to declare the color channels to avoid this ?
Below my yaml part for the entity. This was a known to work implementation.

```
output:
  - platform: ledc
    pin: 16
    frequency: 40000Hz
    id: LED_gpio_16
  - platform: ledc
    pin: 17
    frequency: 40000Hz
    id: LED_gpio_17 

light:
  - platform: cwww
    name: "Led Control"
    cold_white: LED_gpio_16
    warm_white: LED_gpio_17
    cold_white_color_temperature: 6000 K
    warm_white_color_temperature: 3000 K
    default_transition_length: 1s
```

oxan, 2021-09-02:
I've submitted a PR that should fix this, but as I don't use MQTT, I haven't been able to test this. I'd appreciate it if somebody hitting this issue could try the PR above and report back whether it works.

> What is the origin of the breaking change ?

The introduction of color modes in HA 2021.5, and the changes of ESPHome to support that.

joe-sydney, 2021-09-03:
In relation to the following comment in the PR, could I clarify whether or not CWWW should work before I try this?

"RGBCT is somewhat problematic as HA only offers a RGBWW mode, and we don't have CWWW -> CT emulation in ESPHome at the moment (and I don't have time/interest to add that in the short term). Just expose RGBCT as RGBW for now, it's better than nothing."

> I've submitted a PR that should fix this, but as I don't use MQTT, I haven't been able to test this. I'd appreciate it if somebody hitting this issue could try the PR above and report back whether [...]

joe-sydney, 2021-09-03:
As I only have CWWW, I incorporated the following change only, and this has worked for me:
if (traits.supports_color_mode(ColorMode::COLOR_TEMPERATURE) ||
      traits.supports_color_mode(ColorMode::COLD_WARM_WHITE))
    color_modes.add("color_temp");

I can now see this on MQTT:
"color_mode":true,"supported_color_modes":["color_temp"],"brightness":true

Will update again if there are issues after a while.

joe-sydney, 2021-09-03:
BTW, I noticed comments about RGBCT within the PR. Right now with my RGBWW lights, I actually get 2 separate "lights" created for each device, 1 being the RGB only light, and the 2nd being the CWWW only light. 

I guess with RGBCT, something similar could be done? 1 x RGB light and 1 x CT light (which would use the new color_temperature light platform, which I also use now instead of the CWWW2 custom integration). Presumably in this way, we can get around the current RGBCT issue)?

closed 2021-09-04 by jesserockz

jesserockz, 2021-09-05:
> I guess with RGBCT, something similar could be done? 1 x RGB light and 1 x CT light (which would use the new color_temperature light platform, which I also use now instead of the CWWW2 custom integration). Presumably in this way, we can get around the current RGBCT issue)?

This is controlled by setting `color_interlock: true` in your yaml.