esphome/issues #2366: Sonoff S31s with total_daily_energy enabled reboot

tschundler, 2021-08-28:
### The problem

The devices work file with `total_daily_energy` disabled. But with `total_daily_energy` enabled, they periodically reboot. I added uptime measurement, and it isn't a consistent amount of time - as little as a few minutes and as long as many hours. But I have noticed it reboots sooner on devices using more power.

Is there some sort of possible overflow + divide by zero happening.

It is measuring Watt-hours, not kWh. I'm suspicious this might occur 1000x less if I switch to kWh, but I haven't tested that yet.

### Which version of ESPHome has the issue?

v2021.8.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.8.8

### What platform are you using?

ESP8266

### Board

Sonoff S31

### Component causing the issue

total_daily_energy

### Example YAML snippet

```yaml
`esphome config s31_2.yaml`

substitutions:
  device_prefix: s31_2
  device_hostname: s31-2
  device_title: S31 2
esphome:
  name: s31-2
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout
  esp8266_restore_from_flash: true
  arduino_version: platformio/espressif8266@2.6.2
  build_path: s31-2
  platformio_options: {}
  includes: []
  libraries: []
  name_add_mac_suffix: false
captive_portal: {}
debug: {}
logger:
  baud_rate: 0
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: UART0
  level: DEBUG
  logs: {}
  esp8266_store_log_strings_in_flash: true
web_server:
  port: 80
  css_url: https://esphome.io/_static/webserver-v1.min.css
  js_url: https://esphome.io/_static/webserver-v1.min.js
status_led:
  pin:
    number: 13
    mode: OUTPUT
    inverted: false
binary_sensor:
- platform: gpio
  pin:
    number: 0
    mode: INPUT_PULLUP
    inverted: true
  id: s31_2_button
  disabled_by_default: true
  internal: true
  on_press:
  - then:
    - light.toggle:
        id: s31_2_relayl
  name: s31_2_button
- platform: status
  name: S31 2 Status
  disabled_by_default: false
  device_class: connectivity
output:
- platform: gpio
  id: s31_2_relay
  pin:
    number: 12
    mode: OUTPUT
    inverted: false
light:
- platform: binary
  name: S31 2 Relay
  output: s31_2_relay
  id: s31_2_relayl
  disabled_by_default: false
  restore_mode: RESTORE_DEFAULT_OFF
uart:
- rx_pin: 3
  baud_rate: 4800
  rx_buffer_size: 256
  stop_bits: 1
  data_bits: 8
  parity: NONE
text_sensor:
- platform: template
  name: S31 2 Uptime Human
  id: s31_2_uptime_human
  icon: mdi:clock-start
  internal: false
  disabled_by_default: true
  update_interval: 60s
sensor:
- platform: uptime
  name: S31 2 Uptime
  id: s31_2_uptime_sensor
  update_interval: 60s
  internal: false
  disabled_by_default: true
  on_raw_value:
  - then:
    - text_sensor.template.publish:
        id: s31_2_uptime_human
        state: !lambda |-
          int seconds = round(id(s31_2_uptime_sensor).raw_state);
          int days = seconds / (24 * 3600);
          seconds = seconds % (24 * 3600);
          int hours = seconds / 3600;
          seconds = seconds % 3600;
          int minutes = seconds /  60;
          seconds = seconds % 60;
          return (
            (days ? String(days) + "d " : "") +
            (hours ? String(hours) + "h " : "") +
            (minutes ? String(minutes) + "m " : "") +
            (String(seconds) + "s")
          ).c_str();
  force_update: false
  unit_of_measurement: s
  icon: mdi:timer-outline
  accuracy_decimals: 0
  state_class: ''
- platform: cse7766
  current:
    name: S31 2 Current
    internal: false
    disabled_by_default: true
    force_update: false
    unit_of_measurement: A
    accuracy_decimals: 2
    device_class: current
    state_class: measurement
  voltage:
    name: S31 2 Voltage
    internal: false
    disabled_by_default: true
    force_update: false
    unit_of_measurement: V
    accuracy_decimals: 1
    device_class: voltage
    state_class: measurement
  power:
    name: S31 2 Power
    id: s31_2_power
    internal: false
    disabled_by_default: false
    force_update: false
    unit_of_measurement: W
    accuracy_decimals: 1
    device_class: power
    state_class: measurement
  update_interval: 60s
- platform: total_daily_energy
  name: S31 2 Daily Energy
  power_id: s31_2_power
  disabled_by_default: false
  force_update: false
  device_class: energy
  state_class: measurement
  last_reset_type: auto
  min_save_interval: 0s
time:
- platform: sntp
  servers:
  - mythbox.local
  - north-america.pool.ntp.org
  timezone: PST8PDT7,M3.2.0/2,M11.1.0/2
  update_interval: 15min
wifi:
  ap:
    ssid: S31 2 Setup
    password: !secret 'wifi_password'
    ap_timeout: 1min
  enable_mdns: true
  domain: .local
  reboot_timeout: 15min
  power_save_mode: NONE
  fast_connect: false
  output_power: 20.0
  networks:
  - ssid: XXREDACTEDXX
    password: XXREDACTEDXX
    priority: 0.0
  use_address: s31-2.local
mqtt:
  broker: 192.168.1.180
  username: ''
  password: ''
  discovery: true
  port: 1883
  discovery_retain: true
  discovery_prefix: homeassistant
  topic_prefix: s31-2
  keepalive: 15s
  reboot_timeout: 15min
  birth_message:
    topic: s31-2/status
    payload: online
    qos: 0
    retain: true
  will_message:
    topic: s31-2/status
    payload: offline
    qos: 0
    retain: true
  shutdown_message:
    topic: s31-2/status
    payload: offline
    qos: 0
    retain: true
  log_topic:
    topic: s31-2/debug
    qos: 0
    retain: true
ota:
  password: XXREDACTEDXX
  safe_mode: true
  port: 8266
  reboot_timeout: 5min
  num_attempts: 10
```
```

### Anything in the logs that might be useful for us?

```txt
Serial logging on the S31 interferes with the power sensor, so I'm not sure I'll be able to get good logs. I tried enabling debug logging over MQTT, but didn't notice anything noteworthy.
```

tschundler, 2021-08-28:
Updated with logs. I took a look at the total_daily_energy code, and there isn't even division by anything other than constants, so this is very odd. But it's very reproducible if I comment out total_daily_energy sensor, > 1day  uptme. Re-enable it, and I get reboots.

Maybe I'll look at the diff of the generated code

tschundler, 2021-08-28:
Code diff looks as-expected

```
$ diff s31_4/src/esphome.h s31_4.dp/src/esphome.h
59a60
> #include "esphome/components/total_daily_energy/total_daily_energy.h"
```

```
$ diff s31_4/src/main.cpp s31_4.dp/src/main.cpp
46a47,48
> total_daily_energy::TotalDailyEnergy *total_daily_energy_totaldailyenergy;
> mqtt::MQTTSensorComponent *mqtt_mqttsensorcomponent_5;
491a494,518
>   // sensor.total_daily_energy:
>   //   platform: total_daily_energy
>   //   name: S31 4 Daily Energy
>   //   power_id: s31_4_power
>   //   disabled_by_default: false
>   //   mqtt_id: mqtt_mqttsensorcomponent_5
>   //   force_update: false
>   //   device_class: energy
>   //   state_class: measurement
>   //   last_reset_type: auto
>   //   id: total_daily_energy_totaldailyenergy
>   //   time_id: sntp_sntpcomponent
>   //   min_save_interval: 0s
>   total_daily_energy_totaldailyenergy = new total_daily_energy::TotalDailyEnergy();
>   App.register_component(total_daily_energy_totaldailyenergy);
>   App.register_sensor(total_daily_energy_totaldailyenergy);
>   total_daily_energy_totaldailyenergy->set_name("S31 4 Daily Energy");
>   total_daily_energy_totaldailyenergy->set_disabled_by_default(false);
>   total_daily_energy_totaldailyenergy->set_device_class("energy");
>   total_daily_energy_totaldailyenergy->set_state_class(sensor::STATE_CLASS_MEASUREMENT);
>   total_daily_energy_totaldailyenergy->set_last_reset_type(sensor::LAST_RESET_TYPE_AUTO);
>   total_daily_energy_totaldailyenergy->set_force_update(false);
>   mqtt_mqttsensorcomponent_5 = new mqtt::MQTTSensorComponent(total_daily_energy_totaldailyenergy);
>   App.register_component(mqtt_mqttsensorcomponent_5);
>   total_daily_energy_totaldailyenergy->set_parent(s31_4_power);
509a537,538
>   total_daily_energy_totaldailyenergy->set_time(sntp_sntpcomponent);
>   total_daily_energy_totaldailyenergy->set_min_save_interval(0);
```

tschundler, 2021-08-28:
Next reboot logs.. another watchdog reset

```
13:57:44][D][text_sensor:015]: 'S31 4 Uptime Human': Sending state '48m 44s'
[13:57:44][D][sensor:131]: 'S31 4 Uptime': Sending state 2924.18799 s with 0 decimals of accuracy
[13:58:05][D][sensor:131]: 'S31 4 Power': Sending state 172.48734 W with 1 decimals of accuracy
[13:58:05][D][sensor:131]: 'S31 4 Daily Energy': Sending state 411.97653 Wh with 3 decimals of accuracy

..reboot..

[14:00:13][I][mqtt:212]: MQTT Connected!
...
[14:00:14][D][debug:196]: Reset Reason: Software Watchdog
[14:00:14][D][debug:197]: Reset Info: Fatal exception:4 flag:3 (Software Watchdog) epc1:0x40003b4d epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
...
[14:00:54][D][text_sensor:015]: 'S31 4 Uptime Human': Sending state '49s'
...
[14:00:57][D][sensor:131]: 'S31 4 Power': Sending state 159.15079 W with 1 decimals of accuracy
[14:00:57][D][sensor:131]: 'S31 4 Daily Energy': Sending state 417.18314 Wh with 3 decimals of accuracy
```

Wh in the first reboot: ~157
Wh in the next reboot: ~262

So it's not a rollover thing, since those are very different amounts. Also it is hitting WDT, not something else

tschundler, 2021-08-29:
Since I'm using NTP, I wonder if this is related to #2003 / #2299

...adding memory debugging. I wouldn't expect to see WDT for out of ram though. And why does it seem to die faster on the devices pulling more energy?

I'll add free memory logging.

tschundler, 2021-08-30:
```
[17:22:42][V][mqtt:402]: Publish(topic='s31-5/sensor/s31_5_power/state' payload='157.4' retain=1)
[17:22:59][D][heap:128]: Free heap: 27144 bytes
[17:23:09][D][text_sensor:015]: 'S31 5 Uptime Human': Sending state '5h 8m 7s'
[17:23:09][V][mqtt:402]: Publish(topic='s31-5/sensor/s31_5_uptime_human/state' payload='5h 8m 7s' retain=1)
[17:23:09][V][sensor:037]: 'S31 5 Uptime': Received new state 18487.203125
[17:23:09][D][sensor:131]: 'S31 5 Uptime': Sending state 18487.20312 s with 0 decimals of accuracy
[17:23:09][V][mqtt:402]: Publish(topic='s31-5/sensor/s31_5_uptime/state' payload='18487' retain=1)
[17:25:30][I][mqtt:212]: MQTT Connected!
[17:25:30][C][sntp:022]: Setting up SNTP...
```

...so not a RAM thing. Though will try changing from NTP to HASS as time source

tschundler, 2021-08-30:
Not NTP either - switched to homeassistant API as the time source, and still have regular WDT reboots.

Now trying disabling MQTT completely... maybe it is related to sending more data than it did before?

tschundler, 2021-08-31:
With MQTT disabled, and only using Native API, the device is now up > 24h without reboot. So the problem seems to be related to MQTT. *This also means I have a solution - don't use MQTT.*

*But there is still a bug, and somehow it affects some devices more that others. Maybe related to power usage, now I suspect more likely related to wifi. Bug might be with MQTT or might be with wifi itself.*

I suppose I could tcpdump / wireshark so see if there are incomplete transmissions or something. I took a quick look, and the biggest packet sizes for MQTT writes are ~501b (excluding header, with daily power) vs 455 (excluding header, without daily power). So it is far from the MTU limit. Most of the content is debug messaging for the power monitoring. Looking at a device running the native protobuf API, packets are up to 449 bytes. ...is it a problem only with packets > ~500b?

Maybe next test is MQTT on, daily power on, debug logging off.

tschundler, 2021-09-01:
MQTT on / debug logging off still failed, and the device rebooted. So it's not related to packet size. Also an experiment leaving API only running longer lead to a reboot.

tschundler, 2021-09-04:
And it still seems very tied to how much energy is used, which makes it all the more confusing - one device that I forgot I enabled it on rebooted a few times today after being up for days but switched off. Now switched on, it rebooted once at ~380Wh & again ~90Wh later.

Maybe it's not related to how much power is used but rather how often the number is updated? No reboots when the number stays at 0?

tschundler, 2021-09-28:
oops, I wasn't clear in the last comments.

> Also an experiment leaving API only running longer lead to a reboot.

That was a reference to MQTT off. It still rebooted. It just took over a day (maybe by luck, who knows)

twasilczyk, 2024-02-02:
I had the same problem: device didn't boot after flashing with [example yaml snippet](https://devices.esphome.io/devices/Sonoff-S31). Disabling `time:` and `platform: total_daily_energy` made it check into wifi and operate just fine. I didn't dig deeper to figure out which one of the two helped.