esphome/issues #2367: WS2811 Brightness not adjust and light.is_off reporting incorrectly

ruddj, 2021-08-28:
### The problem

After upgrading to recent 2021.8.2 my WS2811 addressable light is no longer returning a brightness value and when using the conditionals light.is_off or light.is_on it is always reporting the light is off even when it is on.

Code has not changed from previous install of 1.18 but functionality has stopped working.

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8.8

### What platform are you using?

ESP32

### Board

esp32doit-devkit

### Component causing the issue

fastled_clockless

### Example YAML snippet

```yaml
light:
  - platform: fastled_clockless
    chipset: WS2811
    pin: GPIO19
    num_leds: 38
    name: "Headboard Light"
    id: fl_hb_light
    restore_mode: ALWAYS_OFF

  - platform: partition
    name: "Left Light"
    id: fl_hb_light_left
    restore_mode: ALWAYS_OFF
    segments:
      - id: fl_hb_light
        from: 0
        to: 17
  - platform: partition
    name: "Right Light"
    id: fl_hb_light_right
    restore_mode: ALWAYS_OFF
    segments:
      - id: fl_hb_light
        from: 18
        to: 37

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO14
      inverted: True
      mode: INPUT_PULLUP
    name: ${upper_devicename} Left Button
    id: left_button
    on_multi_click:
    - timing:
        - ON for at most 0.5s
        - OFF for at least 0.5s
      then:
        - logger.log: "Single Short Clicked Left Button"
        - logger.log:
            format: "The left light brightness value %.2f"
            args: [ 'id(fl_hb_light_left).current_values.get_brightness()' ]
        - if:
            condition:
              light.is_on: fl_hb_light_left
            then:
            - logger.log: "Left Light was on turning off"
            - light.turn_off: fl_hb_light_left              
            else:
            - logger.log: "Left Light was off turning on based on sunset"
            - light.turn_on: 
                id: fl_hb_light_left
                brightness: 0.5
    
    on_press:
      then:
      - delay: 1.5s # Will interfere with clicks if no delay
      - if:
          condition: 
          - binary_sensor.is_on: left_button
          then:
          - logger.log: "Long Press Button"
          - if:
              # When the light is off dim from low light and bring up
              condition:
              - light.is_off: fl_hb_light_left
              then:
              - light.turn_on: 
                  id: fl_hb_light_left
                  brightness: 0.1
                  transition_length: 0.2s
              - lambda: |-
                  id(dim_up_left) = true;
              else:
                # Every time button pressed will change from dim up to dim down
              - lambda: |-
                  id(dim_up_left) = ! id(dim_up_left);
      - while:
          condition:
          - binary_sensor.is_on: left_button
          then:
          - if:
              condition: 
                lambda: |-
                  if (id(fl_hb_light_left).current_values.get_brightness() >= 1) {
                    id(dim_up_left) = false;
                  } else if (id(fl_hb_light_left).current_values.get_brightness() <= 0.06) {
                    id(dim_up_left) = true;
                  } 
                  return id(dim_up_left);
              then:
              # Increase Brightness
              - light.dim_relative:
                  id: fl_hb_light_left
                  relative_brightness: 5%
                  transition_length: 0.2s
              - delay: 0.2s                  
              else:
              - light.dim_relative:
                  id: fl_hb_light_left
                  relative_brightness: -5%
                  transition_length: 0.2s
              - delay: 0.2s
```

### Anything in the logs that might be useful for us?

```txt
[21:46:09][D][binary_sensor:036]: 'Bed Left Button': Sending state ON
[21:46:10][D][binary_sensor:036]: 'Bed Left Button': Sending state OFF
[21:46:10][D][main:164]: Single Short Clicked Left Button
[21:46:10][D][main:167]: The left light brightness value 1.00
[21:46:10][D][main:178]: Left Light was off turning on based on sunset
[21:46:10][D][light:038]: 'Headboard Light' Setting:
[21:46:10][D][light:053]:   Brightness: 50%
[21:46:10][D][light:087]:   Transition length: 1.0s
[21:46:15][D][binary_sensor:036]: 'Bed Left Button': Sending state ON
[21:46:15][D][binary_sensor:036]: 'Bed Left Button': Sending state OFF
[21:46:16][D][main:164]: Single Short Clicked Left Button
[21:46:16][D][main:167]: The left light brightness value 1.00
[21:46:16][D][main:178]: Left Light was off turning on based on sunset
[21:46:16][D][light:038]: 'Headboard Light' Setting:
[21:46:16][D][light:053]:   Brightness: 50%
[21:46:16][D][light:087]:   Transition length: 1.0s
```

### Additional information

Pressing the button once turns on the light as shown above, but pressing a second time which previously turned it off now has no effect. Logs show it detecting the light as still off and turning on again.

I also have commands to dim the light by holding in the button and these have stopped working as well. It is still possible to control and dim the light from HA control, just not using the button.

The light is a white monochrome addressable LED strip that has been partitioned into 2 lights, however the problem occurs with the main light as well as the partitions.

borky, 2021-08-29:
I have the same problem, the automation condition "light.is_on" is always false.
I am using esphome on esp8266-12F with the neopixelbus platform and a WS2813 light strip.

screenagerbe, 2021-09-03:
Not exactly sure, but I encounter a similar issue since the update to 2021.8 from 1.19

Integration to HA went well and I can activate the LED strips, but when I want to deactivate them the LEDs stay on. When the Color Wipe Effect is active, the wipe effect stops, but the LEDs stay on (same brightness and color) the only way to shut them down is to disconnect the LED strips from the power.

board: nodeMCU ESP32-S 

Below the yaml 
```
light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: GPIO23
    num_leds: 16
    rgb_order: GRB
    name: "LEDstrips"
    id: light1
    default_transition_length: 0s
    color_correct: [70%, 70%, 70%]
    # start with effect when enabling LED in HA
    on_turn_on: 
      - light.turn_on:
          id: light1
          effect: "Color Wipe Effect Random"
    effects:
      - addressable_color_wipe:
      - addressable_color_wipe:
          name: "Color Wipe Effect Random"
          colors:
            - random: true
              num_leds: 1
            - red: 0%
              green: 0%
              blue: 0%
              num_leds: 1
          add_led_interval: 500ms
          reverse: False    
```

sgurgul, 2021-09-03:
I have the same issues @screenagerbe described - impossible to turn off the LED. Board NodeMCU v3.
Issue appeares only after upgrading to 2021.8.

ruddj, 2021-09-05:
I've been doing some further testing and problem occurs when using neopixelbus as well, so looks to be a bug in a higher level light component, most likely addressable_light or light itself.

Neopixel Config
```
  - platform: neopixelbus
    chipset: WS2812 
    pin: GPIO19
    num_leds: 38
    name: "Headboard Light"
    id: fl_hb_light
    restore_mode: ALWAYS_OFF
    effects:
      - random:
      - addressable_scan:
      - addressable_twinkle:
```

Here are some Very Verbose logs showing the addressable light and turning it on to 0.5 brightness. Not sure if these will help.
```
INFO Reading configuration /config/esphome/masterbed_headboard.yaml...
WARNING 'masterbed_headboard': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Detected timezone 'AEST' with UTC offset 10 and daylight saving time from 03 October 02:00:00 to 03 April 03:00:00
INFO Detected timezone 'AEST' with UTC offset 10 and daylight saving time from 03 October 02:00:00 to 03 April 03:00:00
INFO Starting log output from masterheadboard.home.jrudd.org using esphome API
INFO Connecting to masterheadboard.home.jrudd.org:6053 (192.168.0.82)
INFO Successfully connected to masterheadboard.home.jrudd.org
[23:00:39][I][app:106]: ESPHome version 2021.8.2 compiled on Sep  5 2021, 22:57:37
[23:00:39][C][wifi:499]: WiFi:
[23:00:39][C][wifi:359]:   SSID: [redacted]
[23:00:39][C][wifi:360]:   IP Address: 192.168.0.82
[23:00:39][C][wifi:362]:   BSSID: [redacted]
[23:00:39][C][wifi:363]:   Hostname: 'masterbed_headboard'
[23:00:39][C][wifi:367]:   Signal strength: -58 dB ▂▄▆█
[23:00:39][V][wifi:369]:   Priority: 0.0
[23:00:39][C][wifi:371]:   Channel: 11
[23:00:39][C][wifi:372]:   Subnet: 255.255.255.0
[23:00:39][C][wifi:373]:   Gateway: 192.168.0.1
[23:00:39][C][wifi:374]:   DNS1: 192.168.0.183
[23:00:39][C][wifi:375]:   DNS2: 192.168.0.182
[23:00:39][C][template.binary_sensor:018]: Template Binary Sensor 'masterbed_headboard_timer'
[23:00:39][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Bed Left Button'
[23:00:39][C][gpio.binary_sensor:016]:   Pin: GPIO14 (Mode: INPUT_PULLUP, INVERTED)
[23:00:39][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Bed Right Button'
[23:00:39][C][gpio.binary_sensor:016]:   Pin: GPIO12 (Mode: INPUT_PULLUP, INVERTED)
[23:00:39][C][fastled:019]: FastLED light:
[23:00:39][C][fastled:020]:   Num LEDs: 38
[23:00:39][C][fastled:021]:   Max refresh rate: 400
[23:00:39][C][uptime.sensor:030]: Uptime Sensor 'uptime_sec'
[23:00:39][C][uptime.sensor:030]:   State Class: ''
[23:00:39][C][uptime.sensor:030]:   Unit of Measurement: 's'
[23:00:39][C][uptime.sensor:030]:   Accuracy Decimals: 0
[23:00:39][C][uptime.sensor:030]:   Icon: 'mdi:timer-outline'
[23:00:39][V][uptime.sensor:030]:   Unique ID: '240ac4c61510-uptime'
[23:00:40][C][template.text_sensor:020]: Template Sensor 'Bed Uptime'
[23:00:40][C][template.text_sensor:020]:   Icon: 'mdi:clock-start'
[23:00:40][C][logger:189]: Logger:
[23:00:40][C][logger:190]:   Level: VERY_VERBOSE
[23:00:40][C][logger:191]:   Log Baud Rate: 115200
[23:00:40][C][logger:192]:   Hardware UART: UART0
[23:00:40][C][light:097]: Light 'Headboard Light'
[23:00:40][C][light:099]:   Default Transition Length: 1.0s
[23:00:40][C][light:100]:   Gamma Correct: 2.80
[23:00:40][C][light:097]: Light 'Bed Left Light'
[23:00:40][C][light:099]:   Default Transition Length: 1.0s
[23:00:40][C][light:100]:   Gamma Correct: 2.80
[23:00:40][C][light:097]: Light 'Bed Right Light'
[23:00:40][C][light:099]:   Default Transition Length: 1.0s
[23:00:40][C][light:100]:   Gamma Correct: 2.80
[23:00:40][C][homeassistant.time:010]: Home Assistant Time:
[23:00:40][C][homeassistant.time:011]:   Timezone: 'AEST-10AEDT-11,M10.1.0/2,M4.1.0/3'
[... 872 lines cut ...]
[23:01:08][VV][light.addressable:020]:   [ 1] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [ 2] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [ 3] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [ 4] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [ 5] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [ 6] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [ 7] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [ 8] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [ 9] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [10] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [11] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [12] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [13] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [14] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [15] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [16] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [17] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [18] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:020]:   [19] Color: R=  0 G=  0 B=  0 W=  0
[23:01:08][VV][light.addressable:022]: 
[23:01:09][VV][api.service:327]: on_ping_request: PingRequest {}
[23:01:09][VV][api.service:043]: send_ping_response: PingResponse {}
[23:01:11][VV][api.service:327]: on_ping_request: PingRequest {}
[23:01:11][VV][api.service:043]: send_ping_response: PingResponse {}
```

sgurgul, 2021-09-05:
In my case the issue appears only on devices using light partitions.
I have another LEDs - same NodeMCU board, FastLED as well but no partitions and there are no similar issues observed.

In case of partitions, the workaround which works for me is to disable each partition light first, before the main light is disabled. That seems to be a reliable procedure to turn on the light. If any partitions remains ON when main light is turned off than it's woken up automatically.

ruddj, 2021-09-06:
Thanks for the suggestion Sebastian. I just tried testing light with no partitions in yaml and problem still occurs. It would not detect light as being on and gave no brightness info.

ruddj, 2021-09-07:
As a workaround I have replaced all instances of `light.is_on: myid` with `lambda: 'return id(myid).remote_values.is_on();' ` as this seems to get around the faulty values.
Testing it seems `id(myid).current_values` is returning the faulty numbers and these are then used by light.is_on funtion.

oxan, 2021-09-15:
@screenagerbe @sgurgul That's a different problem, please open a separate issue (and please give some more information (e.g. logs), as I can't reproduce it with just the info you've given here).

screenagerbe, 2021-09-15:
> 
> 
> @screenagerbe @sgurgul That's a different problem, please open a separate issue (and please give some more information (e.g. logs), as I can't reproduce it with just the info you've given here).

Thanks, but I've updated to version 2021.8.2 this morning and the issue was gone.

sgurgul, 2021-09-16:
Mine still exists. Looks like a different issue, indeed.
Reporter as #2429

closed 2021-09-19 by OttoWinter

ruddj, 2021-09-20:
Thanks @OttoWinter. Just confirming #2320 in 2021.9.1 has solved the issue.