esphome/issues #2371: OTA programming failing.

markusg1234, 2021-08-30:
### The problem

Recently one of my ESP32 devices comes up with an error if using OTA for programming. I am unsure where to view the MQTT/USB logs to find out why it failed.

USB programming via HTTPS or ESP flasher works correctly.

So far I have only noticed one device with the issue.  

With this issue I have noticed also that the programming upload speed seems to be very fast compared to normal speed for when it was working.

### Which version of ESPHome has the issue?

v2021.9.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.0b3

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/computer-room-blind.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/computer-room-blind
Processing computer-room-blind (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Retrieving maximum program size /data/computer-room-blind/.pioenvs/computer-room-blind/firmware.elf
Checking size /data/computer-room-blind/.pioenvs/computer-room-blind/firmware.elf
RAM:   [==        ]  16.4% (used 53780 bytes from 327680 bytes)
Flash: [========  ]  83.1% (used 1524190 bytes from 1835008 bytes)
========================= [SUCCESS] Took 1.71 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of computer-room-blind.local
INFO  -> 192.168.1.23
INFO Uploading /data/computer-room-blind/.pioenvs/computer-room-blind/firmware.bin (1524304 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.
```

OttoWinter, 2021-08-30:
> See the MQTT/USB logs for more information.

Please post that.

markusg1234, 2021-08-30:
Hi,
Where do I find the MQTT/USB logs? Have looked for them but can't find where they are stored.

markusg1234, 2021-08-31:
Hi,
I have found this log for when it fails.

INFO 200 GET /static/js/esphome/c.9481058d.js (0.0.0.0) 0.83ms
INFO 101 GET /upload (0.0.0.0) 1.68ms
INFO Running command 'esphome --dashboard run /config/esphome/computer-room-blind.yaml --device OTA'
INFO Process exited with return code 1

markusg1234, 2021-09-01:
Hi,
I am now have the same flashing issue with esphome-flasher and it fails due to MD5 error.

Using 'COM9' as serial port.
Connecting......
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 08:3A:F2:AC:65:F4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.0s

Writing at 0x000ec000... (100 %)Wrote 1524576 bytes (911960 compressed) at 0x00010000 in 21.0 seconds (effective 580.5 kbit/s)...
File  md5: 7ef26f1b7ea0d6eac01caeeb60233755
Flash md5: 33145459d74759db3637ef051027670b
MD5 of 0xFF is 7e9d7999650705ceb77614023e130480
Unexpected error: Error while writing flash: MD5 of file does not match data in flash!

Regards,
Markus

markusg1234, 2021-09-02:
Hi,
Been able to get it to work again by deleting the device in the ESPhome integration and then doing a install again.

Regards,
Markus

pnt-thiago, 2021-09-04:
> Hi,
> I now have the same flashing issue with esphome-flasher and it fails due to a MD5 error.
> 
> Using 'COM9' as serial port.
> Connecting......
> Detecting chip type... ESP32
> Connecting....
> 
> Chip Info:
> 
> * Chip Family: ESP32
> * Chip Model: ESP32D0WDQ6 (revision 1)
> * Number of Cores: 2
> * Max CPU Frequency: 240MHz
> * Has Bluetooth: YES
> * Has Embedded Flash: NO
> * Has Factory-Calibrated ADC: YES
> * MAC Address: 08:3A:F2:AC:65:F4
>   Uploading stub...
>   Running stub...
>   Stub running...
>   Changing baud rate to 460800
>   Changed.
> * Flash Size: 4MB
> * Flash Mode: dio
> * Flash Frequency: 40MHz
>   Erasing flash (this may take a while)...
>   Chip erase completed successfully in 0.0s
> 
> Writing at 0x000ec000... (100 %)Wrote 1524576 bytes (911960 compressed) at 0x00010000 in 21.0 seconds (effective 580.5 kbit/s)...
> File md5: 7ef26f1b7ea0d6eac01caeeb60233755
> Flash md5: 33145459d74759db3637ef051027670b
> MD5 of 0xFF is 7e9d7999650705ceb77614023e130480
> Unexpected error: Error while writing flash: MD5 of file does not match data in flash!
> 
> Regards,
> Markus

Dear Markus,
That error sometimes apears when I have something connected to one of these pins (Straping Pins):
GPIO 0
GPIO 2
GPIO 4
GPIO 5 (must be HIGH during boot)
GPIO 12 (must be LOW during boot)
GPIO 15 (must be HIGH during boot)
Just yesterday I was experiencing the same MD5 error when I had a relay module connected to Pin 12 that sometimes strapped it to high.
Hi @OttoWinter, do you think it should help to make a little warning message when the user attempts to define one of this pins? I would be glad to provide a pull request to address that.
Best regards.
Thiago Pontes

markusg1234, 2021-09-04:
Hi Thiago,

Thanks for the info.

I haven't had the issue again since deleting the device and putting it back in. 

I do though have GPIO 2 and 15 programmed as inputs but are not being used with pulldown programmed. 

I might change these.

The device is a ESP32-Wroom-32D where in the past I have used older ESP's.

Regards,
Markus

pnt-thiago, 2021-09-04:
> Hi Thiago,
>
> Thanks for the info.
>
> I haven't had the issue again since deleting the device and putting it back in.
>
> I do though have GPIO 2 and 15 programmed as inputs but are not being used with [...]

Using them as inputs is a bit more problematic because it can mess with the boot if they happen to be in the "incorrect" state when booting... As I read ESP docs, it doesn't care with programmable pull up/down as they're set after initial boot, the problem is with the state of the pin itself, when externally driven.

Regards,
Thiago Pontes

closed 2022-01-17 by github-actions[bot]