esphome/issues #2372: Unable to perform OTA update soon after upgrade from 1.16.2 to 2021.8.2

Zirafarafa, 2021-08-30:
### The problem

I had experienced no issues installing code updates over OTA in version 1.16.2.  I upgraded to 2021.8.2 this weekend, and now I have one device which is unable to push updates via OTA (this device is quite difficult to access, so working OTA is necessary)

The device is stable on wifi, until I try perform an update.  This looks like it may be a recurrence of #455 but that was closed in 2019 and it is unclear whether the underlying issue has been fixed in esphome/upstream code, or whether I should perform one of the following two options:

* Set `wifi.output_power` to a lower value, eg `17.5dB`
* Set the flash speed lower using `esptool.py --flash_freq 26m`

Once I can access the device to reflash it via usb, I will be able to get additional logs if needed, but please let me know whether one of the above options is necessary for me as well.

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

api? wifi?

### Example YAML snippet

```yaml
This is the latest config which flashed successfully.  It uses some custom code (which is currently under development) which interfaces with a serial device.  Every 60 seconds it sends a query, and populates a sensor.

esphome:
  name: d1_mini
  platform: ESP8266
  board: d1_mini
  includes:
    - uart_read_line_sensor.h
    
#  libraries:
#    - "CSV Parser"

wifi:
  ssid: "ABCDEFG"
  password: "XXXXXXXXXXX"

logger:
  baud_rate: 0
  level: VERBOSE

api:
  password: "aaaaaaaaaaaaaa"

ota:
  password: "aaaaaaaaaaaaaaaa"

uart:
  id: uart_bus
  baud_rate: 115200
  tx_pin: GPIO1
  rx_pin: GPIO3
  
sensor:
- platform: custom
  lambda: |-
    auto neato = new UartReadLineSensor(id(uart_bus));
    App.register_component(neato);
    return {neato->s_battery_temp};
  sensors:
    name: "Battery Temperature"
    id: "s_battery_temp"
    unit_of_measurement: °C
```

### Anything in the logs that might be useful for us?

```txt
Console logs when performing update:
---------------
INFO Successfully compiled program.
INFO Resolving IP address of d1_mini.local
INFO  -> 172.17.107.135
INFO Uploading /config/d1_mini/.pioenvs/d1_mini/firmware.bin (344784 bytes)
ERROR Error receiving acknowledge version: timed out
--------------

Logs in a second console while doing update:
---------------
INFO Reading configuration /config/d1_mini.yaml...
INFO Starting log output from d1_mini.local using esphome API
INFO Connecting to d1_mini.local:6053 (172.17.107.135)
INFO Successfully connected to d1_mini.local
[04:29:26][I][app:106]: ESPHome version 2021.8.2 compiled on Aug 29 2021, 14:53:00
[04:29:26][C][wifi:499]: WiFi:
[04:29:26][C][wifi:359]:   SSID: [redacted]
[04:29:26][C][wifi:360]:   IP Address: 172.17.107.135
[04:29:26][C][wifi:362]:   BSSID: [redacted]
[04:29:26][C][wifi:363]:   Hostname: 'd1_mini'
[04:29:26][C][wifi:367]:   Signal strength: -60 dB ▂▄▆█
[04:29:26][V][wifi:369]:   Priority: 0.0
[04:29:26][C][wifi:371]:   Channel: 6
[04:29:26][C][wifi:372]:   Subnet: 255.255.255.0
[04:29:26][C][wifi:373]:   Gateway: 172.17.107.254
[04:29:26][C][wifi:374]:   DNS1: 172.17.107.254
[04:29:26][C][wifi:375]:   DNS2: (IP unset)
[04:29:26][C][uart_esp8266:075]: UART Bus:
[04:29:26][C][uart_esp8266:077]:   TX Pin: GPIO1
[04:29:26][C][uart_esp8266:080]:   RX Pin: GPIO3
[04:29:26][C][uart_esp8266:081]:   RX Buffer Size: 256
[04:29:26][C][uart_esp8266:083]:   Baud Rate: 115200 baud
[04:29:26][C][uart_esp8266:084]:   Data Bits: 8
[04:29:26][C][uart_esp8266:085]:   Parity: NONE
[04:29:26][C][uart_esp8266:086]:   Stop bits: 1
[04:29:26][C][uart_esp8266:088]:   Using hardware serial interface.
[04:29:26][C][logger:189]: Logger:
[04:29:26][C][logger:190]:   Level: VERBOSE
[04:29:26][C][logger:191]:   Log Baud Rate: 0
[04:29:26][C][logger:192]:   Hardware UART: UART0
[04:29:26][C][ota:029]: Over-The-Air Updates:
[04:29:26][C][ota:030]:   Address: d1_mini.local:8266
[04:29:26][C][ota:032]:   Using Password.
[04:29:26][C][api:095]: API Server:
[04:29:26][C][api:096]:   Address: d1_mini.local:6053
[04:30:14][D][ReadSerialCommand:045]: Sending GetCharger
[04:30:14][D][DebugSerialCommand:095]: Read 280 bytes
[04:30:14][D][log_lines:035]:   GetCharger#
[04:30:14][D][log_lines:035]:   Label,Value#
[04:30:14][D][log_lines:035]:   FuelPercent,16#
[04:30:14][D][log_lines:035]:   BatteryOverTemp,0#
[04:30:14][D][log_lines:035]:   ChargingActive,0#
[04:30:14][D][log_lines:035]:   ChargingEnabled,1#
[04:30:14][D][log_lines:035]:   ConfidentOnFuel,0#
[04:30:14][D][log_lines:035]:   OnReservedFuel,0#
[04:30:14][D][log_lines:035]:   EmptyFuel,0#
[04:30:14][D][log_lines:035]:   BatteryFailure,0#
[04:30:14][D][log_lines:035]:   ExtPwrPresent,0#
[04:30:14][D][log_lines:035]:   ThermistorPresent,1#
[04:30:14][D][log_lines:035]:   BattTempCAvg,24#
[04:30:14][D][log_lines:035]:   VBattV,12.46#
[04:30:14][D][log_lines:035]:   VExtV,0.77#
[04:30:14][D][log_lines:035]:   Charger_mAH,0#
[04:30:14][D][log_lines:035]:   Discharge_mAH,135#
[04:30:14][V][sensor:037]: 'Battery Temperature': Received new state 24.000000
[04:30:14][D][sensor:131]: 'Battery Temperature': Sending state 24.00000 °C with 0 decimals of accuracy
WARNING Disconnected from API: Timeout while waiting for message response!
WARNING Error resolving IP address of d1_mini.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Couldn't connect to API (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -2] Name or service not known). Trying to reconnect in 1 seconds
INFO Connecting to d1_mini.local:6053 (172.17.107.135)
INFO Successfully connected to d1_mini.local
[04:32:16][D][ReadSerialCommand:045]: Sending GetCharger
[04:32:16][D][DebugSerialCommand:095]: Read 280 bytes
[04:32:16][D][log_lines:035]:   GetCharger#
[04:32:16][D][log_lines:035]:   Label,Value#
[04:32:16][D][log_lines:035]:   FuelPercent,16#
[04:32:16][D][log_lines:035]:   BatteryOverTemp,0#
[04:32:16][D][log_lines:035]:   ChargingActive,0#
[04:32:16][D][log_lines:035]:   ChargingEnabled,1#
[04:32:16][D][log_lines:035]:   ConfidentOnFuel,0#
[04:32:16][D][log_lines:035]:   OnReservedFuel,0#
[04:32:16][D][log_lines:035]:   EmptyFuel,0#
[04:32:16][D][log_lines:035]:   BatteryFailure,0#
[04:32:16][D][log_lines:035]:   ExtPwrPresent,0#
[04:32:16][D][log_lines:035]:   ThermistorPresent,1#
[04:32:16][D][log_lines:035]:   BattTempCAvg,24#
[04:32:16][D][log_lines:035]:   VBattV,12.45#
[04:32:16][D][log_lines:035]:   VExtV,0.77#
[04:32:16][D][log_lines:035]:   Charger_mAH,0#
[04:32:16][D][log_lines:035]:   Discharge_mAH,138#
[04:32:16][V][sensor:037]: 'Battery Temperature': Received new state 24.000000
[04:32:16][D][sensor:131]: 'Battery Temperature': Sending state 24.00000 °C with 0 decimals of accuracy
--------------
```

Zirafarafa, 2021-09-11:
It looks like my problem was actually caused by a faulty 3.3V connection.  After redoing the wiring, I have not issues flashing OTA.

closed 2021-09-11 by Zirafarafa

SeByDocKy, 2021-09-11:
I downgraded to the stable version and everything reworks fine.... It's a know problem with UART connexion introduced recently