esphome/issues #2373: SDM Meter energy sensor did not listed in home assistant Energy config panel.

hawkefly, 2021-08-30:
### The problem

SDM Meter import and export active energy sensor did not listed in home assistant Energy config panel.

This is the picture of  consumed energy(kWh) 

[image]

But DSMR component works fine with home assistant energy config. 

I think , to be listed in consumed energy(kWh) , the sensor's unit should be kWh, and device type should be energy , and last_reset_type should not be null .  

this is the diff

https://github.com/esphome/esphome/blob/f923ba87c0c55e219e55993e13596b2a60724d37/esphome/components/sdm_meter/sensor.py#L108

https://github.com/esphome/esphome/blob/f923ba87c0c55e219e55993e13596b2a60724d37/esphome/components/dsmr/sensor.py#L42

### Which version of ESPHome has the issue?

ESPHome  dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8.8

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

sdm_meter

hawkefly, 2021-09-01:
@jesserockz

jesserockz, 2021-09-01:
All of the requirements for SDM to appear are met by SDM. Not sure why you would not see the entities

[image]

closed 2021-09-04 by hawkefly