esphome/issues #2375: namespace issues with climate::ClimatePreset

slaroy, 2021-08-31:
### The problem

The following lambda code:

```
auto call = id(climate1).make_call();
  esphome::climate::ClimatePreset preset = call.get_preset();
  ESP_LOGD("ADC","Climate preset %s", esphome::climate::climate_preset_to_string(preset));

```
Gives the following error when trying to compile.
_error: cannot convert 'const esphome::optional\<esphome::climate::ClimatePreset\>' to 'esphome::climate::ClimatePreset' in initialization_

I think this is a problem with the namespaces, and not my lambda code, but I don't know enough about classes to find where the problem lies.

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

climate

nuttytree, 2021-08-31:
This is actually not a namespace issue, the ClimateCall.get_preset() method returns an "optional" ClimatePreset not an actual ClimatePreset, this is because there might not be a preset available. The correct code is:
```c++
auto call = id(climate1).make_call();
esphome::optional<esphome::climate::ClimatePreset> preset = call.get_preset();
if (preset.has_value()) {
  ESP_LOGD("ADC","Climate preset %s", esphome::climate::climate_preset_to_string(preset.value()));  
} else {
  ESP_LOGD("ADC","No climate preset available");
}
```
You could also use auto for the preset instead of specifying the type but you would still need to check if it has a value.

slaroy, 2021-09-01:
Thank you! Your code does compile, even with the auto type. I think my major issue is that the documentation on the website is starting to diverge from the codebase, as it still references the old "away" functionality and doesn't explain how the presets work, how to configure them in yaml, or how to use lambda functions to access or change them. I would offer to help document this, but I haven't fully figured it out myself.

closed 2021-09-01 by slaroy