esphome/issues #2376: SCD30 CO2 no "statistics" available in Home Assistant

shaunbro, 2021-08-31:
### The problem

ESPHome 2021.8.2

Home Assistant does not provide statistics for CO2 measurements for this device. 

Running Validate against the config shows that device_class is not set, manually setting to "carbon_dioxide" does not seem to change the behaviour. Temperature and Humidity show correctly.

`- platform: scd30
  co2:
    name: SCD30 CO2
    accuracy_decimals: 1
    device_class: carbon_dioxide
    disabled_by_default: false
    force_update: false
    unit_of_measurement: ppm
    icon: mdi:molecule-co2
    state_class: measurement`

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.8.8

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

SCD30

### Example YAML snippet

```yaml
sensor:
  - platform: scd30
    co2:
      name: "SCD30 CO2"
      accuracy_decimals: 1
      device_class: "carbon_dioxide"
    temperature:
      name: "SCD30 Temperature"
      accuracy_decimals: 1
    humidity:
      name: "SCD30 Humidity"
      accuracy_decimals: 1
    address: 0x61
    update_interval: 5s
```

sjtrny, 2021-09-01:
This is the current

sjtrny, 2021-09-01:
I suspect that this is because the `device_class` is not set in the config ([sensor.py](https://github.com/esphome/esphome/blob/dev/esphome/components/scd30/sensor.py)) of the component.

Hopefully someone with more experience can confirm.

This is the current config:

```
            cv.Optional(CONF_CO2): sensor.sensor_schema(
                unit_of_measurement=UNIT_PARTS_PER_MILLION,
                icon=ICON_MOLECULE_CO2,
                accuracy_decimals=0,
                state_class=STATE_CLASS_MEASUREMENT,
            ),
```

mschilt, 2021-09-02:
Same issue here.. 

There seems to be a device class for CO2 sensors: DEVICE_CLASS_CARBON_DIOXIDE defined here: 
https://github.com/esphome/esphome/blob/edcd88123d37e7484160331a21a1a4f0673d0a98/esphome/const.py#L836

but I am not sure if this will help us. or did your see any log message that complains about a missing device class?

oxan, 2021-09-06:
This was probably caused by HA restricting statistics to temperature, humidity, power and energy related sensors prior to 2021.9. Feel free to reopen this if you can still reproduce the problem with HA 2021.9.

closed 2021-09-06 by oxan

shaunbro, 2021-09-07:
Confirming this now works having upgraded to core-2021.9.3 thanks all.