esphome/issues #2377: Cover Component documentation error

pewu78, 2021-09-01:
### The problem

[Cover documentation](https://esphome.io/components/cover/index.html)
lambda calls publish_state description / examples are wrong:

> publish_state(): Manually cause the cover to publish a new state and store it internally. If it’s different from the last internal state, it’s additionally published to the frontend.
> 
> // Within lambda, make the cover report a specific state
> id(my_cover).publish_state(COVER_OPEN);
> id(my_cover).publish_state(COVER_CLOSED);

Actual code:

> /** Publish the current state of the cover.
>    *
>    * First set the .position, .tilt, etc values and then call this method
>    * to publish the state of the cover.
>    *
>    * @param save Whether to save the updated values in RTC area.
>    */
>   void publish_state(bool save = true);
> 

Next example is also wrong:

> state: Retrieve the current state of the cover.
> 
> if (id(my_cover).position == COVER_OPEN) {
>   // Cover is open
> } else {
>   // Cover is closed
> }

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8

### What platform are you using?

ESP8266

oxan, 2021-09-06:
> Next example is also wrong:

What specifically is wrong with that example? From a quick look at the code it seems correct to me.

pewu78, 2021-09-07:
> > Next example is also wrong:
> 
> What specifically is wrong with that example? From a quick look at the code it seems correct to me.

It mentions _state_ but example checks _position_ variable.

closed 2021-09-16 by jesserockz