esphome/issues #2378: ESPHome on Intel NUC - All Devices have connection issues

GazTheGeek, 2021-09-01:
### The problem

I recently upgrade my HA install from a Raspberry Pi4b to an Intel NUC. I simply restored a snapshot from the Pi and everything was working fine. That is except for all my ESPHome devices. They are all disconnecting and reconnecting every second.

If I disconnect the NUC from the network and connect the Pi4 - Everything goes back to normal. All working fine.

I also have ESPHome desktop installed on docker for compiling (The Pi4b was struggling) When the NUC in on the network I can't compile.

Clearly its a NUC networking issue the relates to ESPHOme only. All other Integrations and add-on seem to be working fine.

Log info below:

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.8.8

### What platform are you using?

ESP8266

### Board

All boards are D1-Mini

### Component causing the issue

All

### Example YAML snippet

```yaml
I don't think this is relevant.
```

### Anything in the logs that might be useful for us?

```txt
Example log from one device - The log is from my laptop docker install of ESPHome.

[09:19:42][D][api:067]: Disconnecting Home Assistant 2021.8.8 (10.0.0.13)
[09:19:42][D][api.connection:670]: Client 'Home Assistant 2021.8.8 (10.0.0.13)' connected successfully!
[09:19:47][D][api:067]: Disconnecting Home Assistant 2021.8.8 (10.0.0.13)
[09:19:47][D][api.connection:670]: Client 'Home Assistant 2021.8.8 (10.0.0.13)' connected successfully!

And logs from HA

Logger: homeassistant.components.esphome
Source: components/esphome/__init__.py:403 
Integration: ESPHome (documentation, issues) 
First occurred: 4:10:51 PM (286 occurrences) 
Last logged: 7:19:56 PM

Can't connect to ESPHome API for light-sensor-02 (10.0.0.120): Error connecting to ('10.0.0.120', 6053): [Errno 111] Connect call failed ('10.0.0.120', 6053)
Can't connect to ESPHome API for light-sensor-02 (10.0.0.120): Timeout while waiting for API response!
Can't connect to ESPHome API for motion-entry (10.0.0.34): Timeout while waiting for API response!
Can't connect to ESPHome API for blinds01 (10.0.0.30): Timeout while waiting for API response!
Can't connect to ESPHome API for light-sensor-01 (10.0.0.117): Timeout while waiting for API response!
```

### Additional information

If anyone has some things in HA to look regarding the networking components that would be helpful.

Again if I reconnect the Pi4 to the network with exactly the same setup these issue go away. Its definitely this HA install on the NUC causing the issue.

Thanks in advance

GazTheGeek, 2021-09-03:
OK Problem solved. After doing a full reinstallation I discovered the issue was caused by setting a static IP address in Supervisor / System / Host. Strange all other services were OK but ESPHome was failing. Putting the IPAddress back to DHCP and setting whatever address was assigned as a static address in m y router fixed the issue.

FreddyLauwers, 2021-09-03:
Glad that worked out for you.
Both my Lan and IoT vlan interfaces get their IP address through a static IP reservation on the firewall, so they already use DHCP...
ESPHome is also the only service that has any problem at this moment.
I'll probably have to try a full reinstall also. But so strange we seem to be the only people who reported this problem...
Thank You.

GazTheGeek, 2021-09-04:
In hindsight had I known the issue I could have fixed it on the spot by reverting the HA IPAddress back to DHCP. There was no need for a full reinstall. 

The NUC install appears to sit on top of a version of Docker so the IP Addresses are a little bit hard to figure out for a newbie like me.

I found that the MAC address I was using the reserve the IP address was one HA ifconfig was showing but when I used Wire Shark to sniff the traffic I was getting a different MAC address. I'm guessing docker is exposing a virtual adapter to HA but the NUC Hardware MAC is the one I needed.

I agree Its strange that this is not more wide spread - It's possible there is some odd use case that has something to do with my Router but I don't know how diagnose it.

Everything is working fine now and IP address appears to be stable so things like InfluxDB are working. 

Also odd no-one else is commenting or has picked this up to help out. I realise that's the nature of open source - no guaranteed support in return for free to use. A fair trade in my book. HA is a very cool piece of kit. Thanks to all who contribute.

FreddyLauwers, 2021-09-04:
In the meantime, I have reinstalled HA from scratch and then restored my latest backup and everything works just fine, every time. All ESPHome devices came online immediately and stayed online… But I hadn’t configured my IoT VLAN yet and I let the devices connect through a firewall rule to the main adapter. After I configured my IoT VLAN on HA and let the device connect to the interface on this VLAN, problems came back immediately. So in my humble opinion, this has something to do with ESPHome and VLANs. Maybe this is also an explanation why most people don’t experience this problem if they use everything on the same lan.
Just wondering, were you using a VLAN configuration when you had these problems?

GazTheGeek, 2021-09-05:
No I am not using a VLAN. But I do think there is something happening between HA and what appears to be a docker container between HA and the hardware adapter on the NUC. But that's above my pay grade I'm afraid.

andriej, 2023-02-23:
I've came across this issue looking for "Connection error occurred" on my nodes. 
I have same setup (proxmox with HA OS + esphome in VM) and would like to debug further