esphome/issues #2379: mDNS fails at random

insanemal, 2021-09-01:
### The problem

When using the esphome plugin for Home Assistant the devices flick randomly to offline. When this happens I can't get logs or flash devices.

From HomeAssistant it's working fine, the sensors are sending data.

when using esphome command line tool on my desktop it also doesn't work.

mDNS only resolves for a short period after booting the device

### Which version of ESPHome has the issue?

2021.8.2-1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8.8

### What platform are you using?

ESP8266

### Board

D1 mini

### Component causing the issue

mDNS

### Example YAML snippet

```yaml
esphome:
  name: aq-sensor-1
  platform: ESP8266
  board: d1_mini

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

ota:
  password: "c844052655ba898260f2927740807f6e"

wifi:
  ssid: "ii0D74F6primary"
  password: "bachpad57"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Aq-Sensor-1 Fallback Hotspot"
    password: "XPWa8RmJtF7V"

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 9600

sensor:
  - platform: pmsx003
    type: PMSX003
    pm_1_0:
      name: "Particulate Matter <1.0µm Concentration"
    pm_2_5:
      name: "Particulate Matter <2.5µm Concentration"
    pm_10_0:
      name: "Particulate Matter <10.0µm Concentration"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
esphome logs sensor1.yaml
INFO Reading configuration sensor1.yaml...
INFO Starting log output from aq-sensor-1.local using esphome API
WARNING Error resolving IP address of aq-sensor-1.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -2] Name or service not known). Re-Trying in 1 seconds
WARNING Error resolving IP address of aq-sensor-1.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -2] Name or service not known). Re-Trying in 1 seconds
```

### Additional information

It does appear to fire up at random, but its very inconsistent

insanemal, 2022-01-05:
This is still an issue. mDNS just stops responding

otzen, 2022-12-28:
I see the same problem.
Getting a log (quoted further down) from the ESP using the serial port (USB) i see that, it has serious problems in connecting to wifi. Also I can see (not in the quoted log) that the wifi drops out again, and then there is several attempts (like below) to reconnect.

I have two access points, and so far I see this only on devices connecting to one of them, this AP is also fairly old (D-link DIR-365), so I suspect the AP to be the root cause. And plan to replace that.

If i Can get further logs to support this bug please let me know.

> [13:48:29][I][wifi:257]: WiFi Connecting to 'wifiname'...
[13:48:29][W][wifi_esp8266:212]: wifi_apply_hostname_(vtermostat): lwIP error -16 on interface st (index 0)
[13:48:30][W][wifi_esp8266:482]: Event: Disconnected ssid='wifiname' bssid=00:18:xx:xx:xx:xx reason='Association Failed'
[13:48:30][W][wifi:549]: Error while connecting to network.
[13:48:30][W][wifi:585]: Restarting WiFi adapter...
[13:48:30][I][wifi:257]: WiFi Connecting to 'wifiname'...
[13:48:30][W][wifi_esp8266:212]: wifi_apply_hostname_(vtermostat): lwIP error -16 on interface st (index 0)
[13:48:32][W][wifi_esp8266:482]: Event: Disconnected ssid='wifiname' bssid=00:18:xx:xx:xx:xx reason='Association Failed'
[13:48:32][W][wifi:549]: Error while connecting to network.
[13:48:32][W][wifi:585]: Restarting WiFi adapter...
[13:48:37][D][wifi:386]: Starting scan...
[13:48:43][D][wifi:401]: Found networks:
[13:48:43][I][wifi:444]: - 'wifiname' (00:18:xx:xx:xx:xx) ▂▄▆█
[13:48:43][D][wifi:446]:     Channel: 13
[13:48:43][D][wifi:447]:     RSSI: -61 dB
[13:48:43][D][wifi:449]: - 'wifiname_B' (10:6F:xx:xx:xx:xx) ▂▄▆█
[13:48:43][I][wifi:257]: WiFi Connecting to 'wifiname'...
[13:48:43][W][wifi_esp8266:212]: wifi_apply_hostname_(vtermostat): lwIP error -16 on interface st (index 0)
[13:48:44][W][wifi_esp8266:482]: Event: Disconnected ssid='wifiname' bssid=00:18:xx:xx:xx:xx reason='Association Failed'
[13:48:44][W][wifi:549]: Error while connecting to network.
[13:48:44][W][wifi:585]: Restarting WiFi adapter...
[13:48:44][I][wifi:257]: WiFi Connecting to 'wifiname'...
[13:48:44][W][wifi_esp8266:212]: wifi_apply_hostname_(vtermostat): lwIP error -16 on interface st (index 0)
[13:48:45][W][wifi_esp8266:482]: Event: Disconnected ssid='wifiname' bssid=00:18:xx:xx:xx:xx reason='Association Failed'
[13:48:45][W][wifi:549]: Error while connecting to network.
[13:48:45][W][wifi:585]: Restarting WiFi adapter...
[13:48:51][D][wifi:386]: Starting scan...
[13:48:56][D][wifi:401]: Found networks:
[13:48:56][I][wifi:444]: - 'wifiname' (00:18:xx:xx:xx:xx) ▂▄▆█
[13:48:56][D][wifi:446]:     Channel: 13
[13:48:56][D][wifi:447]:     RSSI: -55 dB
[13:48:56][D][wifi:449]: - 'wifiname_B' (10:6F:xx:xx:xx:xx) ▂▄▆█
[13:48:56][I][wifi:257]: WiFi Connecting to 'wifiname'...
[13:48:56][W][wifi_esp8266:212]: wifi_apply_hostname_(vtermostat): lwIP error -16 on interface st (index 0)
[13:48:58][W][wifi_esp8266:482]: Event: Disconnected ssid='wifiname' bssid=00:18:xx:xx:xx:xx reason='Association Failed'
[13:48:58][W][wifi:549]: Error while connecting to network.
[13:48:58][W][wifi:585]: Restarting WiFi adapter...
[13:48:58][I][wifi:257]: WiFi Connecting to 'wifiname'...
[13:48:58][W][wifi_esp8266:212]: wifi_apply_hostname_(vtermostat): lwIP error -16 on interface st (index 0)
[13:48:59][W][wifi_esp8266:482]: Event: Disconnected ssid='wifiname' bssid=00:18:xx:xx:xx:xx reason='Association Failed'
[13:48:59][W][wifi:549]: Error while connecting to network.
[13:48:59][W][wifi:585]: Restarting WiFi adapter...
[13:49:02][I][wifi:141]: Starting fallback AP!
[13:49:02][C][wifi:210]: Setting up AP...
[13:49:02][C][wifi:212]:   AP SSID: 'VTermostat'
[13:49:02][C][wifi:213]:   AP Password: 'PanikPanikPanik'
[13:49:03][C][wifi:222]:   IP Address: 192.168.4.1
[13:49:04][D][wifi:386]: Starting scan...
[13:49:10][D][wifi:401]: Found networks:
[13:49:10][I][wifi:444]: - 'wifiname' (00:18:xx:xx:xx:xx) ▂▄▆█
[13:49:10][D][wifi:446]:     Channel: 13
[13:49:10][D][wifi:447]:     RSSI: -57 dB
[13:49:10][D][wifi:449]: - 'wifiname_B' (10:6F:xx:xx:xx:xx) ▂▄▆█
[13:49:10][I][wifi:257]: WiFi Connecting to 'wifiname'...
[13:49:10][W][wifi_esp8266:212]: wifi_apply_hostname_(vtermostat): lwIP error -16 on interface st (index 0)
[13:49:12][I][wifi:518]: WiFi Connected!
[13:49:12][C][wifi:362]:   Local MAC: A0:20:xx:xx:xx:xx
[13:49:12][C][wifi:363]:   SSID: 'wifiname'
[13:49:12][C][wifi:364]:   IP Address: 192.168.XX.129
[13:49:12][C][wifi:365]:   BSSID: 00:18:xx:xx:xx:xx
[13:49:12][C][wifi:367]:   Hostname: 'vtermostat'
[13:49:12][C][wifi:369]:   Signal strength: -64 dB ▂▄▆█
[13:49:12][C][wifi:373]:   Channel: 13
[13:49:12][C][wifi:374]:   Subnet: 255.255.255.0
[13:49:12][C][wifi:375]:   Gateway: 192.168.XX.1
[13:49:12][C][wifi:376]:   DNS1: XX.XX.XX.37
[13:49:12][C][wifi:377]:   DNS2: XX.XX.XX.176
[13:49:12][D][wifi:527]: Disabling AP...
[13:49:14][D][api:102]: Accepted 192.168.XX.197
[13:49:14][D][api.connection:918]: Home Assistant 2022.12.7 (192.168.XX.197): Connected successfully

insanemal, 2022-12-29:
Interestingly my WiFi routers are very new Asus models. So I'm unconvinced it's a router issue

closed 2023-11-12 by jesserockz