esphome/issues #2380: snapshot_exclude' is deprecated, 'backup_exclude' should be used instead. Please report this to the maintainer of ESPHome

paarak, 2021-09-02:
### The problem

Fresh install Home Assistant OS 6.2, core-2021.8.8, supervisor-2021.08.1.

appears in supervisor log

snapshot_exclude' is deprecated, 'backup_exclude' should be used instead. Please report this to the maintainer of ESPHome

### Which version of ESPHome has the issue?

0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant OS 6.2

### What platform are you using?

ESP8266

jesserockz, 2021-09-08:
This is fixed already but stable and beta will only update when they are next released

https://github.com/esphome/hassio/pull/40

OttoWinter, 2021-09-22:
In stable now

closed 2021-09-22 by OttoWinter