esphome/issues #2381: ESPHome Dev edit not saving yaml file.

markusg1234, 2021-09-02:
### The problem

When saving an edited yaml it displays that it was saved but it doesn't update the yaml file.

Going back into edit the change hasn't been saved

This is working fine with Beta but not working with Dev.

Error seen in log.

### Which version of ESPHome has the issue?

2021.9.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.0

### What platform are you using?

ESP32

### Board

Any board

### Component causing the issue

Editing

### Anything in the logs that might be useful for us?

```txt
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 
-----------------------------------------------------------
 Add-on: ESPHome (dev)
 Development version of ESPHome add-on
-----------------------------------------------------------
 Add-on version: dev
 You are running the latest version of this add-on.
 System: Home Assistant OS 6.2  (amd64 / generic-x86-64)
 Home Assistant Core: 2021.9.0
 Home Assistant Supervisor: 2021.08.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-esphome.sh: executing... 
[cont-init.d] 30-esphome.sh: exited 0.
[cont-init.d] 40-migrate.sh: executing... 
[cont-init.d] 40-migrate.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[21:09:17] INFO: Waiting for dashboard to come up...
[21:09:17] INFO: Starting ESPHome dashboard...
WARNING Calling ESPHome with the configuration before the command is deprecated and will be removed in the future. 
WARNING Please instead use:
WARNING    esphome dashboard /config/esphome --socket /var/run/esphome.sock --hassio
INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[21:09:18] INFO: Starting NGINX...
INFO 304 GET / (0.0.0.0) 21.66ms
INFO 200 GET /static/fonts/material-icons/material-icons.css?hash=ead4c276 (0.0.0.0) 3.89ms
INFO 200 GET /static/css/vendor/materialize/materialize.min.css?hash=ec1df3ba (0.0.0.0) 2.25ms
INFO 200 GET /static/js/vendor/jquery-ui/jquery-ui.min.js?hash=cab3392d (0.0.0.0) 3.55ms
INFO 200 GET /static/css/esphome.css?hash=2ea4c658 (0.0.0.0) 3.75ms
INFO 200 GET /static/js/vendor/jquery/jquery.min.js?hash=12108007 (0.0.0.0) 1.80ms
INFO 200 GET /static/js/vendor/materialize/materialize.min.js?hash=4be20daf (0.0.0.0) 2.38ms
INFO 200 GET /static/js/vendor/jquery-validate/jquery.validate.min.js?hash=aad785d4 (0.0.0.0) 1.42ms
INFO 200 GET /static/js/vendor/ace/ace.js?hash=65322f49 (0.0.0.0) 7.81ms
INFO 200 GET /static/js/esphome/index-aa9e8c11.js (0.0.0.0) 3.42ms
INFO 200 GET /static/fonts/material-icons/MaterialIcons-Regular.woff2 (0.0.0.0) 2.52ms
INFO 200 GET /static/js/vendor/ace/theme-dreamweaver.js (0.0.0.0) 3.09ms
INFO 200 GET /static/js/vendor/ace/mode-yaml.js (0.0.0.0) 2.18ms
INFO 101 GET /ace (0.0.0.0) 1.58ms
INFO Running command 'esphome --dashboard -q vscode --ace /config/esphome'
INFO 200 GET /static/js/esphome/c.9481058d.js (0.0.0.0) 1.28ms
INFO 200 GET /static/js/esphome/c.2c9aeedd.js (0.0.0.0) 2.17ms
INFO 200 GET /static/js/esphome/c.d1723f10.js (0.0.0.0) 1.29ms
INFO 200 GET /static/js/esphome/c.d32d4248.js (0.0.0.0) 1.21ms
INFO 200 GET /static/js/esphome/c.8551ca08.js (0.0.0.0) 1.58ms
INFO 200 GET /static/js/esphome/c.505d82a4.js (0.0.0.0) 0.91ms
INFO 200 GET /edit?configuration=wrover.yaml (0.0.0.0) 0.81ms
ERROR Uncaught exception POST /edit?configuration=wrover.yaml (0.0.0.0)
HTTPServerRequest(protocol='http', host='homeassistant.local:8123', method='POST', uri='/edit?configuration=wrover.yaml', version='HTTP/1.1', remote_ip='0.0.0.0')
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/tornado/web.py", line 1702, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/usr/local/lib/python3.7/dist-packages/esphome/dashboard/dashboard.py", line 134, in decorator
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/esphome/dashboard/dashboard.py", line 162, in decorator
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/esphome/dashboard/dashboard.py", line 612, in post
    file=settings.rel_path(configuration), mode="wb", encoding="utf-8"
ValueError: binary mode doesn't take an encoding argument
ERROR 500 POST /edit?configuration=wrover.yaml (0.0.0.0) 1.49ms
INFO 200 GET /static/js/esphome/c.04e11fe1.js (0.0.0.0) 1.10ms
INFO 200 GET /static/js/esphome/c.38391347.js (0.0.0.0) 2.36ms
```

oxan, 2021-09-02:
This is a result of esphome/esphome#2226, there should be no `encoding` here: https://github.com/esphome/esphome/commit/54337befc262f7935b61c2b9f970db0c66c8d74f#diff-a5a4ec3df8dfc1fde32a73116406fad0c54684124e801140b4590dba8c76ac2aR612
@jesserockz

jesserockz, 2021-09-05:
Woops, sorry =)

closed 2021-09-05 by jesserockz