esphome/issues #2382: Air Quality Sensor overview show wrong sensor for PM1006

morganrallen, 2021-09-02:
### The problem

When looking at the overview of air quality sensors, the PM1006 thumbnail shows a different and unsupported air quality sensor, the Sharp GP2Y10. And while I understand this is in fact the sensor it uses internally, there is no compatibility between the PM1006 and GP2Y10, making the use of this preview image very confusing.

### Which version of ESPHome has the issue?

2021.8.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

sjtrny, 2021-09-03:
> the PM1006 thumbnail shows a different and unsupported air quality sensor, the Sharp GP2Y10

To me it looks like the correct sensor shown in the PM1006 data sheet (screenshot below). The Sharp GP2Y10 has a similar form factor but has other markings/holes.

[image]

morganrallen, 2021-09-03:
Wow yup, that's it, I couldn't find a picture of it and though PM1006 was referring to the driver board, interfacing with the Sharp sensor. Now if I could find these in bulk, would take UART over the analog GP2Y10 any day...

closed 2021-09-03 by morganrallen