esphome/issues #2384: CVE-2021-28139

RubenKelevra, 2021-09-03:
### The problem

I was wondering if CVE-2021-28139 is already patched or when a patch is expected to be available?

Reference: https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2021-28139 
News: https://amp.thehackernews.com/thn/2021/09/new-braktooth-flaws-leave-millions-of.html#aoh=16306671538768&csi=0

### Which version of ESPHome has the issue?

yes

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

yes

### What platform are you using?

ESP32

### Component causing the issue

Bluetooth library

oxan, 2021-09-23:
Unfortunately, for the moment, we don't use the IDF from Espressif directly, but through the Arduino framework, which means we're dependent upon them merging the fixes before we can use them. Complicating matters somewhat is that ESPHome has been using the v1.0.x series of the Arduino framework, while Arduino has very recently released v2.0.0. This new major version has incompatibilities for which ESPHome will need to be changed, and history has shown us that new major versions often come with hard-to-debug regressions rendering devices or their connection unstable.

At the time of writing, ESPHome uses Arduino v1.0.6 by default, which seems to be vulnerable. There is no stable version of the Arduino v1.0.x series that uses a patched version of the IDF. Arduino v2.0.0 has the patched IDF, but isn't usable with ESPHome yet. (All based on the patched versions referenced in the Espressif [advisory](https://www.espressif.com/sites/default/files/advisory_downloads/AR2021-004%20Bluetooth%20Security%20Advisory.pdf)).

That said, there is a mitigation available. The `idf-release/v3.3` [development branch](https://github.com/espressif/arduino-esp32/commits/idf-release/v3.3) of Arduino contains v1.0.6 (the current default) with a patched IDF (since commit [1a49760e](https://github.com/espressif/arduino-esp32/commit/1a49760eb671229da4728b64ebb9dffc66d501cd). You can use this branch with ESPHome by putting this in your YAML:
```yaml
esphome:
  platformio_options:
    platform_packages:
      framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#idf-release/v3.3
```

Note that this uses a development branch, so occasionally things might break. You can also pin the first fixed commit in that branch by using this:
```yaml
esphome:
  platformio_options:
    platform_packages:
      framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#1a49760eb671
```

Hedda, 2021-09-24:
I understand that Tasmota is/was in a similar situation and has now upgraded to Arduino v1.0.7.4 which suppose to have the fix too?

https://github.com/arendst/Tasmota/discussions/13120

https://github.com/arendst/Tasmota/pull/13154

Tasmota core developer first wrote this Jason2866
> Tasmota do not use BLE mesh. Anyway the used Tasmota Arduino core 2.0.0 (IDF 4.4 branch S2 and C3 build) has the [espressif/esp-idf@d4232ee](https://github.com/espressif/esp-idf/commit/d4232ee8f95a223557e58144cf124fce3280d3ea)
> commit included.
> I am considering to build a new Tasmota Arduino Core 1.0.7.x based on actual IDF 3.3 (branch) which includes [espressif/esp-idf@35bbd1b](https://github.com/espressif/esp-idf/commit/35bbd1ba2630c9413386aab456dbdf511e6790ce)

and then he made PR with Arduino v1.0.7.4 

> PR #13154 includes [espressif/esp-idf@35bbd1b](https://github.com/espressif/esp-idf/commit/35bbd1ba2630c9413386aab456dbdf511e6790ce)

oxan, 2021-09-24:
> Arduino v1.0.7.4

That release does not exist upstream. Tasmota uses their own fork of the Arduino core.