esphome/issues #2387: Missing server port drop down button

Johnno88, 2021-09-04:
### The problem

ESPHome 2021.8.2
Core 2021.9.2

I am trying to install a yaml file to a new esp32,
and there is no drop down  available to select port... 
[image]

As I am using a Mac to interface with my Homeassistant running on a Pi, I have no idea how to manual flash to the esp32

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.2

### What platform are you using?

ESP32

### Board

N/A

### Component causing the issue

N/A

### Anything in the logs that might be useful for us?

```txt
N/A
```

### Additional information

I have tried multiple boards, an the drop down window does not show.

I can upload to pervious esp32 boards via wifi, no issues

it is only trying to flash a new boards that I need to choose a USB port but the re is no drop down box

Johnno88, 2021-09-05:
so, thanks for the help..
after a frustrating day, I have found that the cable I had was a power supply cable, not a data/power cable.

also, that all the information supplied by ESPHome regarding this issue is out of date..... their website and the information on GitHub is also out of date..

The simply answer for anyone else having issues is, once I had a data cable, the server port selection appears automatically when the upload via cable form the connected server is selected...   This would have been helpful if this update was also included in the esphome documentation

oxan, 2021-09-06:
It would be helpful if you could provide a link to the documentation that is out of date :)

closed 2022-01-20 by github-actions[bot]