esphome/issues #2388: WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received

joe-sydney, 2021-09-04:
### The problem

Note this is may be related to the CWWW bug which I reported and got fixed (in that the lights are discovered by HA, and working).

I notice that my HA log is getting a lot of warnings of invalid color mode:
WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received

When I turn on debug for the mqtt component, I found 2 of the above warnings is preceded by 1 of following debug logs:

DEBUG (MainThread) [homeassistant.components.mqtt] Received message on my_light/light/my_light/state (retained): b'{"color_mode":"cwww","state":"OFF","brightness":255,"color":{"c":255,"w":69}}'

This does not seem to be affecting the operation of the light, but thought perhaps there is something that needs cleaning up.

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.8.8

### What platform are you using?

ESP8266

### Component causing the issue

mqtt, light

### Example YAML snippet

```yaml
output:
  - platform: esp8266_pwm
    pin: 5
    id: output_warmwhite
  - platform: esp8266_pwm
    pin: 4
    id: output_coldwhite

light:
  - platform: cwww
    name: "${name}"
    id: light2
    cold_white: output_coldwhite
    warm_white: output_warmwhite
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
```

### Anything in the logs that might be useful for us?

```txt
DEBUG (MainThread) [homeassistant.components.mqtt] Received message on my_light/light/my_light/state (retained): b'{"color_mode":"cwww","state":"OFF","brightness":255,"color":{"c":255,"w":69}}'
WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received
WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received
```

jesserockz, 2021-09-05:
Can you try the dev version of ESPHome on that light? https://github.com/esphome/esphome/pull/2231 made some changes for mqtt lights

joe-sydney, 2021-09-05:
> ~Can you try the dev version of ESPHome on that light? [esphome/esphome#2231](https://github.com/esphome/esphome/pull/2231) made some changes for mqtt lights~
> 
> Nevermind that, that was a different thing, yeah there is still a bug.

Yes, confirmed on current dev version.

oxan, 2021-09-06:
Can you try esphome/esphome#2247 ?

joe-sydney, 2021-09-06:
> Can you try [esphome/esphome#2247](https://github.com/esphome/esphome/pull/2247) ?

Unfortunately, that didn't help. The following are debug logs showing a light (my_light_A) which had this problem and a light (my_light_B) that didn't have this problem as it was on 1.20.4.

You can also see that the warning has changed from "Invalid color mode received" (per my initial report) to "Invalid or incomplete color value received".

Note however, I can still change the brightness and color temperature via HA.

--- my_light_A ---
DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on my_light_A/light/my_light_A/command: '{"state": "ON", "brightness": 232, "color_temp": 195}', mid: 697
DEBUG (MainThread) [homeassistant.components.mqtt] Received message on my_light_A/light/my_light_A/state: b'{"color_mode":"color_temp","state":"ON","brightness":232,"color":{"c":255,"w":59}}'
WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value received
WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value received

--- my_light_B ---
DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on my_light_B/light/my_light_B/command: '{"state": "ON", "brightness": 255, "color_temp": 200}', mid: 707
DEBUG (MainThread) [homeassistant.components.mqtt] Received message on my_light_B/light/my_light_B/state: b'{"state":"ON","brightness":255,"color_temp":200}'

joe-sydney, 2021-09-15:
I applied the changes on top of 2021.9.0.

The light does shows up on HA with light entity, but it shows as unavailable. Hence, I can't control it via HA and look out for any warning messages. (I can however control it via its web_server, and when I turn the light on and off, but this has nothing to do with HA integration).

This is what HA sees:

 INFO (MainThread) [homeassistant.components.mqtt.mixins] Got update for entity with hash: ('light', 'dicky_lamp dicky_lamp') '{'schema': 'json', 'color_mode': True, 'supported_color_modes': ['color_temp'], 'brightness': True, 'name': 'Dicky lamp', 'state_topic': 'dicky_lamp/light/dicky_lamp/state', 'command_topic': 'dicky_lamp/light/dicky_lamp/command', 'availability_topic': 'dicky_lamp/status', 'unique_id': 'ESPlightdicky_lamp', 'device': {'identifiers': 'ABCDEFABCDEF', 'name': 'dicky_lamp', 'sw_version': 'esphome v2021.9.0 Sep 16 2021, 08:15:05', 'model': 'esp01_1m', 'manufacturer': 'espressif'}, 'platform': 'mqtt'}'

joe-sydney, 2021-09-19:
By the way, I have used this "opportunity" to move to native API, so I don't personally have this issue anymore :)

closed 2022-09-03 by joe-sydney

derandiunddasbo, 2022-09-30:
I don't how has this issue been resolved, I still get the same warnings in the log file. Does this get fixed in any way or did the ticket opener just close the ticket because he doesn't use mqtt anymore`for his esphome devices?