esphome/issues #2390: BLE_CLIENT: Issue Decoding Multiple Characteristics

mwolter805, 2021-09-04:
### The problem

When decoding BLE from a Weber iGrill2, only the first temperature probe is reported. The remaining three probes and the battery are not available (entities are unknown in HA). 

[image]

### Which version of ESPHome has the issue?

2021.8.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.8.6

### What platform are you using?

ESP32

### Board

Olimex ESP32-POE

### Component causing the issue

BLE_CLIENT

### Example YAML snippet

```yaml
substitutions:
  esp32_name: ble-test
  
  prefix: ble_test
  friendly_name: BLE Test
  
  # Weber iGrill2
  igrill_friendly_name: Weber iGrill 2

esphome:
  name: ${esp32_name}
  platform: ESP32
  board: esp32-poe

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12
  use_address: 192.168.6.10
  enable_mdns: false

# Enable logging
logger:
  # level: VERY_VERBOSE
  level: debug
  logs:
    esp32_ble_tracker: error

# Enable Home Assistant API
api:
  password: "pass"

# Enable over-the-air flashing
ota:
  password: "pass"

time:
  - platform: homeassistant

esp32_ble_tracker:

ble_client:
  - mac_address: 70:91:8F:18:EA:E8
    id: weber_igrill2

switch:
  - platform: restart
    name: "${friendly_name} Restart"
    id: ${prefix}_restart_switch

sensor:
  - platform: uptime
    name: "${friendly_name} Uptime Sensor"
    id: ${prefix}_uptime
    

# Weber iGrill2    

  # First Temperature Probe
  - platform: ble_client
    ble_client_id: weber_igrill2
    id: weber_igrill2_p1
    name: "${igrill_friendly_name} Probe 1"
    service_uuid: 'ada7590f-2e6d-469e-8f7B-1822b386a5e9'
    characteristic_uuid: '06ef0002-2e06-4b79-9e33-fce2c42805ec'
    notify: true
    accuracy_decimals: 0
    unit_of_measurement: '°F'
    device_class: temperature
    lambda: |-
      return (float)x[0];

  # Second Temperature Probe
  - platform: ble_client
    ble_client_id: weber_igrill2
    id: weber_igrill2_p2
    name: "${igrill_friendly_name} Probe 2"
    service_uuid: 'ada7590f-2e6d-469e-8f7B-1822b386a5e9'
    characteristic_uuid: '06ef0004-2e06-4b79-9e33-fce2c42805ec'
    notify: false
    accuracy_decimals: 0
    unit_of_measurement: '°F'
    device_class: temperature
    lambda: |-
      return (float)x[0];
      
  # Third Temperature Probe
  - platform: ble_client
    ble_client_id: weber_igrill2
    id: weber_igrill2_p3
    name: "${igrill_friendly_name} Probe 3"
    service_uuid: 'ADA7590F-2E6D-469E-8F7B-1822B386A5E9'
    characteristic_uuid: '06EF0006-2E06-4B79-9E33-FCE2C42805EC'
    notify: false
    accuracy_decimals: 0
    unit_of_measurement: '°F'
    device_class: temperature

  # Fourth Temperature Probe
  - platform: ble_client
    ble_client_id: weber_igrill2
    id: weber_igrill2_p4
    name: "${igrill_friendly_name} Probe 4"
    service_uuid: 'ADA7590F-2E6D-469E-8F7B-1822B386A5E9'
    characteristic_uuid: '06EF0008-2E06-4B79-9E33-FCE2C42805EC'
    notify: false
    accuracy_decimals: 0
    unit_of_measurement: '°F'
    device_class: temperature
  
  # Battery    
  - platform: ble_client
    ble_client_id: weber_igrill2
    id: weber_igrill2_b
    name: "${igrill_friendly_name} Battery"
    service_uuid: '180F'
    characteristic_uuid: '2A19'
    notify: false
    icon: 'mdi:battery'
    accuracy_decimals: 0
    unit_of_measurement: '%'
    device_class: battery
#    update_interval: 10min
    lambda: |-
      return (float)x[0];
```

### Anything in the logs that might be useful for us?

```txt
There are only a few errors in the logs.
[W][ble_sensor:082]: Error reading char at handle 21, status=15
[W][ble_sensor:082]: Error reading char at handle 37, status=15
[W][ble_sensor:082]: Error reading char at handle 41, status=15
[W][ble_sensor:082]: Error reading char at handle 45, status=15
[W][ble_sensor:082]: Error reading char at handle 93, status=15
```

### Additional information

Wifi is not enabled as this board has PoE. Verified the board has not run out of heap. BLE does appear to be working properly with the board as the xiaomi_lywsd03mmc platform reports values correctly. 

Complete Logs at Very Verbose
[Olimex ESP32-POE iGrill2 Very Verbose 210904.txt.zip](https://github.com/esphome/issues/files/7110046/Olimex.ESP32-POE.iGrill2.Very.Verbose.210904.txt.zip)

buxtronix, 2021-09-15:
It looks like some security authentication is required before reading these values. The working sketch seems to implement this:

https://github.com/1mckenna/esp32_iGrill/blob/main/esp32_iGrill.ino#L471

It seems to have both BLE pairing as well as some custom challenge protocol, which would need to be implemented in a custom component.

mwolter805, 2021-09-15:
Thanks for looking into this. Seems like most of the code can be used to make a custom component. I've made several Arduino projects but never an ESPHome component. Are there documents I can follow or a project you'd recommend as a template or guide?

buxtronix, 2021-09-16:
This is a good external BLE component example that can show what's required:

https://github.com/j5lien/esphome-idasen-desk-controller

closed 2021-09-16 by mwolter805

anthonyangel, 2021-12-16:
@mwolter805 Did you manage to get this working? I previously tried this, but couldn't get any of the temperature sensors to return data.

https://github.com/esphome/feature-requests/issues/792 for reference

mwolter805, 2021-12-16:
Currently do not have this working with ESPHome. The ability to write BLE commands was recently added (or will be shortly). This should help greatly.