esphome/issues #2391: Fanspeed deprecated warnings on Dev

markusg1234, 2021-09-06:
### The problem

Have noticed warnings of deprecated fanspeed even though I haven't programed a fan speed.
Only using a binary fan.

Also have noticed for a while now incompatable  pointer types as seen in log below.

### Which version of ESPHome has the issue?

V2021.9.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.3

### What platform are you using?

ESP32

### Board

multiple

### Component causing the issue

fan

### Example YAML snippet

```yaml
esphome:
  name: test
  platform: ESP32
  board: esp-wrover-kit
  
output:
  - id: relay_1
    platform: gpio
    pin: GPIO5
    inverted: true

fan:
   - platform: binary
     output: relay_1
     name: "test"
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/test.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting /data/test/.pioenvs
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/test
Processing test (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Dependency Graph
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
[... 146 lines cut ...]
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling /data/test/.pioenvs/test/FrameworkArduino/esp32-hal-timer.c.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/esp32-hal-touch.c.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/esp32-hal-uart.c.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/libb64/cdecode.c.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/libb64/cencode.c.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/main.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/stdlib_noniso.c.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/wiring_pulse.c.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/wiring_shift.c.o
Archiving /data/test/.pioenvs/test/libFrameworkArduino.a
Linking /data/test/.pioenvs/test/firmware.elf
Retrieving maximum program size /data/test/.pioenvs/test/firmware.elf
Checking size /data/test/.pioenvs/test/firmware.elf
RAM:   [          ]   4.3% (used 14128 bytes from 327680 bytes)
Flash: [==        ]  15.5% (used 284842 bytes from 1835008 bytes)
Building /data/test/.pioenvs/test/firmware.bin
========================= [SUCCESS] Took 8.60 seconds =========================
INFO Successfully compiled program.
```

krom, 2021-09-06:
It was announced with last HA release https://www.home-assistant.io/blog/2021/09/01/release-20219/#breaking-changes.

oxan, 2021-09-11:
> It was announced with last HA release https://www.home-assistant.io/blog/2021/09/01/release-20219/#breaking-changes.

This is actually a different issue, ESPHome still contains legacy code to handle the old three-speed interface, and that code causes warnings during a default build; even if it isn't used.

closed 2021-09-13 by OttoWinter