esphome/issues #2395: Unable to find filter with the name 'min' [or 'max']

to0kan, 2021-09-07:
### The problem

Unable to apply min/max filters when using the ESPHome Editor in Home Assistant. 

Error code: 'Unable to find filter with the name [...]'.
[image]

The issue persists regardless of whether I use the filter on it's own or follow it up with arguments like send_every or window_size.

When 'min' or 'max' in the code is replaced with another filter, e.g. 'median', the error drops off.

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.4.6

### What platform are you using?

ESP8266

### Board

Wemos D1 Pro / Wemos D1 Mini

### Component causing the issue

Filters

### Example YAML snippet

```yaml
esphome:
  name: solar_voltage_v2
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: *
  password: *

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: *
    password: *

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

sensor:
  - platform: adc
    pin: A0
    name: "Solar Battery Voltage"
    update_interval: 3s
    filters:
      - multiply: 22.0
      - min:
          window_size: 5
          send_every: 5
          send_first_at: 5
```

oxan, 2021-09-07:
It compiles fine for me...

oxan, 2021-09-11:
I cannot reproduce this... Are you sure you're on 2021.8.2? The min/max filters were added later than the older filters which are working for you.

closed 2021-09-11 by oxan

to0kan, 2021-09-12:
@oxan you are absolutely correct, I was on 1.16.2 dead certain I had that addon on auto-update. Thanks for pointing me in the right direction!