esphome/issues #2396: Display ST7735 not working with esphome

pioneershahid, 2021-09-07:
### The problem

Hi, 
this is my first ever post on GitHub so please bear with me.

I am trying to use the ST7735 display bought from Aliexpress. here is the [link](https://www.aliexpress.com/item/32927602923.html?spm=a2g0s.9042311.0.0.27424c4dL2OaBG) of the product. 

I can successfully test it using Adafruit default graphictest on Nodemcu using the following pinout. No issue.
  #define TFT_CS         D1
  #define TFT_RST        D4                                            
  #define TFT_DC         D2 
  #define TFT_MOSI     D7
  #define TFT_SCLK      D5

when I try to use the esphome ver 2021.8.2 on nodmcu using the code below there is nothing coming up on the screen. The display stays white.  

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.3

### What platform are you using?

ESP8266

### Board

nodmcu

### Component causing the issue

st7735

### Example YAML snippet

```yaml
font:
  - file: "webfont.ttf"
    id: my_font
    size: 20
 
spi:
  clk_pin: D5
  mosi_pin: D7

display:
  - platform: st7735
    model: "INITR_18BLACKTAB"
    reset_pin: D4
    cs_pin: D1
    dc_pin: D2
    rotation: 0
    device_width: 128
    device_height: 160
    col_start: 0
    row_start: 0
    eight_bit_color: true
    update_interval: 5s
    lambda: |-
      it.print(10, 10, id(my_font), "HELLO!");
```

### Additional information

can anyone please advise if there is something wrong in the code or an error in esphome settings? any pointer would be greatly appreciated with thanks.

WeekendWarrior1, 2021-09-13:
Weird ask, but can you try swapping the mosi_pin for a miso_pin and let me know if it works?

```yaml
spi:
  clk_pin: D5
  miso_pin: D7
```

pioneershahid, 2021-09-13:
> Weird ask, but can you try swapping the mosi_pin for a miso_pin and let me know if it works?
> 
> ```yaml
> spi:
>   clk_pin: D5
>   miso_pin: D7
> ```

Hi, same result. no display on the screen. wifi blue lights turn on and blink after few seconds.

WeekendWarrior1, 2021-09-14:
I tried your config with a ST7735 display I had lying around, and it worked fine for me.
Only difference was I used a different font.
Have you checked your wiring?

pioneershahid, 2021-09-16:
@WeekendWarrior1 Thanks for looking into it. my wiring was correct as it's the same wiring I was using to test via Adafruit default graphictest. when I changed the font to Comic Sans MS.ttf, it worked. looks like an issue with the fonts I was originally using. thanks once again for all your help.

closed 2021-09-16 by pioneershahid