esphome/issues #2398: How to update Development version of ESPHome add-on in Home Assistant

WIsbrecht, 2021-09-08:
### The problem

I've been using the ESPHome (dev) add-on in Home Assistant for some time now.
It used to be, that when I stopped and restarted the add-on in Home Assistant, the add-on automatically downloaded the newest version from GitHub. It reported this in the add-on log.
For some time now the add-on starts way quicker than before but doesn't update to the latest version from GitHub.
The only way I found to update the ESPHome (dev) add-on in Home Assistant is to uninstall and re-install the add-on.

Have there been any changes regarding this? If so, how do I update the add-on without reinstalling it?

### Which version of ESPHome has the issue?

v2021.9.0-dev 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.9.4

### What platform are you using?

ESP8266

### Component causing the issue

ESPHome (dev) add-on in Home Assistant

jesserockz, 2021-09-08:
Yes, because of the removal of esphome_version. You now have to rebuild it then restart it.

WIsbrecht, 2021-09-08:
Do you mean the "REBUILD" button on the right?
[image]

closed 2021-09-09 by WIsbrecht