esphome/issues #2400: Update via USB cable doesn't work after OS/Hassio update

Spirituss, 2021-09-08:
### The problem

Firmware successfully compiles, esphome sees USB cable but bin uploading can't be done following by "Connecting....____.... " and finally by error "A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header".
Nevertheless, the blue status led on ESP board is blinking during upload retries.
I tried to reinstall esphome docket container (both Hassio Addon and docker container versions), to fulfil esphome site recommendations with no effect.
This host worked successfully for a long time and becomes broken after host update my Raspbian OS by "apt-get update + upgrade".
Googling leads to the similar issue:
 https://github.com/Koenkk/zigbee2mqtt/issues/8663

### Which version of ESPHome has the issue?

v2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.3

### What platform are you using?

ESP8266

### Board

Wemos D1 mini, generic Nodemcu

### Component causing the issue

Possibly, USB driver

### Anything in the logs that might be useful for us?

```txt
NFO Running:  esptool.py --before default_reset --after hard_reset --baud 460800 --chip esp8266 --port /dev/ttyUSB1 write_flash 0x0 /data/test/.pioenvs/test/firmware.bin
esptool.py v2.8
Serial port /dev/ttyUSB1
Connecting........_____....._____....._____....._____....._____....._____....._____
A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header
```

Spirituss, 2021-09-24:
Critical issue but there is no any reaction. I am confused.

trave, 2021-10-07:
May be related to https://github.com/espressif/esptool/issues/656
Can you check manually fix with `--before no_reset` option for esptool?

Spirituss, 2021-10-07:
> May be related to [espressif/esptool#656](https://github.com/espressif/esptool/issues/656)
> Can you check manually fix with `--before no_reset` option for esptool?

I will check, thanx!

Spirituss, 2021-10-09:
Found issues with similar diagnosis
https://github.com/espressif/esptool/issues/653
https://bugzilla.kernel.org/show_bug.cgi?id=214131

InnerLive, 2021-11-01:
Same error. Tried with different ESP8266 boards.
HA 2021.10.6, ESPHOME 2021.10.3