esphome/issues #2402: TinyPico does not boot with default yaml

crlogic, 2021-09-08:
### The problem

Manually flashing ESPHome to a TinyPico does not boot.

Serial output:
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:248852
ets Jun  8 2016 00:22:57`

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.4

### What platform are you using?

ESP32

### Board

TinyPico

### Component causing the issue

N/A

### Example YAML snippet

```yaml
esphome:
  name: tinypico-outside
  platform: ESP32
  board: tinypico

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "longstring"

wifi:
  ssid: "wifi"
  password: "wifi secret"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Tinypico-Outside"
    password: "pass"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
Flashing process used:

esptool.py -p COM23 read_mac
esptool.py -p COM23 erase_flash
esptool.py --chip esp32 --port COM23 write_flash -fm dio -z 0x1000 tinypico-outside.bin
```
`-fm dio` used at the recommendation of UnExpectedMaker on Discord. The default QIO exhibited the same behavior.
Restoring to MicroPython v1.16 works fine.
```

oxan, 2021-09-09:
This might be caused by https://github.com/platformio/platform-espressif32/pull/615.

Can you try this?
```
esphome:
  platformio_options:
    board_build.flash_mode: dio
```

crlogic, 2021-09-09:
> This might be caused by platformio/platform-espressif32#615

Added to yaml. Built ok. ESPTool flashed ok. Failed to boot w/ the
following error :(

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:248820

closed 2022-01-23 by github-actions[bot]