esphome/issues #2403: 2021.8.2 seems to randomly set a color for a short while, then revert back to correct color

cooljimy84, 2021-09-09:
### The problem

The LED strips i have get set to a color, but then randomly (and nothing in Home Assistant or ESP logs shows) but the color changes. Sometimes it changes to a greeny color, other times a redy color.

I've had this on neopixel and on fastled_clockless

Roll back to 1.13.6 and everything behaves as expected.

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.8.8

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

light

### Example YAML snippet

```yaml
esphome:
  name: esp2
  platform: ESP8266
  board: nodemcuv2

#Our Bedroom

status_led:
  pin:
    number: D0
    inverted: True

wifi:
  ssid: 'WIFI'
  password: 'PASSWORD'
  power_save_mode: none
  fast_connect: true

mqtt:
  broker: 'IP'
  username: 'USERNAME'
  password: 'PASSWORD'
  reboot_timeout: 1min

# Enable logging
logger:
  level: INFO

ota:
  password: 'PASSWORD'

text_sensor:
  - platform: version
    name: "ESPHome Version ESP2"
    hide_timestamp: true

sensor:
  - platform: uptime
    name: ESP2 Uptime Sensor
    update_interval: 360s
  - platform: dht
    pin: D7
    temperature:
      name: "ESP2 temperature"
    humidity:
      name: "ESP2 humidity"
    model: DHT22
  - platform: adc
    pin: A0
    name: "ESP2 ldr"
    unit_of_measurement: lux
    update_interval: 30s
    filters:
      - lambda: >-
          return (x / 10000.0) * 2000000.0;
  - platform: wifi_signal
    name: "WiFi Signal ESP2"
    update_interval: 360s

binary_sensor:
  - platform: gpio
    pin: D5
    name: "ESP2 pir"
    device_class: motion
    filters:
      - delayed_on: 100ms
  - platform: gpio
    pin: D6
    name: "ESP2 touchpad"
    on_multi_click:
    - timing:
        - ON for at most 1s
        - OFF for at most 0.5s
        - ON for 0.1s to 1s
        - OFF for at least 0.2s
      then:
        - logger.log: "Double Click"
        - mqtt.publish:
            topic: esp2/binary_sensor/esp2_touchpad/state
            payload: "Double Click"
    - timing:
         - ON for 1s to 3s
         - OFF for at least 0.5s
      then:
        - logger.log: "Single Long Click"
        - mqtt.publish:
            topic: esp2/binary_sensor/esp2_touchpad/state
            payload: "Single Long Click"
    - timing:
         - ON for at most 0.5s
         - OFF for at least 0.5s
      then:
        - logger.log: "Single Short Click"
        - mqtt.publish:
            topic: esp2/binary_sensor/esp2_touchpad/state
            payload: "Single Short Click"

# warmwhite
light:
  - platform: neopixelbus
    type: GRBW
    pin: GPIO3
    num_leds: 24
    name: "LED Wheel"
    gamma_correct: 1.0
    effects:
      - addressable_flicker:
          name: Flicker Effect With Custom Values
          update_interval: 25ms
          intensity: 50%
      - strobe:
          name: Flash Red
          colors:
            - state: True
              brightness: 100%
              red: 100%
              green: 0%
              blue: 0%
              white: 0%
              duration: 500ms
            - state: True
              brightness: 1%
              red: 1%
              green: 0%
              blue: 0%
              white: 0%
              duration: 500ms
      - strobe:
          name: Flash Green
          colors:
            - state: True
              brightness: 100%
              red: 0%
              green: 100%
              blue: 0%
              white: 0%
              duration: 500ms
            - state: True
              brightness: 1%
              red: 0%
              green: 1%
              blue: 0%
              white: 0%
              duration: 500ms
      - strobe:
          name: Flash Blue
          colors:
            - state: True
              brightness: 100%
              red: 0%
              green: 0%
              blue: 100%
              white: 0%
              duration: 500ms
            - state: True
              brightness: 1%
              red: 0%
              green: 0%
              blue: 1%
              white: 0%
              duration: 500ms
      - random:
          name: Color Wipe
          transition_length: 0.1s
          update_interval: 0.5s
      - strobe:
          name: Flash RedYellow
          colors:
            - state: True
              brightness: 85%
              red: 100%
              green: 0%
              blue: 0%
              white: 0%
              duration: 500ms
            - state: True
              brightness: 85%
              red: 100%
              green: 100%
              blue: 0%
              white: 0%
              duration: 500ms
      - strobe:
          name: Flash RedBlue
          colors:
            - state: True
              brightness: 85%
              red: 100%
              green: 0%
              blue: 0%
              white: 0%
              duration: 500ms
            - state: True
              brightness: 85%
              red: 0%
              green: 0%
              blue: 100%
              white: 0%
              duration: 500ms
      - addressable_lambda:
          name: "Scroll Off"
          update_interval: 50ms
          lambda: |-
            for (int i = it.size() - 1; i > 0; i--) {
              it[i] = it[i - 1].get();
            }
            it[0] = Color::BLACK;

switch:
  - platform: restart
    name: "ESP2 Restart"
```

### Anything in the logs that might be useful for us?

```txt
log doesn't show the color change.
```

oxan, 2021-09-11:
Usually this is an electrical/power problem...

cooljimy84, 2021-09-13:
I would agree, however I have two strips (built at the same time) when i have 1.13.6 on both they work fine.
if i put 2021.8.2 on one the issues shows up on that one, if i put it back to 1.13.6 its fine again, if i put 2021.8.2 on the other one the fault show on that one (fault follows never firmware)

It might be that gamma is being miss calculated ? the only reason i say that is cause the lights are color set by Circadian, so they update to a "warmer" in the evenings. (it's a bit of a stab in the dark, but i can try and test further)

oxan, 2021-09-15:
I'll try to reproduce this on my dev strip tonight, but I've never seen this behaviour, so I'm not optimistic I'll be able to reproduce it...

Can you try to run with the log level set at `DEBUG` instead of `INFO`? Light changes are logged at debug level, so if you set it to info you'll miss all those messages. Furthermore, 1.13 is an awfully long way back, do you know if there's any more recent version that doesn't have this behaviour?

> It might be that gamma is being miss calculated ? the only reason i say that is cause the lights are color set by Circadian, so they update to a "warmer" in the evenings. (it's a bit of a stab in the dark, but i can try and test further)

I'm not familar with the circadian integration, but if it changes the color of the LED strip a message should be logged by the ESP (at debug level).

cooljimy84, 2021-10-10:
After a hard reboot of the device (unplugged and replugged) it seems fine.

closed 2021-10-10 by cooljimy84