esphome/issues #2405: Connecting one esphome device causes all other connected esphome devices to fail.

thejonesyboy, 2021-09-10:
### The problem

I have an esphome device "alarm_system" (esp8266 wemos D1 mini) that has worked normally, amongst about ten other esp devices consisting of esp8266 devices (wemos d1 mini, esp01 smart plugs etc) that also work normally in harmonious glory. Today I merely disconnected and reconnected the power supply to "alarm_system", and ALL other esphome devices spontaneously fail. I cannot comprehend what is happening. The issue is reproduceable every time. When I disconnect the alarm_system power supply, all other devices resume normal operation. See homeassistant esphome integration logs below, using the following config:

configuration.yaml:
```
logger:
  default: warning
  logs:
    homeassistant.components.esphome: debug
    aioesphomeapi.connection: debug
```

### Which version of ESPHome has the issue?

v1.20.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.1

### What platform are you using?

ESP8266

### Board

d1mini

### Example YAML snippet

```yaml
substitutions:
  name: alarm_system
  long_name: AV-GAD AV-2008 Security System

esphome:
  name: ${name}
  platform: ESP8266
  board: d1_mini
  comment: ${long_name}

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: ${name}
    password: !secret wifi_password

captive_portal:

logger:

api:

ota:

switch:

  - platform: gpio
    pin: D1
    id: alarm_key

  - platform: template
    name: "Arm/Disarm Toggle"
    icon: "mdi:shield-lock"
[... 140 lines cut ...]
2021-09-10 17:53:55 INFO (MainThread) [homeassistant.components.esphome] Disconnected from ESPHome API for 192.168.1.187
2021-09-10 17:53:55 INFO (MainThread) [aioesphomeapi.connection] arlec_plug_5 @ 192.168.1.187: Error while reading incoming messages: Error while receiving data: 0 bytes read on a total of 1 expected bytes
2021-09-10 17:53:55 DEBUG (MainThread) [aioesphomeapi.connection] arlec_plug_5 @ 192.168.1.187: Connecting to 192.168.1.187:6053 (AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.187', port=6053)))
2021-09-10 17:54:00 DEBUG (MainThread) [aioesphomeapi.connection] 192.168.1.242: Sending <class 'api_pb2.SwitchCommandRequest'>: key: 640615439
2021-09-10 17:54:03 DEBUG (MainThread) [aioesphomeapi.connection] 192.168.1.8: Sending <class 'api_pb2.PingRequest'>: 
2021-09-10 17:54:03 DEBUG (MainThread) [aioesphomeapi.connection] 192.168.1.8: Got message of type <class 'api_pb2.PingResponse'>: 
2021-09-10 17:54:03 DEBUG (MainThread) [aioesphomeapi.connection] 192.168.1.36: Sending <class 'api_pb2.PingRequest'>: 
2021-09-10 17:54:04 DEBUG (MainThread) [aioesphomeapi.connection] 192.168.1.242: Sending <class 'api_pb2.PingRequest'>: 
2021-09-10 17:54:04 DEBUG (MainThread) [aioesphomeapi.connection] 192.168.1.242: Got message of type <class 'api_pb2.PingResponse'>: 
2021-09-10 17:54:08 INFO (MainThread) [aioesphomeapi.connection] arlec_plug_2 @ 192.168.1.36: Ping Failed!
2021-09-10 17:54:08 DEBUG (MainThread) [aioesphomeapi.connection] arlec_plug_2 @ 192.168.1.36: Closed socket
2021-09-10 17:54:08 INFO (MainThread) [homeassistant.components.esphome] Disconnected from ESPHome API for 192.168.1.36
2021-09-10 17:54:08 INFO (MainThread) [aioesphomeapi.connection] arlec_plug_2 @ 192.168.1.36: Error while reading incoming messages: Error while receiving data: 0 bytes read on a total of 1 expected bytes
2021-09-10 17:54:08 DEBUG (MainThread) [aioesphomeapi.connection] arlec_plug_2 @ 192.168.1.36: Connecting to 192.168.1.36:6053 (AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.36', port=6053)))
2021-09-10 17:54:15 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for brilliant_plug_1 (192.168.1.221): Timeout while connecting to ('192.168.1.221', 6053)
2021-09-10 17:54:15 INFO (MainThread) [homeassistant.components.esphome] Trying to reconnect to 192.168.1.221 in the background
2021-09-10 17:54:15 DEBUG (MainThread) [homeassistant.components.esphome] Retrying 192.168.1.221 in 2 seconds
2021-09-10 17:54:17 DEBUG (MainThread) [aioesphomeapi.connection] brilliant_plug_1 @ 192.168.1.221: Connecting to 192.168.1.221:6053 (AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.221', port=6053)))
2021-09-10 17:54:18 DEBUG (MainThread) [aioesphomeapi.connection] 192.168.1.8: Sending <class 'api_pb2.PingRequest'>: 
2021-09-10 17:54:18 DEBUG (MainThread) [aioesphomeapi.connection] 192.168.1.8: Got message of type <class 'api_pb2.PingResponse'>: 
2021-09-10 17:54:18 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for arlec_plug_1 (192.168.1.227): Timeout while connecting to ('192.168.1.227', 6053)
2021-09-10 17:54:18 INFO (MainThread) [homeassistant.components.esphome] Trying to reconnect to 192.168.1.227 in the background
2021-09-10 17:54:18 DEBUG (MainThread) [homeassistant.components.esphome] Retrying 192.168.1.227 in 2 seconds
## this continues indefinitely until I disconnect alarm_system. Note that alarm_system or its most recent IP address do not come up.
```

oxan, 2021-09-10:
My first thought here is that the ESP in question somehow gets the IP address of your HA host assigned.

thejonesyboy, 2021-09-14:
> My first thought here is that the ESP in question somehow gets the IP address of your HA host assigned (or a similar network issue).

Good thinking but unfortunately I could see nothing on the router logs after power up.

thejonesyboy, 2021-09-14:
UPDATE

I didn't want to disconnect and reconnect the device or I/O just in case the issue disappeared and could never be troubleshooted - but that is exactly what has now happened. I disconnected all I/O pins (D1, D5, D6 & D7) and the device connects and receives an IP address and the issue is gone. I reconnected all I/O and the issue is still gone and I cannot reproduce it. What is going on? Could a bad I/O connection have caused a misbehaving wifi signal that shutdown all other esp devices? Who knows.

closed 2021-09-16 by thejonesyboy