esphome/issues #2408: ESP01-m with I2C sensors now crash from 2021.10.x

SeByDocKy, 2021-09-11:
### The problem

I just recompiled the same yaml associated with the ESP01m where some I2C sensors are attached on the UART port (GPIO01/GPIO03). Was working flawless before this upgrade.... Nonw  it's more sending information to HA... cycling between connexion/disconnextion.... 

### Which version of ESPHome has the issue?

2021.10.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.4

### What platform are you using?

ESP8266

### Board

ESP01m

### Component causing the issue

UART ?

### Example YAML snippet

```yaml
:
  name: mosquitto
  platform: ESP8266
  board: esp01_1m
  
  on_boot:
    priority: -100
    then:
      - light.turn_on: mosquito_led_status

wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "mosquitto Hotspot"
    password: "mosquitto"

captive_portal:

logger:
  baud_rate: 0 #need this to free up UART pins

# Enable Home Assistant API
api:

# enable over the air updates
ota:

# Enable Web server.
#web_server:
#  port: 80
  
i2c:
  sda: GPIO01  
  scl: GPIO03
  scan: True
  id: bus_i2c  
[... 422 lines cut ...]
[12:52:36][D][api.connection:703]: Client 'Home Assistant 2021.9.5 (192.168.1.2)' connected successfully!
[12:52:36][D][ads1115:163]: 'mosquitto_battery': Got Voltage=3.745125V
[12:52:36][D][sensor:118]: 'mosquitto_battery': Sending state 12.27326 V with 3 decimals of accuracy
[12:52:37][D][sensor:118]: 'WiFi puissance_mosquitto': Sending state -71.00000 dBm with 0 decimals of accuracy
[12:52:39][D][sensor:118]: 'uptime_sec': Sending state 47.67800 s with 0 decimals of accuracy
[12:52:40][D][bme280.sensor:197]: Got temperature=39.6°C pressure=1005.8hPa humidity=49.0%
[12:52:40][D][sensor:118]: 'mosquitto_temperature': Sending state 39.62000 °C with 1 decimals of accuracy
[12:52:41][D][text_sensor:015]: 'mosquitto Uptime': Sending state '0d 00h 00m 47s'
[12:52:41][D][ads1115:163]: 'mosquitto_battery': Got Voltage=3.745375V
[12:52:41][D][sensor:118]: 'mosquitto_battery': Sending state 12.27408 V with 3 decimals of accuracy
[12:52:42][D][sensor:118]: 'WiFi puissance_mosquitto': Sending state -71.00000 dBm with 0 decimals of accuracy
[12:52:44][D][api:095]: Accepted 192.168.1.2
[12:52:44][D][api.connection:703]: Client 'Home Assistant 2021.9.5 (192.168.1.2)' connected successfully!
[12:52:44][D][sensor:118]: 'uptime_sec': Sending state 52.67800 s with 0 decimals of accuracy
[12:52:45][D][bme280.sensor:197]: Got temperature=39.6°C pressure=1005.8hPa humidity=49.0%
[12:52:46][D][text_sensor:015]: 'mosquitto Uptime': Sending state '0d 00h 00m 52s'
[12:52:46][D][ads1115:163]: 'mosquitto_battery': Got Voltage=3.745250V
[12:52:47][D][sensor:118]: 'WiFi puissance_mosquitto': Sending state -71.00000 dBm with 0 decimals of accuracy
[12:52:49][D][sensor:118]: 'uptime_sec': Sending state 57.67800 s with 0 decimals of accuracy
[12:52:50][D][bme280.sensor:197]: Got temperature=39.6°C pressure=1005.8hPa humidity=49.0%
[12:52:51][D][text_sensor:015]: 'mosquitto Uptime': Sending state '0d 00h 00m 57s'
[12:52:51][D][ads1115:163]: 'mosquitto_battery': Got Voltage=3.745375V
[12:52:52][D][sensor:118]: 'WiFi puissance_mosquitto': Sending state -71.00000 dBm with 0 decimals of accuracy
[12:52:54][D][sensor:118]: 'uptime_sec': Sending state 62.67800 s with 0 decimals of accuracy
```

oxan, 2021-09-11:
There is no ESPHome 2021.10.1 (yet). Which version are you actually using?

SeByDocKy, 2021-09-11:
last dev

mmakaay, 2021-09-11:
Looks indeed like the same kind of issues that I have reported on Discord in the beta channel (you referenced this issue there).
Can you check if the device becomes stable when you do not attach the API logger to it? The biggest problems for me happened when both HA and the logger were attached to the API service on the device. Not a solution, but just a step to see if we're looking at the same kind of issue here.

reaper7, 2021-09-12:
I have the same problem on esp01 with the last dev.
It looks like a change **Convert API to use sockets** introduced this problem.
But for me, only on this one device, I do not observe this issue on my other devices.

If it helps, I also use UART port (GPIO01/GPIO03) for SDM Meter integration.

OttoWinter, 2021-09-13:
Potential fix in https://github.com/esphome/esphome/pull/2288

reaper7, 2021-09-13:
@OttoWinter 
update via 
```pip3 install --force-reinstall https://github.com/esphome/esphome/archive/dev.zip```
will already contain this fix ??

OttoWinter, 2021-09-13:
Yes

closed 2021-09-13 by OttoWinter

SeByDocKy, 2021-09-13:
Fix my problem thankx :)