esphome/issues #2409: TTGO T-Display can't compile with fonts

G2Dolphin, 2021-09-11:
### The problem

I had my T-Display with old ESPHome firmware (like half-year old?) working with displaying some text info.
Sadly it can't compile anymore if I use "font" section and a text in "lambda" section of "display".

It works nicely if I use some graphics though (like an "it.rectangle" example from [https://esphome.io/components/display/st7789v.html](url)).

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.5

### What platform are you using?

ESP32

### Board

TTGO T-Display

### Component causing the issue

display

### Example YAML snippet

```yaml
esphome:
  name: ttgo-test
  platform: ESP32
  board: featheresp32

wifi:
  ssid: !secret home_wlan_ssid
  password: !secret home_wlan_pass

logger:
ota:
api:

web_server:
  port: 80

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO19
  
display:
  - platform: st7789v
    id: st7789vdisplay
    reset_pin: GPIO23
    dc_pin: GPIO16
    cs_pin: GPIO5
    bl_pin: GPIO4
    brightness: 65%
    update_interval: 1s
    rotation: 90°
    lambda: |-
      it.print(0, 0, id(font1), "Hello World!");
    
font:
  - file: 'ttgo-test2/fonts/Arial-Black.ttf'
    id: font1
    size: 12
```

### Anything in the logs that might be useful for us?

```txt
src/main.cpp: In function 'void setup()':
src/main.cpp:250:25: error: 'GlyphData' in namespace 'esphome::display' does not name a type
   static const display::GlyphData display_glyphdata[] = {display::GlyphData{
                         ^
src/main.cpp:783:29: error: 'display_glyphdata' was not declared in this scope
   font1 = new display::Font(display_glyphdata, 76, 14, 18);
                             ^
*** [/data/ttgo-test2/.pioenvs/ttgo-test2/src/main.cpp.o] Error 1
```

oxan, 2021-09-11:
That's weird, your exact YAML compiles fine for me (if I replace `bl_pin` with `backlight_pin`). Can you try removing the temporary directory?

G2Dolphin, 2021-09-11:
> That's weird, your exact YAML compiles fine for me (if I replace `bl_pin` with `backlight_pin`). Can you try removing the temporary directory?

I don't have any temporary directories around (other than .esphome, if it is the one), but your message made me wonder, why you would need to replace this variable.
After all I've found old custom_component of "st7789v" I added when it wasn't available on ESPHome itself, looks like latest update broke something with this library. After removing old library and restarting ESPHome (and renaming variable in YAML, of course) it could finally compile again.

I'm deeply sorry for rushing with creating a bug-report since it was my fault after all, but without your hint I would not have come across that custom_component soon enough. Thank you. :)

closed 2021-09-11 by G2Dolphin