esphome/issues #2413: Color_Brightness Control has no effect on addressable SK6812 Color Channels

tk4arts, 2021-09-12:
### The problem

Hello there,
I have a Wemos D1 Mini with ESP Home flashed to use it as a controller for my SK6812 individually adressable LEDs. Everything is working like a charm except for one issue I have. The color_brightness slider in Home Assistant is not changing anything. The slider for the white brightness is working and the overall brightness slider as well. With this configuration it is not possible to have only the white channel of the LEDs on while the others are off. Unfortunately this is what I need and what is technically possible by the SK6812 and the D1 Mini. It just feels like ESP Home does not do anything with the value from this slider.
If one of the developers could try and fix the lose end between the color_brightness slider and the neopixelbus integration that would be great. 
Thanks in advance!

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8.2

### What platform are you using?

ESP8266

### Board

Wemos D1-mini

### Component causing the issue

neopixelbus

oxan, 2021-09-15:
I don't have a SK6812, but I can't reproduce this with my LED strips. Are you sure you're running 2021.8? This was a problem with earlier versions but should be resolved now. 

In any case we're going to need more information, including at least your config and logs, to say anything sensible about this.

tk4arts, 2021-09-15:
Thanks for the reply. I will try to gather the needed information.

Below is a screenshot of my esphome addon page in homeassistant. That says 2021.8.2 so I am pretty sure this is the version I am running.
[image]

I now managed to get it working like it should with no effect running. 
The problem is that I want to run a custom lambda effect.
To set the color of the Leds I used:
`it.all() = current_color;`

And then the color brightness slider is not working.
Is there another way to access the currently set color and apply it to individual leds or the complete strip?

Thanks in advance!
Till

oxan, 2021-09-15:
Ah, it doesn't apply to effects. That's the missing piece of information.

tk4arts, 2021-09-16:
Yes sorry that I just discovered that yesterday. 
Is there another possiblitiy to make use of the rgb values, white brightness and color brightness in adressable lambda effects?

closed 2021-09-19 by OttoWinter