esphome/issues #2415: Automations fail when cabled Ethernet gets disconnected

adezerega, 2021-09-12:
### The problem

I have set an ESP32 board with built-in ethernet, with a manual IP address. The idea is that the automations work even if the network goes down. However, as soon as the network goes down (simulated by disconnecting the ethernet cable), the automations stop working.

As long as the network cable is connected, when I trigger input INT01 (with MCP23017 chip), the switch THC01 toggles. When the network cable is not connected, this simple automation stops working. 

As per the documentation, automations should work even if the network is disconnected. The only "side effect" is that the controller resets every 15 mins, since the api: cannot reach home assistant.

I can't disable the Wifi reboot timer, since ESPHome says that I can't have the wifi: component at the same time as the ethernet: component.

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.9.5

### What platform are you using?

ESP32

### Board

LILYGO TTGO T-Internet-POE ESP32 WROOM LAN8270A

### Component causing the issue

ethernet:

### Example YAML snippet

```yaml
esphome:
  name: esp32cabled
  platform: ESP32
  board: esp-wrover-kit
logger:
# Enable Home Assistant API
api:
ota:
i2c:
  sda: 15
  scl: 14
  scan: True
  id: bus_a
esp32_ble_tracker:
time:
  - platform: ds1307
    id: ds1307_time
    address: 0x68
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  manual_ip:
    static_ip: 172.16.1.150
    gateway: 172.16.1.1
    subnet: 255.255.255.0
    dns1: 172.16.1.1
    dns2: 8.8.8.8
mcp23017:
  - id: 'MCP23017_1'
    address: 0x20
  - id: 'MCP23017_2'
    address: 0x21
switch:
  - platform: gpio
    name: "THC01"
    pin:
      mcp23xxx: MCP23017_1
      number: 8
      mode: OUTPUT
      inverted: True
    restore_mode: RESTORE_DEFAULT_OFF
    id: CSW1
binary_sensor:
  - platform: gpio
    name: "INT01"
    pin:
      mcp23xxx: MCP23017_1
      number: 0
      mode: INPUT
      inverted: True
    filters:
      - delayed_on_off: 100ms
    on_state:
      then:
        - switch.toggle: CSW1
```

jesserockz, 2021-09-13:
Can you capture logs via serial for the whole process?

adezerega, 2021-09-13:
I’ll sure do - but need some time (the microcontroller is embedded in a wall!)

oxan, 2021-09-13:
> I can't disable the Wifi reboot timer, since ESPHome says that I can't have the wifi: component at the same time as the ethernet: component.

The reboot timeout is a setting under `api`, so you should be able to disable that.

adezerega, 2021-09-17:
Attached find the logger output - as you can see, the ESP32 crashes when the network is disconnected. After inspecting the log, there is an issue with the Real Time Clock - which I will review in parallel

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[0;32m[I][logger:174]: Log initialized[0m
[0;35m[C][ota:379]: There have been 1 suspected unsuccessful boot attempts.[0m
[0;32m[I][app:029]: Running through setup()…[0m
[0;35m[C][mcp23017:010]: Setting up MCP23017…[0m
[0;35m[C][mcp23017:010]: Setting up MCP23017…[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch ‘THC01’…[0m
[0;36m[D][switch:025]: ‘THC01’ Turning OFF.[0m
[0;36m[D][switch:045]: ‘THC01’: Sending state OFF[0m
[0;36m[D][switch:025]: ‘THC01’ Turning OFF.[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch ‘THC02’…[0m
[0;36m[D][switch:025]: ‘THC02’ Turning OFF.[0m
[0;36m[D][switch:045]: ‘THC02’: Sending state OFF[0m
[0;36m[D][switch:025]: ‘THC02’ Turning OFF.[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch ‘THC03’…[0m
[0;36m[D][switch:025]: ‘THC03’ Turning OFF.[0m
[0;36m[D][switch:045]: ‘THC03’: Sending state OFF[0m
[0;36m[D][switch:025]: ‘THC03’ Turning OFF.[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch ‘THC04’…[0m
[0;36m[D][switch:025]: ‘THC04’ Turning OFF.[0m
[0;36m[D][switch:045]: ‘THC04’: Sending state OFF[0m
[0;36m[D][switch:025]: ‘THC04’ Turning OFF.[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch ‘THC05’…[0m
[0;36m[D][switch:025]: ‘THC05’ Turning OFF.[0m
[0;36m[D][switch:045]: ‘THC05’: Sending state OFF[0m
[0;36m[D][switch:025]: ‘THC05’ Turning OFF.[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch ‘THC06’…[0m
[0;36m[D][switch:025]: ‘THC06’ Turning OFF.[0m
[0;36m[D][switch:045]: ‘THC06’: Sending state OFF[0m
[0;36m[D][switch:025]: ‘THC06’ Turning OFF.[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch ‘THC07’…[0m
[0;36m[D][switch:025]: ‘THC07’ Turning OFF.[0m
[0;36m[D][switch:045]: ‘THC07’: Sending state OFF[0m
[0;36m[D][switch:025]: ‘THC07’ Turning OFF.[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch ‘THC08’…[0m
[0;36m[D][switch:025]: ‘THC08’ Turning OFF.[0m
[0;36m[D][switch:045]: ‘THC08’: Sending state OFF[0m
[0;36m[D][switch:025]: ‘THC08’ Turning OFF.[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch ‘THC09’…[0m
[0;36m[D][switch:021]: ‘THC09’ Turning ON.[0m
[0;36m[D][switch:045]: ‘THC09’: Sending state ON[0m
[0;36m[D][switch:021]: ‘THC09’ Turning ON.[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch ‘THC10’…[0m
[0;36m[D][switch:025]: ‘THC10’ Turning OFF.[0m
[0;36m[D][switch:045]: ‘THC10’: Sending state OFF[0m
[0;36m[D][switch:025]: ‘THC10’ Turning OFF.[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch ‘THC11’…[0m
[0;36m[D][switch:025]: ‘THC11’ Turning OFF.[0m
[0;36m[D][switch:045]: ‘THC11’: Sending state OFF[0m
[0;36m[D][switch:025]: ‘THC11’ Turning OFF.[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch ‘THC12’…[0m
[0;36m[D][switch:025]: ‘THC12’ Turning OFF.[0m
[0;36m[D][switch:045]: ‘THC12’: Sending state OFF[0m
[0;36m[D][switch:025]: ‘THC12’ Turning OFF.[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch ‘THC13’…[0m
[0;36m[D][switch:025]: ‘THC13’ Turning OFF.[0m
[0;36m[D][switch:045]: ‘THC13’: Sending state OFF[0m
[0;36m[D][switch:025]: ‘THC13’ Turning OFF.[0m
[... 611 lines cut ...]
[0;35m[C][ethernet:246]:   DNS1: 172.16.10.1[0m
[0;35m[C][ethernet:247]:   DNS2: 8.8.8.8[0m
[0;35m[C][ethernet:250]:   MAC Address: 9C:9C:1F:XX:XX:XX [0m
[0;35m[C][ethernet:251]:   Is Full Duplex: YES[0m
[0;35m[C][ethernet:252]:   Link Up: YES[0m
[0;35m[C][ethernet:253]:   Link Speed: 100[0m
[0;35m[C][ethernet:128]:   MDC Pin: 23[0m
[0;35m[C][ethernet:129]:   MDIO Pin: 18[0m
[0;35m[C][ethernet:130]:   Type: LAN8720[0m
[0;35m[C][ota:029]: Over-The-Air Updates:[0m
[0;35m[C][ota:030]:   Address: 172.16.10.156:3232[0m
[0;35m[C][ota:032]:   Using Password.[0m
[0;33m[W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts[0m
[0;35m[C][api:135]: API Server:[0m
[0;35m[C][api:136]:   Address: 172.16.10.156:6053[0m
[0;36m[D][esp32_ble_tracker:628]: Found device XX:XX:XX:XX:XX:XX RSSI=-72[0m
[0;36m[D][esp32_ble_tracker:649]:   Address Type: PUBLIC[0m
[0;36m[D][esp32_ble_tracker:651]:   Name: ‘LYWSD03MMC’[0m
[0;36m[D][esp32_ble_tracker:628]: Found device XX:XX:XX:XX:XX:XX RSSI=-74[0m
[0;36m[D][esp32_ble_tracker:649]:   Address Type: PUBLIC[0m
[0;36m[D][esp32_ble_tracker:651]:   Name: ‘LYWSD03MMC’[0m
[0;36m[D][sensor:118]: ‘POE-1’: Sending state 36.80100 s with 0 decimals of accuracy[0m
[0;36m[D][bmp280.sensor:149]: Got temperature=25.4Â°C pressure=956.5hPa[0m
[0;36m[D][sensor:118]: ‘Temperatura control’: Sending state 25.38000 Â°C with 1 decimals of accuracy[0m
[0;36m[D][sensor:118]: ‘Presion control’: Sending state 956.47467 hPa with 1 decimals of accuracy[0m

adezerega, 2021-09-17:
I fixed the Real Time Clock issue, the error continues as before. It seems that when the network cable is disconnected, the ESP32 crashes.

Extract of log (VERBOSE):

[0;36m[D][sensor:118]: ‘POE-1’: Sending state 44.79600 s with 0 decimals of accuracy[0m
[0;36m[D][binary_sensor:036]: ‘INT07’: Sending state ON[0m
[0;36m[D][switch:029]: ‘THC07’ Toggling ON.[0m
[0;36m[D][switch:045]: ‘THC07’: Sending state ON[0m
[0;36m[D][binary_sensor:036]: ‘INT07’: Sending state OFF[0m
[0;36m[D][switch:029]: ‘THC07’ Toggling OFF.[0m
[0;36m[D][switch:045]: ‘THC07’: Sending state OFF[0m

Bro ets Jun  8 2016 00:22:57 ## "Bro" is logged when network cable is disconnected, and reboot follows

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[0;32m[I][logger:174]: Log initialized[0m
[0;35m[C][ota:379]: There have been 1 suspected unsuccessful boot attempts.[0m
[0;32m[I][app:029]: Running through setup()…[0m
[0;37m[V][app:030]: Sorting components by setup priority…[0m

adezerega, 2021-09-17:
I guess I found the "bug", which was not in the code, but in my circuit design :(

When I disconnected the network cable, the power to the ESP32 went down. Then I realized that I hadn't powered the ESP32 board from the power supply - a missing cable - but it was being powered from the PoE subsystem. The logs showed exactly what happens - the POWERON_RESET gave me the hint.

Thanks for the help. ESPHome is a great project

closed 2021-09-17 by adezerega