esphome/issues #2416: Adding example esp add on CONFIG puts home-assist into safe mode

kohekohebigsky, 2021-09-13:
### The problem

Cannot connect to esp server with https

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.09.0

### What platform are you using?

ESP32

### Board

do it dev V1

### Component causing the issue

n/a

### Example YAML snippet

```yaml
Not getting that far
```

### Anything in the logs that might be useful for us?

```txt
Was using Firefox browser.  After installing chrome I am still not able to make a https connection to esp
I have generated the OEM files and when I add the example CONFIG to HA it goes into safe mode
```

oxan, 2021-09-13:
You'll need to give a more explicit information about what exactly you did and what exactly happened (including any and all logs, if possible) for anybody to be able to help you or possibly fix this.

kohekohebigsky, 2021-09-13:
Hi thanks for your reply,
I would appreciate some advice from an experienced user like yourself.
It seems like everything I am trying to do is complicated by my architecture.
Can you comment?
HP290 X86 i5-8400 running Ubuntu 20.04 Virtualbox VM running home assistant OS
Most all the addons require CONFIG.
So I have installed VS Studio Code addon
Problems start when documents advise: go to /some folder and run command to add pem files.  I am abstracted from the file system through the HA is and VM. I have never seen the ha docker containers or files as I am on the Ubuntu GUI browser or terminal
Should I use a different setup?
Went for this as all the docs say OS is best for non commandline folks.
Regards Don

closed 2022-01-26 by github-actions[bot]